JenniferBuehler/graspit-pkgs #11: How to view .iv results for grasp_planning.cpp ?

belledon, 2016-06-15:
I was using the grasp_planning_graspit package to automate grasps. The xml world files opened in Graspit perfectly. However, I could not view the iv versions with ivviewer. I'm assuming its because the iv file is not a single mesh. I was wondering what would be an appropriate way to parse the files? Could one separate the objects and then rerender them with their relative positions? My ultimate goal is to convert these files (or the xml version, which ever is more reliable) to a more popular mesh variant (like waveform) while keeping the hand and the graspable object as distinct from each other.

JenniferBuehler, 2016-06-15:
You should be able to see the individual meshes (parts of the robots) with an IV viewer, but yes, you won't be able to see the whole hand/robot. The links are connected by joints, and the transforms for the joints are defined in the graspit xml file with the Denavit-Hartenberg parameters.

Do you have the robot as URDF file as well? You can use the package urdf2inventor in [urdf-tools-pkgs](https://github.com/JenniferBuehler/urdf-tools-pkgs/wiki) to convert the full robot to inventor, however the format will not have a concept of joints so you won't be able to move them around.

If ivviewer doesn't even display the links, you can also try ``rosrun urdf_viewer_node urdf_viewer <path-to-iv-file> --iv`` (you will need besaid urdf-tools-pkgs repo). But this will also only display the individual links.

The package urdf2graspit (in the graspit-pkgs repository you already have) can convert from urdf to graspit format, which means it does the conversion from URDF to Denavit-Hartenberg parameters. I've been planning to do it the other way round (DH to URDF) as well, which should be easier, but it's not implemented yet. This may give me the incentive to do it some time soon ;) If I do it I'll post the news here, but it may be a few days (maybe weeks) yet.

Meanwhile, let me know if you have a URDF of your robot, or if you are referring to a robot already provided with the graspit models.

belledon, 2016-06-15:
I'm sorry, I forgot to mention which robot. I am using the 16dof human hand provided by graspit ( but I could use a urdf imported human hand if you know of one). Also I understand that once the grasps are imported into iv they are no long joints which is perfectly ok for my purposes. All I would need is to be able to distinguish between the hand and object meshes (so I could hide the object when needed.. I am also aware that I could just delete the object from within the grasp planner, the former seems cleaner). 

Also I am not using ros (I can set it up if it would be better). I am relatively new to robot simulation but if you think that it would be relatively easy for me to create my own urdf human hand, I'll give it a try so I can convert the robot into iv and then pull the coordinates for the robot and object positions from the xml file. 

I have another question about EigenTransform. Would you like me to open another issue or can I comment here?

Thanks again for your time!

JenniferBuehler, 2016-06-15:
Ok, now I understand better what you're trying to do, thanks. Still I'll repeat, to clarify I got it right: You want the human hand in the grasping state (after planning), but without the object, as one whole IV file (and/or the object separately as well). That shouldn't be hard to do. I can add the changes required. Note however that the relative transform hand<->object will get lost this way, would this still work for you?

If you haven't built a URDF before, it will be a fiddly process, for quick prototyping probably not the best approach if you don't need the URDF otherwise anyway. You don't need ROS for using graspit, it just works with ROS as well.

Can you give me a hint about the EigenTransform question? If it's something major it would be better to create a new issue, if it's just a quick question, we can do it here.

belledon, 2016-06-15:
As long as its possible for me to generate the relative positions from the original xml (object held by the hand) so that I could still render both (in Blender etc) if need be then yes. (I also really don't want you to have to do any extra work for this). And thanks for the input about urdf. 

With regards to the EigenTransform there are two components. One I need to be able to determine the object's bounding box (or something like a xyz min max range) before grasping. Secondly, I would like to then position the hand (or the object relative to the hand) along the bounding box.

JenniferBuehler, 2016-06-15:
I think you should be able to reconstruct it from the XML, but I'll make sure that's the case.

The extra work is minimal. I'm doing it now, but won't have much more time today so I may finish it tomorrow and then let you know here.

The EigenTransform gives you a transformation matrix (rotation/translation of a joint or link in the chain). This is not enough for a bounding box... Off the top of my head I don't know if there's a bounding box function in Coin (Inventor) or GraspIt, but I'll check.
Maybe you could ask this question separately in a new issue, how to get the bounding box of the object. Thanks!

belledon, 2016-06-15:
I really appreciate it! Will do.

JenniferBuehler, 2016-06-15:
Managed to put aside a bit of extra time just now and uploaded the changes. You will need the extra parameter ``--save-separate`` in ``grasp_planning``. The poses will be saved as ``world_i_pose.iv`` in the results directory.
You can reconstruct the object-hand transform from the graspit xml result.

For converting to other formats, I think I've seen an inventor importer for assimp around somewhere, but I don't remember where. Ask google our friend ;)

belledon, 2016-06-16:
I added the changes, however when I pass `--dir string  --rob string --obj string --save-separate true` I get an unrecognized option error. I plan to always save separate so I can just hardcode saveSeparate to true for now.

JenniferBuehler, 2016-06-16:
You only need to pass ``--save-separate``, without the ``true`` :)

belledon, 2016-06-16:
I think its because I'm calling the old grasp. Sorry I'm a novice. Do I recompile using gcc or should I use cmake?

JenniferBuehler, 2016-06-16:
No problem at all.
Yes you need to rebuild, just type make while in the build directory. You don't need another run of cmake, unless you are rebuilding from scratch.

You could for example do:

``./grasp_planning --wld <path-to-a-graspit-world.xml> --dir <path-to-results-dir> --save-separate``

That's using a world file, but you can also use robot and object files separately with ``--rob`` and ``--obj`` as you did above.

belledon, 2016-06-16:
Thanks it worked! The iv files are viewable in graspit_simulator

closed 2016-06-19 by JenniferBuehler