JenniferBuehler/graspit-pkgs #13: Changing annealing parameters

belledon, 2016-06-28:
Looking through the source from the graspit_simulator github I found that the grasp planning proceeds as  follows (I could be wrong): The abstract planning classes call upon `SimAnnPlanner` which then loops using `simAnn`. Within `simAnn` lies `setParameters` which uses a switch to set the annealing parameters. 

```cpp
void SimAnn::setParameters(AnnealingType type)
{
	switch (type) {
		case ANNEAL_DEFAULT:
			//these are the default parameters that experimentation shows work best over 8 dimensions
			//identical schedule for error and neighbors
			//error is supposed to be distance between fingers and object, so raw order of 1-100 mm
			YC = 7.0;
			HC = 7.0;
			YDIMS = 8;
			HDIMS = 8;
			NBR_ADJ = 1.0;
			ERR_ADJ = 1.0e-6;
			DEF_T0 = 1.0e6;
			DEF_K0 = 30000;
			break;
		case ANNEAL_LOOP:
			//for looping we use a slightly shorter interval
			YC = 7.0;
			HC = 7.0;
			YDIMS = 8;
			HDIMS = 8;
			NBR_ADJ = 1.0;
			ERR_ADJ = 1.0e-6;
			DEF_T0 = 1.0e6;
			DEF_K0 = 35000;
			break;
		case ANNEAL_MODIFIED:
			//different version which doesn't really work better than default
			YC = 2.38;
			YDIMS = 8;
			HC = 2.38;
			HDIMS = 8;
			NBR_ADJ = 1.0e-3;
			ERR_ADJ = 1.0e-1;
			DEF_T0 = 1.0e3;
			DEF_K0 = 0;
			break;
		case ANNEAL_STRICT:
			DBGP("Strict Sim Ann parameters!");
			//this are for searches that ONLY look at grasp quality measure
			YC = 0.72; 
			HC = 0.22;
			YDIMS = 2;
			HDIMS = 2;
			NBR_ADJ = 1.0;
			ERR_ADJ = 1.0; //meant to work with ENERGY_STRICT_AUTOGRASP which multiplies eps. gq by 1.0e2
			DEF_T0 = 10.0;
			DEF_K0 = 0; 
			break;
		case ANNEAL_ONLINE:
			//parameters for on-line search. Supposed to work really fast and in fewer dimensions
			//YC = 0.54;
			YC = 0.24;
			HC = 0.54;
			//YDIMS = 3;
			YDIMS = 2;
			HDIMS = 3;
			NBR_ADJ = 1.0;
			ERR_ADJ = 1.0e-2;
			DEF_T0 = 1.0e1;
			DEF_K0 = 100;
			break;
		default:
			fprintf(stderr,"Unknown Annealing params requested, using default!\n");
			setParameters(ANNEAL_DEFAULT);
			break;
	}
}
```
I was thinking that creating a modified version of `simAnnPlanner` that calls on a different `simAnn` can enable a way to iterate through different parameters. I noticed that your version of `EigenGraspPlanner.cpp`, you include the helper files for both of the previously mentioned sources. In order to use the modified version of `simAnnPlanner` and `simAnn` would I have to compile it through graspit first and then compile a modified `EigenGraspPlanner.cpp` in grasp_planning_graspit?

JenniferBuehler, 2016-06-28:
Yes, this would be the way to go. It will require a PR to the main graspit_simulator repo. Temperature (DEF_T0) and step (DEF_K0) would be the parameters I would start playing with. I'd suggest it would make sense to add a new method ``simAnn::getDefaultParameters(AnnealingType type, AnnealingParamters& params)`` with the new struct ``AnnealingParameters`` which contains all fields (eg. DEF_T0, YDIMS, etc). ``getDefaultParameters()`` would then write the defaults according to AnnealingType into the params variable, similar to the existing ``simAnn::setParameters(AnnealingType type)``.
Then, after getting the desired default values read into the ``params``, selected values can be modified/overwritten, before passing ``params`` into ``simAnn::setParameters(AnnealingParameters& params)`` (a modified version, *without* AnnealingType parameter, which just sets all fields according to ``params``).
Makes sense?

This may take a while to be merged into graspit_simulator though, as there's a lot going on in this repository currently.

Once that's done, there will be a few changes to add more parameters to  the ``EigenGraspPlanner::plan()`` method, probably using a more elaborate parameters struct instead of only the existing ``PlannerType`` parameter. This could maybe be a struct/class that allows to specify the ``PlannerType`` along with the search parameters for the individual planner types. This will affect ``EigenGraspPlanner::initPlannerType()`` as well.

Overall, I suspect this will be a bit of a longer process to merge things, but it would be good to get started.

belledon, 2016-06-30:
Sorry for the delay. Ok thanks! I'll get started on it today or tomorrow.

belledon, 2016-07-19:
Sorry about the delay again. I started working on what was discussed above. I was able to compile it within original graspit fork. However, when I try to compile grasp_planning_graspit with that graspit fork it gives the following error:

```
cmake -DBUILD_GRASPIT:BOOL=false -DGRASPIT_PATH=/home/user/graspit -DCMAKE_INSTALL_PREFIX=/home/user/bin ..
-- Compiling grasp_planning_graspit: CATKIN_DEVEL_PREFIX is not defined, so **enforcing compilation with cmake**.
-- Finding graspit...
CMake Error at CMakeLists.txt:63 (find_package):
  By not providing "Findgraspit.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "graspit", but
  CMake did not find one.

  Could not find a package configuration file provided by "graspit" with any
  of the following names:

    graspitConfig.cmake
    graspit-config.cmake

  Add the installation prefix of "graspit" to CMAKE_PREFIX_PATH or set
  "graspit_DIR" to a directory containing one of the above files.  If
  "graspit" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/home/user/graspit-pkgs/grasp_planning_graspit/build/CMakeFiles/CMakeOutput.log".
```

After going through the main CMakeLists.txt I think it is because there is no Findgraspit.cmake in CMakeMacros. I'm not very experience but I will go ahead and try to write one.

JenniferBuehler, 2016-07-19:
Hi,

sorry I didn't get around to look at it any more either. I started a new project so I'm pretty busy, but will have time to look at it in more detail on the weekend.

You don't need ``-DGRASPIT_PATH``, try ``-DCMAKE_PREFIX_PATH`` instead. Let me know if it works.

Jenny

JenniferBuehler, 2016-07-19:
By the way, may I ask where you got ``-DGRASPIT_PATH`` from, is there an old piece of documentation (wiki) somewhere that I am not aware of? That was used ages ago but has been replaced with ``-DCMAKE_PREFIX_PATH``, it will help me if I know where there may be old documentation flying about.

belledon, 2016-07-19:
Hi Jenny, 

I was following the no ros install instructions from [https://github.com/JenniferBuehler/graspit-pkgs/wiki/Installation-without-ROS](url). I'll give it a shot

belledon, 2016-07-19:
I forgot to mention that I also pulled a copy of graspitConfig.cmake from your graspit fork and placed it in the graspit directory. After using `-DCMAKE_PREFIX_PATH`, it gets stuck either are finding qhul (so it ignores the `findQhul.cmake` in `graspit/CMakeMacros`. If I put all of the finders and `graspitConfig.cmake` in one directory it finds them but can't find `graspit-targets.cmake`. I searched to make sure it wasn't just named differently and I found `graspit-lib-targets.cmake` (assuming its the same). I edited the `graspitConfig.cmake' at line 91 with the other target cmake but it gave me the same could not load/ find error. 

-Mario

belledon, 2016-07-19:
To be more helpful, this is the error message:

```
cmake -DBUILD_GRASPIT:BOOL=false -DCMAKE_PREFIX_PATH=/home/user/graspit  ..
-- Compiling grasp_planning_graspit: CATKIN_DEVEL_PREFIX is not defined, so **enforcing compilation with cmake**.
-- Finding graspit...
-- Looking for graspit...
-- Looking for graspit headers -- found /home/include
-- Looking for graspit library include -- found /home/lib
-- Looking for graspit library -- found /usr/local/lib/libgraspit.so
-- Using graspit SHARED lib
CMake Error at /home/user/graspit/graspitConfig.cmake:91 (include):
  include could not find load file:

    /home/user/graspit/../graspit-lib-targets.cmake
Call Stack (most recent call first):
  CMakeLists.txt:63 (find_package)

-- QHULL found (include: /usr/include;/usr/include/libqhull, lib: optimized;/usr/lib/x86_64-linux-gnu/libqhull.so;debug;/usr/lib/x86_64-linux-gnu/libqhull.so)
-- Looking for SoQt headers -- found /usr/include/Inventor/Qt/SoQt.h
-- Looking for SoQt library -- found /usr/lib/x86_64-linux-gnu/libSoQt4.so
-- Looking for SoQt library debug -- not found
-- Result of soqt_config: -lSoQt -lQtOpenGL -lQtGui -lQtCore -lXmu -lXi -lCoin -lGL -lXext -lSM -lICE -lX11 -ldl -lpthread

-- soqt_config failed? 0
-- A library with BLAS API found.
-- A library with LAPACK API found.
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   filesystem
--   system
--   thread
--   program_options
-- Could NOT find Doxygen (missing:  DOXYGEN_EXECUTABLE) 
-- Configuring incomplete, errors occurred!
See also "/home/user/graspit-pkgs/grasp_planning_graspit/build/CMakeFiles/CMakeOutput.log".
See also "/home/user/graspit-pkgs/grasp_planning_graspit/build/CMakeFiles/CMakeError.log".
```

JenniferBuehler, 2016-07-20:
You should not have to copy anything around... is ``/home/user/graspit`` the github repository clone, or the location where you installed graspit? Did you make a ``make install`` for graspit?

Make sure you remove all the files you manually copied around and then do it afresh. In graspit fork repo:

```
cmake -DCMAKE_INSTALL_PREFIX=/home/user/graspitInstall ..
make
make install
```

Now you should have the graspitConfig.cmake in ``/home/user/graspitInstall/lib``.

Then in grasp_planning_graspit:

```
cmake -DCMAKE_PREFIX_PATH=/home/user/graspitInstall
make
```

I'll check if I have to update the documentation on the wiki

JenniferBuehler, 2016-07-20:
Updated wiki

belledon, 2016-07-21:
I cloned the graspit fork into a folder and installed into another folder called `graspitInstall` but the files inside `graspitInstall/lib` were: `graspit-targets.cmake	graspit-targets-release.cmake  libgraspit.so` and a folder containting the find_package macros called `graspit`. I searched in the install directory and could not find `graspitConfig.cmake`

JenniferBuehler, 2016-07-22:
Strange, the graspitConfig.cmake should acutally  be in graspitInstall/lib/graspit. I'm just testing this again now separately on a completely new build

JenniferBuehler, 2016-07-22:
Right so I tried it, and it works for me. To repeat, this is what I've done after checking out the current version of my fork, and of graspit-pkgs afresh:

```
cd /home/jenny/graspit-fork
mkdir build && cd build
cmake -DCMAKE_INSTALL_PREFIX=/home/jenny/graspitInstall ..
make && make install
cd /home/jenny/graspit-pkgs-test
mkdir build && cd build
cmake -DCMAKE_PREFIX_PATH=/home/jenny/graspitInstall ..
make
```

Compiles, and:
```
ls /home/jenny/graspitInstall/lib/graspit
FindBULLET.cmake  FindQhull.cmake  FindSoQt4.cmake  graspitConfig.cmake
```

However there was one thing I changed in graspit-pkgs/grasp_planning_graspit/CMakeLists.txt: I reverted the default of ENFORCE_CATKIN which was still set to true. This flag is required to build on jenkins to add it as ROS package. I set it back to false as it normally should be. However this is not related to the issue you are having.

Can you please try again?

JenniferBuehler, 2016-07-22:
Are you SURE you checked out the fork

``git clone git@github.com:JenniferBuehler/graspit.git``

and **not** the original

``https://github.com/graspit-simulator/graspit.git``

Because the original compiles with cmake, but still does not have a graspitConfig.cmake

belledon, 2016-07-22:
I'm sorry, I was using the original graspit fork and not yours. I now see `graspitConfig.cmake` in the right place. After following your instructions I noticed that without the flag `-DBUILD_GRASPIT:BOOL=false`, cmake still attempts to clone and build graspit internally. When I include the flag, I receive the following error 

```
cmake -DCMAKE_INSTALL_PREFIX=/home/mario/graspitInstall -DBUILD_GRASPIT:BOOL=false .. 
...
-- Using graspit SHARED lib
CMake Error at /home/mario/graspit/cmake/graspitConfig.cmake:90 (include):
  include could not find load file:
```

After changing the name of the directory `home/mario/graspit/cmake`, cmake began looking the install directory instead. But it couldn't find `graspit-targets.cmake`. 

```
-- Using graspit SHARED lib
CMake Error at /home/mario/graspitInstall/lib/graspit/graspitConfig.cmake:90 (include):
  include could not find load file:

    /home/mario/graspitInstall/lib/graspit/../graspit-targets.cmake
Call Stack (most recent call first):
  CMakeLists.txt:63 (find_package)
```

belledon, 2016-07-22:
I just realized I was using `DCMAKE_INSTALL_PREFIX` instead of `DCMAKE_PREFIX_PATH`

JenniferBuehler, 2016-07-22:
Oh yeah, you have to use ``-DBUILD_GRASPIT=false`` if you want to use the code of my fork directly! That is only useful if you want to modify the code in my fork, which I presume you will do (if you don't need to modify, you don't need to add my fork at all, it will then be pulled and built by grasp_planning_graspit).

belledon, 2016-07-22:
I modified a few files to allow passing annealing parameters into `SimAnn`. For now I was just planning on compiling the modified graspit with grasp_planning. Do you recommend going about it differently?

belledon, 2016-07-22:
I still get the same error about there not being `graspit-targets.cmake`. I noticed that in line 90 of `graspitConfig.cmake` there are a few statements about which target file to chose. Could it be that it should be using `graspit-static-targets.cmake`? (The former isn't created on my system)

JenniferBuehler, 2016-07-22:
No, the graspit-static-targets is only required when -DBUILD_GRASPIT is true (which is default). Try to remove the complete install directory and re-install the fork and grasp_planning_graspit.

belledon, 2016-07-22:
Just to be safe, is there any particular way I should remove them?

JenniferBuehler, 2016-07-22:
did you install only the graspit stuff in the directory? Eg. as in my ``/home/jenny/graspitInstall``?
Then just do ``rm -rf <path>/graspitInstall``, and re-run ``make install`` on the fork and grasp_planning_graspit. But be sure you have compiled grasp_planning_graspit with ``-DBUILD_GRASPIT=false``

belledon, 2016-07-22:
Thank you! It worked with the call `cmake -DCMAKE_PREFIX_PATH=/home/mario/graspitInstall -DBUILD_GRASPIT=false -DCMAKE_INSTALL_PREFIX=/home/mario/graspPlanningInstall ..` However I noticed that it used the static library:
```
-- Looking for graspit library -- found /home/mario/graspitInstall/lib/libgraspit.a
-- Using graspit STATIC lib
```

JenniferBuehler, 2016-07-22:
Hm. It must have installed the static library because before you built it once as static library.

Remove the libgraspit.so from your graspitInstall directory and try again.

Otherwise, if you do a complete re-build (remove grasp_planning_graspit build directory and build afresh) the issue should disappear for sure. You should not use the static lib because that's then not going to include your changes you make to the fork.

belledon, 2016-07-22:
I may have not installed them locally at some time in the past. For the most recent installation, I deleted the folders and recloned them to start from scratch.

belledon, 2016-07-22:
But it would have been the original graspit fork

belledon, 2016-07-22:
I tried one more time just to be safe (from your fork) and it still installs the static library `libgraspit.a`

JenniferBuehler, 2016-07-22:
Oh ok... maybe you need ``-DBUILD_SHARED_LIBS`` (it somehow is true by default on my system)

JenniferBuehler, 2016-07-22:
on the fork, that is

belledon, 2016-07-22:
It creates the shared library now but cmake can't find `graspit-targets.cmake` again. This time its really weird because its `<graspitInstall>/lib/graspit`. 

```
-- Using graspit SHARED lib
CMake Error at /home/mario/graspitInstall/lib/graspit/graspitConfig.cmake:90 (include):
  include could not find load file:

    /home/mario/graspitInstall/lib/graspit/../graspit-targets.cmake
Call Stack (most recent call first):
  CMakeLists.txt:63 (find_package)

 ~/graspitInstall/lib/graspit $ ls
FindBULLET.cmake  FindSoQt4.cmake      graspit-targets.cmake
FindQhull.cmake   graspitConfig.cmake  graspit-targets-release.cmake
```

JenniferBuehler, 2016-07-22:
Is graspit-targets.cmake in ``<graspitInstall>/l`` (not in lib dir)?

You have to build *the graspit fork* with ``-DBUILD_SHARED_LIBS``. Are you also sure you are using ``-DCMAKE_PREFIX_PATH`` when building grasp_planning_graspit, and not ``-DCMAKE_INSTALL_PREFIX``

JenniferBuehler, 2016-07-22:
Oh, hang on.. yes the install path is not corrected. I'm testing it now and will update in a moment.

JenniferBuehler, 2016-07-22:
Ok, do a git pull on the graspit fork again... it was indeed still set to the obsolete path for the shared libs.

belledon, 2016-07-22:
Thanks! Will do

belledon, 2016-07-22:
Everything worked. What would be the best way to distribute the changes to the two packages? Are you still interested in the results from the annealing changes? My intentions are to share anything requested. I am working on a project that would require weak annealing such that grasps remain spatially local.

JenniferBuehler, 2016-07-22:
Great!! Sure I am interested, and probably the GraspIt! group is as well, in what you find.

For **graspit**: I'd say that you do pull requests (PRs) to my graspit fork on the changes you do. We can discuss it in the PR conversation first and apply changes according to discussion. Once settled and I have merged your changes, we can do a PR with my fork to the original graspit source (though this may take quite a while, so the main thing is you'll have your own working version meanwile). 
*Note:* Do do PR's, you will need to fork my fork (I know it sounds complicated, but it's not so bad, and it's the best way to do it because you need the stuff from my fork at this point as well). There's plenty of tutorials online on how to create a fork, if you haven't done it yet (you can do it on the [github of my fork](https://github.com/JenniferBuehler/graspit) by clicking on "Fork").  You should do this now - you will then need to clone your own fork, and then re-compile graspit from your fork. Now you know how to do it, so it shouldn't be too hard. Then you should apply your changes on your own fork and send PRs to my fork.

For **grasp_planning_graspit**: Your test files would be modifications in grasp_planning_graspit - that's the changes you make in order to set the annealing parameters. You could also do PR's to graspit-pkgs (again, fork my graspit-pkgs). Or, maybe a bit easier: we could first discuss things as issues in graspit-pkgs to see if anything has to be changed.

**One important note on my graspit fork**
The original graspit code has recently been updated frequently. I want to merge all the changes into my fork ASAP. I hope to get around to it this Sunday. Meanwhile, please keep track of the changes you're making, ideally it only affects one file and then you can copy&paste your changes later into the new version. It will be a big merge, so it's likely you'll have conflicts. To make it as easy as possible for both of us, I would suggest you wait with all this (including forking my repos as described above) until Monday. I will keep you posted on when the merge happens.

belledon, 2016-07-22:
The instructions were helpful. I'm recompiling it now. To be clear, I should fork your graspit fork ( and the rest) on Monday.

JenniferBuehler, 2016-07-24:
Hi,
unfortunately I didn't make it today, another project is keeping me super busy, so it will have to be Tuesday or Wednesday. Meanwhile, can you work on the current version (don't fork yet, just use what you already have)?

If your changes only affect one or a few file, you can then copy this file manually over to the new version coming early this week. This will be easier as merging the complete update into your changes.

Jenny

belledon, 2016-07-25:
Hey Jenny, 
No worries. Yes I am only changing a few files so it should be no problem. 

Mario

JenniferBuehler, 2016-07-26:
I have now update the fork and graspit-pkgs. You are good to make your fork. Let me know if there's any issues.

belledon, 2016-07-26:
graspit is building the static library again even when I use the flag `-DBUILD_SHARED_LIBS=true`

JenniferBuehler, 2016-07-26:
Hups, that's probably because I've had to change the CMakeLists.txt. Silly I thought I checked that. Will do it in the morning! You can just use the static library until then, there's no real difference for your use case.

JenniferBuehler, 2016-07-29:
Hmmmm it's building the shared library for me.

In the fork:
```
cmake -DBUILD_SHARED_LIBS=true -DCMAKE_INSTALL_PREFIX=/home/jenny/graspitInstall ..
make && make install
```

Builds shared lib. Then in grasp_planning_graspit:

```
cmake  -DCMAKE_PREFIX_PATH=/home/jenny/graspitInstall -DBUILD_GRASPIT=false ..
make
```

Works just as it did before, only shared libs build.
Still not working for you?

belledon, 2016-07-29:
Oh I was just being silly and misspelled the command. Its almost done (I accidentally deleted some of my files and had to reimplement them). As of now I have two versions of `grasp_planning` : the original, and a modified version called `om_grasp_planning`. I was wondering if you would want me to add the annealing arguments to the original (as of now its only on the other version) (both compile and are installed separately)?

JenniferBuehler, 2016-07-30:
Ok. I'd suggest you keep your things in a separate executable as it is now, so you have a copy of the original ready to go if you need it. Most changes should anyway be happening not in the main .cpp file, but in related source, so it should not require much work to add the changes to the original later on.

belledon, 2016-07-30:
I agree. Most of my changes have been to `SimAnn`. I did add optional parameters to `EigenGraspPlanner::plan` so I could pass the annealing parameters when calling plan  without effecting the original `grasp_planning` call. However, I was wondering if in your opinion it would be better for me to add a new method, `EigenGraspPlanner::configPlanner(std::vector<double> params)` that passes the annealing parameters to the `graspitEgPlanner` object which would then use the functions within `SimAnn` (ie `SimAnn::useParameters(AnnealingType t, ...)`). 

Using the current structure:
`EigenGraspPlanner::plan -> EigenGraspPlanner::initPlanner -> EigenGraspPlanner::initPlannerType -> SimAnnPlanner::useAnnealingParameters -> SimAnn::useParameters`

The alternate would be:
`EigenGraspPlanner::configPlanning -> SimAnnPlanner::useAnnealingParameters -> SimAnn::useParameters`

But I would have to add a snippet where `EigenGraspPlanner::plan` checks to make sure that `graspitEgPlanner` was modified so that it doesn't call `EigenGraspPlanner::initPlanner`...and so on.

JenniferBuehler, 2016-07-31:
Hi,
I like the idea about the configPlanner method. But I would not initialize with a vector, rather with a map (where they keys are the names of the paramters).  I.e. it would be ``EigenGraspPlanner::configPlanner(std::map<std::string, float>& params)``. Each parameter would have a unique string name, similar to the ROS parameter server. The thing with this approach would be that the float value would need to be used to cast int values, which is not pretty.

As an alternative, it could be a special struct which enumerates the annealing parameters explicitly. This could be a subclass ``AnnealingParamters`` derived from ``PlanningParamters``, so that it can be kept flexible. I.e. it would be ``EigenGraspPlanner::configPlanner(PlanningParamters& params)``. When the ``PlanningParamters`` are then used to initialize ``SimAnn``, it would dynamic_cast the ``params`` to make sure it's an instance of ``AnnealingParameters`` in order to access the right values.

Does this make sense?

I won't have much time this week to keep an eye on this, I'm very busy with a new project and can now only work on this on the weekends. I hope you're good until next weekend with this?

Looking forward to see how this works :)
Jenny

belledon, 2016-08-01:
Hi Jenny, 

It all makes sense, except for the casting to ints. Where does this happen in the `SimAnn`? I thought the annealing parameters where floats?

belledon, 2016-08-01:
O sorry they're doubles. I'll give the map a try. Hope your projects go well!
-Mario

JenniferBuehler, 2016-08-01:
Hi Mario,

I admit I haven't checked the exact parameters in SimAnn yesterday ;) But to keep things general, in any case we have to consider that some parameters may be ints or enums (maybe for other planner types to be added later). That's what I meant with this, so I didn't check as I thought the conclusion is the same.

Good luck & let me know how it goes
Jenny

belledon, 2016-08-01:
Hi Jenny, 

So I got it to work using the quick and dirty way of passing a vector just so I can have numbers to crunch. I plan to add (probably next week) the classes that you suggested:

`AnnealingParams` having one element `PlannerType` (as from `search.h`) with some virtual methods and an inherited class `SimAnnParams` which contains the particulars. So in the end, `AnnealingParams params = new SimAnnParams(type, p)` and after `EGPlanner::configPlanner(AnnealingParams params)`

JenniferBuehler, 2016-08-04:
Hi Mario,

sounds good! I'd say keep it as is for now, I'll have a closer look at the annealing parameters on the weekend and let you know my thoughts. I think keeping it flexible in-between planners as well would be good (not only annealing is available), but I have to take a better look at SimAnn and the other classes before, and during the week I won't get around to do it.

Good work! Let me know how it works out for the planning results now that you can change the parameters. 

Jenny

belledon, 2016-08-04:
Hi Jenny, 

Sorry I forgot to mention earlier. I did implement an abstract class `PlanningParams`. I then added a virtual function to `EGPlanner::configPlanner(PlanningParams * params)` which then was implemented in `SimAnnPlanner::configPlanner(SimAnnParams * params)` where `SimAnnParams` is the derived class from `PlanningParams` specifically for `SimAnnPlanner`. `PlanningParams` has a method `PlanningParams::getPlanningType` which returns a `PlanningType` as defined in `search.h` which allows for the derived planning class (ie `SimAnnPlanner`) to further check for the param type. `SimAnnParams::getParams` returns a `std::map<string, double>` which is called upon in `SimAnnPlanner::configPlanner`. So the idea would be that each derived planner would just need to implement `<derived planner>::configPlanner(<derived params> * params)`  which would mean that they would also need to derive their own params class. However, the params class tend to be bare bones (only really need to define some  like `<some return type> <derived planner>::getParams()`

belledon, 2016-08-04:
Also, I was wondering how difficult you would think it would be for me to be able to implement rotations when loading object (similar to `GraspitSceneManager::loadRobot`) where I can give it a quaternion (or anything) that would allow me to change the rotation of an object prior to planning. To be honest, I don't think rotation will have any effect but I would like to be thorough.

JenniferBuehler, 2016-08-05:
Hi Mario,

the rotations should already be supported as parameters for grasp_planning, if you type only ./grasp_planning then it should print you the help.

Can you send me the link to your github fork? It's easier for me if I can just look at the code :)

Cheers
Jenny

belledon, 2016-08-05:
Here's the link! [https://github.com/belledon/graspit](https://github.com/belledon/graspit)

belledon, 2016-08-05:
I don't see any parameters for rotation. Only for x,y,z position.

JenniferBuehler, 2016-08-06:
Sorry I'm under way at the moment and can't look it up, I think there is an orientation as well, maybe I didn't add it as command line argument. Check in grasp_planning.cpp?

Cheers
Jenny

JenniferBuehler, 2016-08-06:
Ah and thanks for the link!

belledon, 2016-08-09:
Hi Jenny, 

I just realized that I botched up allot in what I did to the fork of graspit. I am attempting to fix the issues with polymorphism in `SimAnnPlanner`. I'll let you know when its up and running

[8 more replies]

closed 2016-09-07 by JenniferBuehler