JenniferBuehler/graspit-pkgs #14: Catkin error when building graspit_external

JSR694, 2016-06-28:
Hi there!  I run into the following errors when I try to build graspit-pkgs with catkin_make:

```
[18%] Building CXX object graspit-pkgs/grasp_planning_graspit/graspit_external/graspit-build/CMakeFiles/graspit_simulator.dir/include/robots/moc_barrett.cxx.o
/home/jack/ros_stuff/my_ws/build/graspit-pkgs/grasp_planning_graspit/graspit_external/graspit-build/qt_image_collection.cpp: In function ‘QImage uic_findImage(const QString&)’:
/home/jack/ros_stuff/my_ws/build/graspit-pkgs/grasp_planning_graspit/graspit_external/graspit-build/qt_image_collection.cpp:13205:51: error: ‘const struct EmbedImage’ has no member named ‘colorTab22’
                         (QRgb*)embed_image_vec[i].colorTab22, 22, 32, (const unsigned char*)image_21_data, 123, 0, 0, true, "editcopy" },
                                                   ^
/home/jack/ros_stuff/my_ws/build/graspit-pkgs/grasp_planning_graspit/graspit_external/graspit-build/qt_image_collection.cpp:13205:136: error: expected ‘)’ before ‘}’ token
                         (QRgb*)embed_image_vec[i].colorTab22, 22, 32, (const unsigned char*)image_21_data, 123, 0, 0, true, "editcopy" },
                                                                                                                                        ^
/home/jack/ros_stuff/my_ws/build/graspit-pkgs/grasp_planning_graspit/graspit_external/graspit-build/qt_image_collection.cpp:13205:136: error: expected ‘,’ or ‘;’ before ‘}’ token
/home/jack/ros_stuff/my_ws/build/graspit-pkgs/grasp_planning_graspit/graspit_external/graspit-build/qt_image_collection.cpp:13205:137: error: expected primary-expression before ‘,’ token
                         (QRgb*)embed_image_vec[i].colorTab22, 22, 32, (const unsigned char*)image_21_data, 123, 0, 0, true, "editcopy" },
                                                                                                                                         ^
/home/jack/ros_stuff/my_ws/build/graspit-pkgs/grasp_planning_graspit/graspit_external/graspit-build/qt_image_collection.cpp:13206:5: error: expected primary-expression before ‘{’ token
     { 22, 22, 32, (const unsigned char*)image_22_data, 149, 0, 0, true, "editpaste" },
     ^
/home/jack/ros_stuff/my_ws/build/graspit-pkgs/grasp_planning_graspit/graspit_external/graspit-build/qt_image_collection.cpp:13206:5: error: expected ‘;’ before ‘{’ token
/home/jack/ros_stuff/my_ws/build/graspit-pkgs/grasp_planning_graspit/graspit_external/graspit-build/qt_image_collection.cpp:13206:86: error: expected primary-expression before ‘,’ token
     { 22, 22, 32, (const unsigned char*)image_22_data, 149, 0, 0, true, "editpaste" },
                                                                                      ^
/home/jack/ros_stuff/my_ws/build/graspit-pkgs/grasp_planning_graspit/graspit_external/graspit-build/qt_image_collection.cpp:13207:5: error: expected primary-expression before ‘{’ token
     { 22, 22, 32, (const unsigned char*)image_23_data, 480, 0, 0, true, "searchfind" },
     ^
/home/jack/ros_stuff/my_ws/build/graspit-pkgs/grasp_planning_graspit/graspit_external/graspit-build/qt_image_collection.cpp:13207:5: error: expected ‘;’ before ‘{’ token
/home/jack/ros_stuff/my_ws/build/graspit-pkgs/grasp_planning_graspit/graspit_external/graspit-build/qt_image_collection.cpp:13207:87: error: expected primary-expression before ‘,’ token
     { 22, 22, 32, (const unsigned char*)image_23_data, 480, 0, 0, true, "searchfind" },
                                                                                       ^
/home/jack/ros_stuff/my_ws/build/graspit-pkgs/grasp_planning_graspit/graspit_external/graspit-build/qt_image_collection.cpp:13208:5: error: expected primary-expression before ‘{’ token
     { 0, 0, 0, 0, 0, 0, 0, 0, 0 }
     ^
/home/jack/ros_stuff/my_ws/build/graspit-pkgs/grasp_planning_graspit/graspit_external/graspit-build/qt_image_collection.cpp:13208:5: error: expected ‘;’ before ‘{’ token
/home/jack/ros_stuff/my_ws/build/graspit-pkgs/grasp_planning_graspit/graspit_external/graspit-build/qt_image_collection.cpp:13212:1: error: a function-definition is not allowed here before ‘{’ token
 {
 ^
/home/jack/ros_stuff/my_ws/build/graspit-pkgs/grasp_planning_graspit/graspit_external/graspit-build/qt_image_collection.cpp:13280:45: error: expected ‘}’ at end of input
 static StaticInitImages_graspit staticImages;
                                             ^
make[2]: *** [graspit-pkgs/grasp_planning_graspit/graspit_external/graspit-build/CMakeFiles/graspit.dir/qt_image_collection.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[ 18%] /home/jack/ros_stuff/my_ws/build/graspit-pkgs/grasp_planning_graspit/graspit_external/graspit-build/qt_image_collection.cpp: In function ‘QImage uic_findImage(const QString&)’:
/home/jack/ros_stuff/my_ws/build/graspit-pkgs/grasp_planning_graspit/graspit_external/graspit-build/qt_image_collection.cpp:13205:51: error: ‘const struct EmbedImage’ has no member named ‘colorTab22’
                         (QRgb*)embed_image_vec[i].colorTab22, 22, 32, (const unsigned char*)image_21_data, 123, 0, 0, true, "editcopy" },
                                                   ^
/home/jack/ros_stuff/my_ws/build/graspit-pkgs/grasp_planning_graspit/graspit_external/graspit-build/qt_image_collection.cpp:13205:136: error: expected ‘)’ before ‘}’ token
                         (QRgb*)embed_image_vec[i].colorTab22, 22, 32, (const unsigned char*)image_21_data, 123, 0, 0, true, "editcopy" },
                                                                                                                                        ^
/home/jack/ros_stuff/my_ws/build/graspit-pkgs/grasp_planning_graspit/graspit_external/graspit-build/qt_image_collection.cpp:13205:136: error: expected ‘,’ or ‘;’ before ‘}’ token
/home/jack/ros_stuff/my_ws/build/graspit-pkgs/grasp_planning_graspit/graspit_external/graspit-build/qt_image_collection.cpp:13205:137: error: expected primary-expression before ‘,’ token
                         (QRgb*)embed_image_vec[i].colorTab22, 22, 32, (const unsigned char*)image_21_data, 123, 0, 0, true, "editcopy" },
                                                                                                                                         ^
/home/jack/ros_stuff/my_ws/build/graspit-pkgs/grasp_planning_graspit/graspit_external/graspit-build/qt_image_collection.cpp:13206:5: error: expected primary-expression before ‘{’ token
     { 22, 22, 32, (const unsigned char*)image_22_data, 149, 0, 0, true, "editpaste" },
     ^
/home/jack/ros_stuff/my_ws/build/graspit-pkgs/grasp_planning_graspit/graspit_external/graspit-build/qt_image_collection.cpp:13206:5: error: expected ‘;’ before ‘{’ token
/home/jack/ros_stuff/my_ws/build/graspit-pkgs/grasp_planning_graspit/graspit_external/graspit-build/qt_image_collection.cpp:13206:86: error: expected primary-expression before ‘,’ token
     { 22, 22, 32, (const unsigned char*)image_22_data, 149, 0, 0, true, "editpaste" },
                                                                                      ^
/home/jack/ros_stuff/my_ws/build/graspit-pkgs/grasp_planning_graspit/graspit_external/graspit-build/qt_image_collection.cpp:13207:5: error: expected primary-expression before ‘{’ token
     { 22, 22, 32, (const unsigned char*)image_23_data, 480, 0, 0, true, "searchfind" },
     ^
/home/jack/ros_stuff/my_ws/build/graspit-pkgs/grasp_planning_graspit/graspit_external/graspit-build/qt_image_collection.cpp:13207:5: error: expected ‘;’ before ‘{’ token
/home/jack/ros_stuff/my_ws/build/graspit-pkgs/grasp_planning_graspit/graspit_external/graspit-build/qt_image_collection.cpp:13207:87: error: expected primary-expression before ‘,’ token
     { 22, 22, 32, (const unsigned char*)image_23_data, 480, 0, 0, true, "searchfind" },
                                                                                       ^
/home/jack/ros_stuff/my_ws/build/graspit-pkgs/grasp_planning_graspit/graspit_external/graspit-build/qt_image_collection.cpp:13208:5: error: expected primary-expression before ‘{’ token
     { 0, 0, 0, 0, 0, 0, 0, 0, 0 }
     ^
/home/jack/ros_stuff/my_ws/build/graspit-pkgs/grasp_planning_graspit/graspit_external/graspit-build/qt_image_collection.cpp:13208:5: error: expected ‘;’ before ‘{’ token
/home/jack/ros_stuff/my_ws/build/graspit-pkgs/grasp_planning_graspit/graspit_external/graspit-build/qt_image_collection.cpp:13212:1: error: a function-definition is not allowed here before ‘{’ token
 {
 ^
/home/jack/ros_stuff/my_ws/build/graspit-pkgs/grasp_planning_graspit/graspit_external/graspit-build/qt_image_collection.cpp:13280:45: error: expected ‘}’ at end of input
 static StaticInitImages_graspit staticImages;
                                             ^
```

Thanks!
Jack

JenniferBuehler, 2016-06-28:
Huh weird..  this is an error with building the graspit code.
Which version of Qt do you have, and which OS (Ubuntu 14.04?)

Cheers
Jenny

JSR694, 2016-06-28:
Yep, 14.04.  And, I've got Qt4 (as well as its Inventor toolkit), as per the wiki.

JenniferBuehler, 2016-06-28:
And when you build grasp_planning_graspit without ROS (cmake only) then it compiles?

JSR694, 2016-06-28:
I just tried that, and it works!  So it's a problem with catkin, then?  At any rate, now that I've got a build of graspit I don't need to build it via catkin.

Thanks again for all your help!
-Jack

JenniferBuehler, 2016-06-28:
Very odd. Thanks for reporting this, I didn't have this on any of my machines.

If you don't mind, could you help and re-compile after I looked into this and change suspected issues? Not sure I'll be able to reproduce it.

Which ROS dist are you using?

JSR694, 2016-06-28:
Will do--glad to help!  I am using Indigo.

JSR694, 2016-06-29:
Jenny,

I'm running into problems building a ROS package which has this (catkin-less) build of Graspit as a dependency.  Here's my CMakeLists.txt for the package in question.  It's a modification of graspit_dbase_tasks, which I ported from rosbuild using the catkinize Python tool:

```
cmake_minimum_required(VERSION 2.8.3)
project(graspit_dbase_tasks_localhost)

set(graspit_DIR /home/jack/ros_stuff/my_ws/build/graspit-pkgs/grasp_planning_graspit/graspit_external/repo/graspit/cmake/)

# Load catkin and all dependencies required for this package
find_package(catkin REQUIRED COMPONENTS )
find_package(Qt4 COMPONENTS QtCore QtGui QtOpenGL REQUIRED)
include(${QT_USE_FILE})
find_package(graspit REQUIRED)

#set the default path for built executables and lib.s to the "bin"+"lib" dir.s
set(EXECUTABLE_OUTPUT_PATH ${PROJECT_SOURCE_DIR}/bin)
set(LIBRARY_OUTPUT_PATH ${PROJECT_SOURCE_DIR}/lib)

include_directories(${Boost_INCLUDE_DIRS})
target_link_libraries(${PROJECT_NAME} ${Boost_LIBRARIES})

add_definitions(-DQT_NO_KEYWORDS)

add_library(graspit_dbase_tasks_localhost src/dbTaskDispatcher.cpp)
generate_messages(
    #TODO DEPENDENCIES geometry_msgs std_msgs
)

# catkin_package parameters: http://ros.org/doc/groovy/api/catkin/html/dev_guide/generated_cmake_api.html#catkin-package
# TODO: fill in what other packages will need to use this package
catkin_package(
    DEPENDS graspit Qt
    CATKIN_DEPENDS # TODO
    INCLUDE_DIRS # TODO include
    LIBRARIES # TODO
)
```
I added "set(graspit_DIR ..." to address the first error I encountered, but that only led to further errors:

```
-- Looking for graspit...
CMake Error at /home/jack/ros_stuff/my_ws/build/graspit-pkgs/grasp_planning_graspit/graspit_external/repo/graspit/cmake/graspitConfig.cmake:58 (include):
  include could not find load file:
    /home/jack/ros_stuff/my_ws/devel/lib/graspit/graspit-targets.cmake
Call Stack (most recent call first):
  graspit_dbase_tasks_localhost/CMakeLists.txt:13 (find_package)

SENDL_ERRORLooking for graspit library -- not foundPlease install graspit or adjust CMAKE_PREFIX_PATHe.g. cmake -DCMAKE_PREFIX_PATH=/path-to-graspit/ ...
CMake Error at /home/jack/ros_stuff/my_ws/build/graspit-pkgs/grasp_planning_graspit/graspit_external/repo/graspit/cmake/graspitConfig.cmake:99 (find_package):
  By not providing "FindQhull.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Qhull", but
  CMake did not find one.

  Could not find a package configuration file provided by "Qhull" with any of
  the following names:
    QhullConfig.cmake
    qhull-config.cmake

  Add the installation prefix of "Qhull" to CMAKE_PREFIX_PATH or set
  "Qhull_DIR" to a directory containing one of the above files.  If "Qhull"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  graspit_dbase_tasks_localhost/CMakeLists.txt:13 (find_package)
```
 (Note that I abridged CMakeLists above, so the line numbers here probably don't correspond.)

I've spent quite a while debugging but have made no headway.  Any ideas as to what's wrong?

Best
Jack

JenniferBuehler, 2016-06-29:
Hi Jack,

if you are using the catkin-less graspit install, why are you setting ``graspit_DIR`` to the catkin directory, is there a particular reason for it? (graspit_DIR is an obsolete variable anyway, where did you read about it, is there a left-over in the wiki?)

I would compile and install graspit separately with cmake only, not catkin (from [my fork](https://github.com/JenniferBuehler/graspit), because main repo doesn't have graspitConfig.cmake yet). Compile and install to ``<your-graspit-dir>``

```
cd <graspit-fork>
mkdir build
cd build
cmake .. -DCMAKE_INSTALL_PREFIX=<your-graspit-dir>
make
make install
```

Try a cmakelists based on this template:

```cmake
cmake_minimum_required(VERSION 2.8.3)
project(your_project)
find_package (graspit REQUIRED)
include_directories(
    ${graspit_INCLUDE_DIRS}
)
add_executable(your_program src/your_source.cpp)
target_link_libraries(your_program
   ${graspit_LIBRARIES}
)
```

You don't need to include QT_USE_FILE and all that, it should be handled by the graspitConfig.cmake.

Then build your own package with:

```
cmake -DCMAKE_PREFIX_PATH=<your-graspit-dir>
make
```

Let me know if it works!
I'll only be back home tomorrow so won't be able to try the first issue (posted above) on my Indigo machine before then.

JenniferBuehler, 2016-06-29:
Ah hang on I just saw, you are compiling with catkin. I can't try it now, but does this work (for the last commands posted above):

```
catkin_make -DCMAKE_PREFIX_PATH=<your-graspit-dir>
```

JSR694, 2016-06-29:
Nope, that makes it throw an error as soon as it tries to generate messages:

```
CMake Warning at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by "image_transport"
  with any of the following names:

    image_transportConfig.cmake
    image_transport-config.cmake

  Add the installation prefix of "image_transport" to CMAKE_PREFIX_PATH or
  set "image_transport_DIR" to a directory containing one of the above files.
  If "image_transport" provides a separate development package or SDK, be
  sure it has been installed.
Call Stack (most recent call first):
  apriltags_ros/apriltags_ros/CMakeLists.txt:7 (find_package)

-- Could not find the required component 'image_transport'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "image_transport"
  with any of the following names:

    image_transportConfig.cmake
    image_transport-config.cmake

  Add the installation prefix of "image_transport" to CMAKE_PREFIX_PATH or
  set "image_transport_DIR" to a directory containing one of the above files.
  If "image_transport" provides a separate development package or SDK, be
  sure it has been installed.
Call Stack (most recent call first):
  apriltags_ros/apriltags_ros/CMakeLists.txt:7 (find_package)
```

I'm guessing that -DCMAKE_PREFIX_PATH=<graspit-dir> gets rid of the entries for /opt/ros/indigo and others.  I also tried _appending_ the graspit directory to CMAKE_PREFIX_PATH, but that doesn't eliminate my original error.

JenniferBuehler, 2016-06-29:
> I'm guessing that -DCMAKE_PREFIX_PATH= gets rid of the entries for

Hm yeah that's my guess too, image_transport is not needed by graspit. I'm
sorry you're having so much trouble, you end up finding odd new issues, but
I guess it's a good thing because then we can fix them.

There must be a way to set additional install dirs for catkin, but I won't
be able to look it up before tomorrow. Will try a quick google search on my
phone though and let you know if I find something.

JSR694, 2016-06-29:
I seem to have fixed my original problem!  For some reason the source file "qt_image_collection.cpp" was malformed-- the last ~20 lines of code (line 13283 onwards) in the file are  duplicated.  I commented out the offending lines and catkin_make completed without issue.

No idea why this bug arose; qt_image_collection seems to be Qt-specific code, and I'm unfamiliar with Qt.  It's created by a custom command in graspit's CMakeLists.txt, if that's any help.

Dinner time for now but I'll let you know later if I run into any more problems.  As you observe, I tend to unearth new and interesting bugs. :)

JenniferBuehler, 2016-06-30:
Ah... yeah that qt_image_collection.cpp has given me headache before. I did this part of the graspit CMakeLists.txt, it took forever until the symbols finally worked using cmake instead of Qt.
Because qt_image_collection is generated by a Qt command (triggered from cmake), maybe something went wrong with the generation of it the first time. Having things double should definitely not happen. If you re-build completely (``cd catkin_ws && rm -rf devel build && catkin_make``), does it happen again?

Anyway, glad the problem is resolved. I will try again in any case when I get home and see if the duplicated lines issue happens for me as well now. If it doesn't happen on my Indigo, I will close this thread considering it as solved.

JSR694, 2016-06-30:
> If you re-build completely (cd catkin_ws && rm -rf devel build && catkin_make), does it happen again?

Yep.  Although the duplication happens at a slightly different place. Here it seems that line 13267:: is duplicated.  Again, the problem is solved by removing the duplicated lines.

JSR694, 2016-06-30:
Jenny,

Now that GraspIt! builds, I'm trying to build my package which depends on it.  I mentioned this earlier-- it's a modification of graspit_dbase_tasks, which I ported from rosbuild using the catkinize Python tool.  To get cmake to find graspit, I set graspit_DIR to be the directory which contains graspitConfig.cmake:

`set(graspit_DIR /home/jack/ros_stuff/my_ws/build/graspit-pkgs/grasp_planning_graspit/graspit_external/repo/graspit/cmake)`

However, cmake errors out when trying to run graspitConfig.cmake.  Here are the errors I'm getting:

```
CMake Error at /home/jack/ros_stuff/my_ws/build/graspit-pkgs/grasp_planning_graspit/graspit_external/repo/graspit/cmake/graspitConfig.cmake:59 (include):
  include could not find load file:

    /home/jack/ros_stuff/my_ws/devel/lib/graspit/graspit-targets.cmake
Call Stack (most recent call first):
  graspit_dbase_tasks_localhost/CMakeLists.txt:15 (find_package)

SENDL_ERRORLooking for graspit library -- not foundPlease install graspit or adjust CMAKE_PREFIX_PATHe.g. cmake -DCMAKE_PREFIX_PATH=/path-to-graspit/ ...
CMake Error at /home/jack/ros_stuff/my_ws/build/graspit-pkgs/grasp_planning_graspit/graspit_external/repo/graspit/cmake/graspitConfig.cmake:100 (find_package):
  By not providing "FindQhull.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Qhull", but
  CMake did not find one.

  Could not find a package configuration file provided by "Qhull" with any of
  the following names:

    QhullConfig.cmake
    qhull-config.cmake

  Add the installation prefix of "Qhull" to CMAKE_PREFIX_PATH or set
  "Qhull_DIR" to a directory containing one of the above files.  If "Qhull"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  graspit_dbase_tasks_localhost/CMakeLists.txt:15 (find_package)
```

From what I've found, it looks like graspitConfig.make assumes a very different directory structure than what is given.  For instance, this snippet from line 56:
```
else (NOT CATKIN_DEVEL_PREFIX)
    # catkin is going to handle the include directories
    include(${CATKIN_DEVEL_PREFIX}/lib/graspit/graspit-targets.cmake)
```
There is no _graspit_ directory in .../my_ws/devel/lib; in fact, I cannot find any file called "graspit-targets.cmake".  The closest thing I can find is "graspit-static-targets.cmake", which is in the following directory:
```
.../my_ws/build/graspit-pkgs/grasp_planning_graspit/graspit_external/graspit-build/CMakeFiles/Export/_home/jack/ros_stuff/my_ws/build/graspit-pkgs/grasp_planning_graspit/graspit_external/install/lib/graspit-static-targets.cmake
```
The other error, about finding "FindQhull.cmake", is also odd, as catkin has no problems finding the package when building GraspIt! on its own.

(If you think that this is a separate issue from my original one, please feel free to move it :) )

Thanks for your help!
Jack

JenniferBuehler, 2016-06-30:
You don't need to include like this if you've already compiled it in your catkin workspace. Remove all your graspit-related stuff from the cmake file (including setting graspit_DIR), and then just include graspit with

```cmake
find_package(catkin REQUIRED COMPONENTS graspit <.. your other packages..>)
```
graspit is now included in all variables, e.g. the include dirs in ``catkin_INCLUDE_DIRS``.
You should also add graspit as build and run dependency in your package.xml.

catkin generates its own graspitConfig so you should not include the other one.

JSR694, 2016-06-30:
Hmm, doing that that results in the following error:
```
CMake Warning at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by "graspit" with any
  of the following names:

    graspitConfig.cmake
    graspit-config.cmake

  Add the installation prefix of "graspit" to CMAKE_PREFIX_PATH or set
  "graspit_DIR" to a directory containing one of the above files.  If
  "graspit" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  graspit_dbase_tasks_localhost/CMakeLists.txt:8 (find_package)

-- Could not find the required component 'graspit'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "graspit" with any
  of the following names:

    graspitConfig.cmake
    graspit-config.cmake

  Add the installation prefix of "graspit" to CMAKE_PREFIX_PATH or set
  "graspit_DIR" to a directory containing one of the above files.  If
  "graspit" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  graspit_dbase_tasks_localhost/CMakeLists.txt:8 (find_package)

-- Configuring incomplete, errors occurred!
```

This error is the reason why I manually specified graspit_DIR in the first place.

JenniferBuehler, 2016-06-30:
Confirming that you have the directory graspit in ``your_catkin_ws/src``, and it is listed as package to be built at the beginning of the ``catkin_make`` command. You have no other graspit installation flying about somewhere?

JenniferBuehler, 2016-06-30:
And, you have used graspit from my fork, not from graspit_simulator, because that CMakeLists.txt does not (yet) support catkin.

JSR694, 2016-06-30:
I was about to comment on this!  I think it's the issue.  I've got the old graspit_simulator stack installed.  I'll remove it, and replace it with your fork, and see if that solves the issue.

JenniferBuehler, 2016-06-30:
AAAAh hang on... That happens when replying from under way and not thinking properly. I forgot you are building graspit from graspit-pkgs!
In this case you won't be able to include graspit directly, because it builds it into grasp_planning_graspit (to avoid clashes with original versions, not the one in my fork). The solution in your case is to build graspit separately:

1. Clone [graspit](https://github.com/JenniferBuehler/graspit) into catkin workspace
2. You already have cloned graspit-pkgs in your catkin_ws
3. Remove devel and build folders
4. ``catkin_make -DBUILD_GRASPIT=false``

Sorry I didn't think of this earlier.

I should probably ask: What does your ``graspit_dbase_tasks_local`` need: do you need original headers from graspit, or do you need anything from grasp_planning_graspit? If you don't need the latter, you can only use graspit from my fork (point 1. above), and remove graspit-pkgs from your catkin workspace altogether

JenniferBuehler, 2016-06-30:
And yes, remove the old graspit_simulator if you don't need it. While it shouldn't have name clashes, it's better to test on a clean state first.

JSR694, 2016-07-01:
Jenny,

You were right-- I've successfully built your fork of graspit.  I've run into other issues now, but they are not relevant to this one.  I'd consider it solved.

Thanks, once again, for your help!

Best,
Jack

JenniferBuehler, 2016-07-01:
Hi Jack,

glad this solved it! I'm suspecting the issue with qt_image_collection was related to this, did it happen also when building graspit separately?

If you have other problems, feel free to open a new issue related to the subject.

JSR694, 2016-07-01:
Jenny,

Yep, qt_image_collection was no longer a problem.

My current issue is more related to the dbase_grasp_planner plugin from graspit_simulator... I'm having an odd compile-time error, which may be due to an improper conversion from rosbuild to catkin.  [Here's the link](https://github.com/ros-interactive-manipulation/graspit_simulator/issues/2) to the issue.

Best
Jack

JenniferBuehler, 2016-07-01:
Ok, great, then the original issue has resolved.

We'll talk about the other issue in the other thread then.

Thanks, Jenny

closed 2016-07-01 by JenniferBuehler