JenniferBuehler/graspit-pkgs #15: GraspitSceneManagerHeadless requiring x server

belledon, 2016-09-08:
Hi Jenny,

Posting this more as notification for a quick work around trying to run grasp_planning via ssh. I don;'t have the error saved but it was something along the lines of `Graspit: cannot establish connection to x server 0`. Because I was runnning it via ssh, I could not user the normal fixes involving xhost. Instead, using a virtual x server. via xvfb-run did the trick. 

From a rough trace, it seems that the problem actually originates within the graspit library rather than graspit-pkgs. I have a feeling it may be a simple check to see if any x server is reachable, throwing that error if its not. Might be a simple fix, but since the work around is also very easy, might not be worth the time. (I'd be happy to post a full trace/ error message if you'd like)

-Mario

JenniferBuehler, 2016-09-08:
Hi Mario,

have you tried to do the ssh connection with the -Y option? This does the X11 forwarding.

Thanks for letting me know about this.

Jenny

belledon, 2016-09-09:
I'm sorry, I wasn't very clear. The ssh isn't the problem. Its that I'm running it from a Singularity container in the system I'm sshing into. Singularity containers are very permissive "safe" meaning that they do not allow escalation of privaleges the user does not have when entering the container (I'm not sudo on the ssh). As of now, x11 is under development for Singularity (at least from what I tried to search). 

That being said. It works perfectly with xvfb, which I just had to install within the container. Feel free to close this issue when ever you'd like.

JenniferBuehler, 2016-09-09:
Ah ok, that makes sense :) Glad you got it to work with xvfb! Thanks for letting me know.

closed 2016-09-09 by JenniferBuehler