JenniferBuehler/graspit-pkgs #17: Misaligned axis when converting from urdf to graspit

hashb, 2017-01-18:
I have been trying to convert allegro hand to graspit!. Everything looks good except the thumb link. The axes on the thumb link are a bit off. 

[urdf2graspit.txt](https://github.com/JenniferBuehler/graspit-pkgs/files/712692/urdf2graspit.txt) is the output of the conversion. I just modified the example to work for my urdf.

Here is an image of the hand in graspit environment
[image]

JenniferBuehler, 2017-01-18:
Hi, thanks for reporting this!

There are a few tricky bits in the conversion process, it's always good to have new hands to test on. I will look at this as soon as I can. It would help if you can share the URDF (even if it's only with dummy primitives in place of the meshes), otherwise it will be difficult for me to see what the possible cause of the problem may be.

hashb, 2017-01-18:
I forgot to include link to the urdf file in the first post. Here is the repository to the meshes and urdf file.

https://github.com/felixduvallet/allegro-hand-ros/tree/master/allegro_hand_description

JenniferBuehler, 2017-01-18:
Great, thanks! I will look at this as soon as I can.

hashb, 2017-01-25:
I solved the problem. 

The issue was with the robot model. The thumb link was a bit complex and we cannot find the DH-Parameters directly. I had to add an extra frame between _link_12.0_ and _link_13.0_ to get the DH-Parameters working.

JenniferBuehler, 2017-01-25:
Cool! Sorry I didn't get to check on this earlier, I had it on my TODO for this week. Glad you solved the problem! Though I'd still be interested to know what the problem was for finding the DH parameters, it should work if the URDF is displaying it correctly (e.g. in RViz). So I'll check it out anyway. I guess I can just use an old commit from your repository to access the initial version of the URDF?

Cheers
Jenny

hashb, 2017-01-25:
Hey Jenny,

I haven't pushed the modified URDF yet. I plan to do that sometime soon. Original URDF files are available [here](https://github.com/felixduvallet/allegro-hand-ros/blob/master/allegro_hand_description/allegro_hand_description_right.xacro).

The problem is with the DH convention. It can only account of 4 Degrees of Freedom. The thumb link in the Allegro hand that I am using has translations in all 3 coordinate frames and 2 rotations between link_12.0 and link_13.0. This broke the converter. 

Thanks for documenting your code so well. I am using the grasp pipeline and its made my life so much easier. 

Cheers
Kautilya

JenniferBuehler, 2017-01-26:
Hi Kautilya,

ah ok, that makes sense if there's several degrees of freedom in rotation. This should be caught in the converter though and it should print an error, or maybe it can be fixed with an automatically inserted link. I'll keep this issue open to remind me of what needs to be done soon, so others don't have to debug this case as you had to.

Just a quick question to make sure I understood the problem right. I'm unsure which joint you are referring to... by looking at the URDF in your link, the connections are as follows (built by macro thumb_right):

``palm_link --> joint_12 (revolute) --> link_12 --> joint_13 (revolute) --> link_13 --> joint_14 (revolute)...``

The joints are all revolute joints around one axis. Or maybe I'm looking at the wrong file? Which joint did you mean has the two rotation degrees, do you maybe have another version with a planar joint or am I missing something?

Or are you referring to ``finger_angle_p`` and ``finger_angle_r``, because that's the fixed transform from one link to the other (not the DOF of the joint, maybe I misunderstood you then). The converter should be able to deal with this... if you are referring to this then it's still an issue with the converter I'll need to fix.

Thanks for your help to narrow this down :)
Jenny

hashb, 2017-01-27:
Hey Jenny,

That is the correct file. I was talking about the transformation from `joint_12 --> joint_13`. I just read my earlier comment and I did a terrible job explaining the problem. When I tried to manually compute DH Parameters for the thumb, I couldn't find a set of DH parameters that could do the transformation between the two frames. I might have been wrong. For now I am using the extra joint hack to make this work. It'd be awesome if this can be fixed. 

Ill try to update the comment with more information.

Cheers
Kautilya

JenniferBuehler, 2017-01-30:
Hi Kautilya,

Hmmm maybe you're right, I'll double check the maths as well. The axis of the joint may have to be changed, or it would maybe be possible to insert the in-between "fake" links in such cases automatically.

I'm glad you have found a workaround for now, I'll post an update here in case I find (or don't) an automatic workaround.

Cheers
Jenny

JenniferBuehler, 2017-03-12:
Hi Kautilya,

not sure you have follwed the other issue #19 (scroll to bottom), but the allegro hand is now working with the [original repository by simlabrobotics](https://github.com/simlabrobotics/allegro_hand_ros). It was possible to compute the DH parameters, there were actually two bugs hidden in urdf2graspit and urdf2inventor which the allegro hand helped to uncover.
There is still an issue with one of the thumb link visuals in the fork of the repository which you are using as well. This issue does not appear when using the original repository (see discussion in the other issue).

Just wanted to let you know. If you're still working on this, you can try it out, and if your problem is resolved, you may close this issue.

Cheers
Jenny

JenniferBuehler, 2017-03-17:
Closed as referencing to conclusions discussed in #19

closed 2017-03-17 by JenniferBuehler

hashb, 2017-03-18:
Hey Jenny,
This is awesome. I guess the way I was calculating DH parameters was wrong. I moved away from graspit because I was not able to find good grasps. I will check it out again, It might be that I did something wrong.

Thanks
Kautilya

JenniferBuehler, 2017-03-18:
Yeah the DH parameter calculation is a bit of a head wrapper ;)
If you try again, make sure you do generate contacts (one or two per finger should already do). You can use the visual tool coming with urdf2graspit. Without contacts, the search space is too large for GraspIt and it doesn't really come up with any reasonable solution.
Good luck!
Jenny