JenniferBuehler/graspit-pkgs #18: catkin error while compiling

ilaplace, 2017-02-01:
Hello there!

I'm using Ubuntu 14.04 after successfully compiling convenience-pkgs, urdf-tools-pkg and graspit-pkgs, my ide can't resolve grasp_planning_graspit nor i can compile with catkin. Any idea what is this about?
By the way i erased graspit-simulator which couses bigger issues. So I think even if i could compile this i can't load world files right?

JenniferBuehler, 2017-02-01:
Hi,

can you provide a few more details please? Are you using catkin_make to compile, which ROS distribution are you using, what's the exact compiler warning, which IDE are you using? How did you delete graspit-simulator, and did you have it in a default path like /usr/, how did you delete it?

Sorry I can't help you without a bit more information ;)
Cheers
Jenny

ilaplace, 2017-02-02:
Certainly,

I'm using catkin_make and ros-indigo. As an IDE I use CLion 2016.1.3. And my error is:

`/home/semih/catkin_ws/src/graspit_planning_tutorial/src/simple_planning_tutorial.cpp:2:61: fatal error: grasp_planning_graspit/GraspItSceneManagerNoGui.h: No such file or directory
 #include <grasp_planning_graspit/GraspItSceneManagerNoGui.h>`

And in my IDE it says: 

> Cannot find 'grasp_planning_graspit'

I used to have graspit simulator on my catkin workspace and later I've also installed via synaptic package manager. I've uninstalled it and deleted the files in my catkin workspace. And I erased the path in my basrc file.

By the way if I before erasing the graspit simulator i was having the same error as our friend over here: [https://github.com/JenniferBuehler/graspit-pkgs/issues/14](url)

Semih

JenniferBuehler, 2017-02-04:
The file it doesn't find has been renamed from GraspItSceneManagerNoGui.h to GraspItSceneManagerHeadless.h a while back. That's also been updated in [simple_planning_tutorial.cpp](https://github.com/JenniferBuehler/graspit-pkgs/blob/master/grasp_planning_graspit/tutorial/src/simple_planning_tutorial.cpp). So you must have an old version for some reason? Do a ``git pull`` in the repository graspit-pkgs. And to clean up properly it's also always worth a try to remove the ``devel`` and ``build`` directories in your catkin workspace and build all from scratch.

ilaplace, 2017-02-08:
Hi,
I've updated the tutorial I'm following, which is:
[(https://github.com/JenniferBuehler/graspit-pkgs/wiki/GraspIt-CPP-API-Tutorial )]
I've deleted my `devel` and `build` directories. I've done a git pull.  

And currently I'm getting this cmake error:
`Error:Could not find a package configuration file provided by "grasp_planning_graspit" with any of the following names:
  grasp_planning_graspitConfig.cmake   grasp_planning_graspit-config.cmake
Add the installation prefix of "grasp_planning_graspit" to CMAKE_PREFIX_PATH or set "grasp_planning_graspit_DIR" to a directory containing one of the above files.  If "grasp_planning_graspit" provides a separate development package or SDK, be sure it has been installed.`

By the way the the file in question exists in my `devel` directory. :(

JenniferBuehler, 2017-02-09:
Are you following the instructions for ROS (not cmake only)?
If it's in your devel directory, maybe your catkin workspace is not set up properly, are you sourcing devel/setup.sh from your .bashrc?

ilaplace, 2017-02-23:
Doing all and some more solved the issue but can't exactly know which one solved it. 

Thanks for your help

closed 2017-02-23 by ilaplace