JenniferBuehler/graspit-pkgs #19: urdf2graspit Segmentation fault

ilaplace, 2017-02-23:
Hi there!

I was trying to convert my Allegro Hand to graspit but i can''t seem to succeed. I'm on ubuntu 14.04 and ros indigo. My command was:
`rosrun urdf2graspit urdf2graspit_node ~/catkin_ws/src/allegro-hand-ros/allegro_hand_description/allegro_hand_description_right.urdf  ~/catkin_ws/src/allegro-grasp hand_root joint_0.0 joint_12.0 joint_4.0 joint_8.0
`
And the output I got is:
`semih@semih-ABRA-A5-V5-2:~/catkin_ws$ rosrun urdf2graspit urdf2graspit_node ~/catkin_ws/src/allegro-hand-ros/allegro_hand_description/allegro_hand_description_right.urdf  ~/catkin_ws/src/allegro-grasp hand_root joint_0.0 joint_12.0 joint_4.0 joint_8.0
[ INFO] [1487865345.239835269]: URDF file: /home/semih/catkin_ws/src/allegro-hand-ros/allegro_hand_description/allegro_hand_description_right.urdf
[ INFO] [1487865345.239878258]: Output dir: /home/semih/catkin_ws/src/allegro-grasp
[ INFO] [1487865345.239888224]: Hand root: hand_root
[ INFO] [1487865345.239907079]: Joint: joint_0.0
[ INFO] [1487865345.239917401]: Joint: joint_12.0
[ INFO] [1487865345.239924455]: Joint: joint_4.0
[ INFO] [1487865345.239932418]: Joint: joint_8.0
[ INFO] [1487865345.240352459]: output_material: <plastic>
[ INFO] [1487865345.240742997]: scale_factor: <1000.000000>
[ INFO] [1487865345.241133731]: negate_joint_movement: <0>
[ INFO] [1487865345.242602582]: Starting model conversion...
[ INFO] [1487865345.242656087]: ### Loading from URDF file /home/semih/catkin_ws/src/allegro-hand-ros/allegro_hand_description/allegro_hand_description_right.urdf...
[ INFO] [1487865345.242876698]: Setting base model directory to /home/semih/catkin_ws/src/allegro-hand-ros/allegro_hand_description
[ INFO] [1487865345.244080686]: ### Converting files starting from link hand_root
[ INFO] [1487865345.244096362]: ### Preparing for DH conversion...
[ INFO] [1487865345.244104523]: ### Joining fixed links..
[ INFO] [1487865345.244118931]: ############### Joining fixed links
[ INFO] [1487865345.244130750]: Start from root link: hand_root
[ INFO] [1487865345.244186002]: ### Transforming rotation axes to z...
[ INFO] [1487865345.244202962]: ############### Aligning rotation axis to [0, 0, 1]
[ INFO] [1487865345.244235026]: ### Converting files...
[ INFO] [1487865345.244247583]: ### Urdf2GraspIt::pretConvert for robot allegro_hand_right
[ INFO] [1487865345.244264270]: ##### Computing DH parameters out of model
[ INFO] [1487865345.244275962]: ############### Getting DH params
[ INFO] [1487865345.244286199]: ### Starting DH conversion from link hand_root
[ INFO] [1487865345.244297490]: ======== Transforming joint joint_0.0 to DH parameters
[ INFO] [1487865345.244307296]: ======== Transforming joint joint_1.0 to DH parameters
[ INFO] [1487865345.244328141]: z-axises intersect! [0, 0, 1] at [0, 0, 0], [0, -1, 2.22045e-16] at [0, 0, 0.0164]
[ INFO] [1487865345.244350501]: ======== Transforming joint joint_2.0 to DH parameters
[ INFO] [1487865345.244362646]: ======== Transforming joint joint_3.0 to DH parameters
[ INFO] [1487865345.244374120]: ======== Transforming joint joint_12.0 to DH parameters
[ INFO] [1487865345.244382848]: ======== Transforming joint joint_13.0 to DH parameters
[ INFO] [1487865345.244397645]: DEBUG-INFO DHParams: Correcting alpha: [-1, 0, 2.22045e-16], [1, 0, 2.83277e-16]
[ INFO] [1487865345.244408883]: ======== Transforming joint joint_14.0 to DH parameters
[ INFO] [1487865345.244424806]: z-axises intersect! [-1, 0, 2.22045e-16] at [-0.0399, 0.005, -0.027], [-2.22045e-16, -1, 4.93038e-32] at [-0.0576, 0.005, -0.027]
[ INFO] [1487865345.244439268]: DEBUG-INFO DHParams: Correcting theta: [2.22045e-16, 0, 1], [-2.22045e-16, 6.12323e-17, -1]
[ INFO] [1487865345.244448649]: ======== Transforming joint joint_15.0 to DH parameters
[ INFO] [1487865345.244459498]: ======== Transforming joint joint_4.0 to DH parameters
[ INFO] [1487865345.244468209]: ======== Transforming joint joint_5.0 to DH parameters
[ INFO] [1487865345.244481644]: z-axises intersect! [0, 0, 1] at [0, 0, 0], [0, -1, 2.22045e-16] at [0, 0, 0.0164]
[ INFO] [1487865345.244521242]: ======== Transforming joint joint_6.0 to DH parameters
[ INFO] [1487865345.244533568]: ======== Transforming joint joint_7.0 to DH parameters
[ INFO] [1487865345.244544049]: ======== Transforming joint joint_8.0 to DH parameters
[ INFO] [1487865345.244552928]: ======== Transforming joint joint_9.0 to DH parameters
[ INFO] [1487865345.244566467]: z-axises intersect! [0, 0, 1] at [0, 0, 0], [0, -1, 2.22045e-16] at [0, 0, 0.0164]
[ INFO] [1487865345.244576756]: ======== Transforming joint joint_10.0 to DH parameters
[ INFO] [1487865345.244587026]: ======== Transforming joint joint_11.0 to DH parameters
[ INFO] [1487865345.244599846]: ############### Transform links to DH reference frames
[ INFO] [1487865345.244640184]: --- DH Parameters: ---
[ INFO] [1487865345.244653384]: joint_0.0: d=0.0164, r=0, theta=0, alpha=1.5708, dof_idx=0
[ INFO] [1487865345.244664050]: joint_1.0: d=0, r=0.054, theta=1.5708, alpha=0, dof_idx=1
[ INFO] [1487865345.244673890]: joint_2.0: d=0, r=0.0384, theta=0, alpha=0, dof_idx=2
[ INFO] [1487865345.244683979]: joint_3.0: d=0, r=0, theta=0, alpha=0, dof_idx=3
[ INFO] [1487865345.244694575]: joint_12.0: d=0.027, r=0.005, theta=1.5708, alpha=-1.5708, dof_idx=4
[ INFO] [1487865345.244704953]: joint_13.0: d=0.0177, r=0, theta=-1.5708, alpha=1.5708, dof_idx=5
[ INFO] [1487865345.244714996]: joint_14.0: d=0, r=0.0514, theta=1.5708, alpha=0, dof_idx=6
[ INFO] [1487865345.244724982]: joint_15.0: d=0, r=0, theta=0, alpha=0, dof_idx=7
[ INFO] [1487865345.244734857]: joint_4.0: d=0.0164, r=0, theta=0, alpha=1.5708, dof_idx=8
[ INFO] [1487865345.244744843]: joint_5.0: d=0, r=0.054, theta=1.5708, alpha=0, dof_idx=9
[ INFO] [1487865345.244755011]: joint_6.0: d=0, r=0.0384, theta=0, alpha=0, dof_idx=10
[ INFO] [1487865345.244765228]: joint_7.0: d=0, r=0, theta=0, alpha=0, dof_idx=11
[ INFO] [1487865345.244775014]: joint_8.0: d=0.0164, r=0, theta=0, alpha=1.5708, dof_idx=12
[ INFO] [1487865345.244785048]: joint_9.0: d=0, r=0.054, theta=1.5708, alpha=0, dof_idx=13
[... 244 lines cut ...]
[ INFO] [1487865360.585648046]: Converting mesh file /home/semih/catkin_ws/src/allegro-hand-ros/allegro_hand_description/meshes/link_0.0.STL with factor 1000
[ INFO] [1487865360.585825605]: Scaling the mesh /home/semih/catkin_ws/src/allegro-hand-ros/allegro_hand_description/meshes/link_0.0.STL with factor 1000
Coin warning in SoBase::setName(): Bad characters in name '_visual_0_link_8.0'. Replacing with name '_visual_0_link_8_0'
Coin warning in SoBase::setName(): Bad characters in name '_link_8.0'. Replacing with name '_link_8_0'
[ INFO] [1487865360.596689663]: Convert mesh for link 'link_9.0'
[ INFO] [1487865360.596798463]: Converting mesh file /home/semih/catkin_ws/src/allegro-hand-ros/allegro_hand_description/meshes/link_1.0.STL with factor 1000
[ INFO] [1487865360.597050495]: Scaling the mesh /home/semih/catkin_ws/src/allegro-hand-ros/allegro_hand_description/meshes/link_1.0.STL with factor 1000
Coin warning in SoBase::setName(): Bad characters in name '_visual_0_link_9.0'. Replacing with name '_visual_0_link_9_0'
Coin warning in SoBase::setName(): Bad characters in name '_link_9.0'. Replacing with name '_link_9_0'
[ INFO] [1487865360.608345435]: Convert mesh for link 'link_10.0'
[ INFO] [1487865360.608415736]: Converting mesh file /home/semih/catkin_ws/src/allegro-hand-ros/allegro_hand_description/meshes/link_2.0.STL with factor 1000
[ INFO] [1487865360.608580671]: Scaling the mesh /home/semih/catkin_ws/src/allegro-hand-ros/allegro_hand_description/meshes/link_2.0.STL with factor 1000
Coin warning in SoBase::setName(): Bad characters in name '_visual_0_link_10.0'. Replacing with name '_visual_0_link_10_0'
Coin warning in SoBase::setName(): Bad characters in name '_link_10.0'. Replacing with name '_link_10_0'
[ INFO] [1487865360.616953647]: Convert mesh for link 'link_11.0'
[ INFO] [1487865360.617013129]: Converting mesh file /home/semih/catkin_ws/src/allegro-hand-ros/allegro_hand_description/meshes/link_3.0.STL with factor 1000
[ INFO] [1487865360.617107700]: Scaling the mesh /home/semih/catkin_ws/src/allegro-hand-ros/allegro_hand_description/meshes/link_3.0.STL with factor 1000
Coin warning in SoBase::setName(): Bad characters in name '_visual_0_link_11.0'. Replacing with name '_visual_0_link_11_0'
[ INFO] [1487865360.617228612]: Converting mesh file /home/semih/catkin_ws/src/allegro-hand-ros/allegro_hand_description/meshes/link_3.0_tip.STL with factor 1000
[ INFO] [1487865360.617413579]: Scaling the mesh /home/semih/catkin_ws/src/allegro-hand-ros/allegro_hand_description/meshes/link_3.0_tip.STL with factor 1000
Coin warning in SoBase::setName(): Bad characters in name '_visual_1_link_11.0'. Replacing with name '_visual_1_link_11_0'
Coin warning in SoBase::setName(): Bad characters in name '_link_11.0'. Replacing with name '_link_11_0'
[ INFO] [1487865360.635377698]: ### Urdf2GraspIt::postConvert for robot allegro_hand_right
Segmentation fault (core dumped)
`

JenniferBuehler, 2017-02-23:
Hi,

sorry for having this trouble, and thanks for helping to test the converter with a new hand. I will look into this over the weekend. If you get around to it before then, it would help if you could send the output of a valgrind run:

```
sudo apt-get install valgrind (if you don't have it already)
cd <your catkin_ws>
catkin_make -DCMAKE_BUILD_TYPE=Debug
rosrun --prefix valgrind urdf2graspit urdf2graspit_node ~/catkin_ws/src/allegro-hand-ros/allegro_hand_description/allegro_hand_description_right.urdf ~/catkin_ws/src/allegro-grasp hand_root joint_0.0 joint_12.0 joint_4.0 joint_8.0
```

And also the link to the allegro URDF (git repo?) you are using would help me to test it.

Cheers
Jenny

ilaplace, 2017-02-24:
Thanks for a quick response,

I'm working on a modified version of the xacro file over there:

[https://github.com/felixduvallet/allegro-hand-ros/tree/master/allegro_hand_description](url)

If you would like to see the exact one:

[https://drive.google.com/file/d/0B1AWehlT7E4xWVNPMXdQOGRscjA/view?usp=sharing](url)

And the output I got with your suggestion:
semih@semih-ABRA-A5-V5-2:~/graspit_ros_ws$ rosrun --prefix valgrind urdf2graspit urdf2graspit_node ~/catkin_ws/src/allegro-hand-ros/allegro_hand_description/allegro_hand_description_right.urdf ~/graspit_ros_ws/src/allegro-grasp hand_root joint_0.0 joint_12.0 joint_4.0 joint_8.0
==15312== Memcheck, a memory error detector
==15312== Copyright (C) 2002-2013, and GNU GPL'd, by Julian Seward et al.
==15312== Using Valgrind-3.10.1 and LibVEX; rerun with -h for copyright info
==15312== Command: /home/semih/graspit_ros_ws/devel/lib/urdf2graspit/urdf2graspit_node /home/semih/catkin_ws/src/allegro-hand-ros/allegro_hand_description/allegro_hand_description_right.urdf /home/semih/graspit_ros_ws/src/allegro-grasp hand_root joint_0.0 joint_12.0 joint_4.0 joint_8.0
==15312== 
[ INFO] [1487926150.532033539]: URDF file: /home/semih/catkin_ws/src/allegro-hand-ros/allegro_hand_description/allegro_hand_description_right.urdf
[ INFO] [1487926150.555730079]: Output dir: /home/semih/graspit_ros_ws/src/allegro-grasp
[ INFO] [1487926150.570397681]: Hand root: hand_root
[ INFO] [1487926150.585749297]: Joint: joint_0.0
[ INFO] [1487926150.586800092]: Joint: joint_12.0
[ INFO] [1487926150.587703414]: Joint: joint_4.0
[ INFO] [1487926150.592983178]: Joint: joint_8.0
[ INFO] [1487926150.598172640]: output_material: <plastic>
[ INFO] [1487926150.634155235]: scale_factor: <1000.000000>
[ INFO] [1487926150.637387060]: negate_joint_movement: <0>
[ INFO] [1487926150.726192050]: Starting model conversion...
[ INFO] [1487926150.742835624]: ### Loading from URDF file /home/semih/catkin_ws/src/allegro-hand-ros/allegro_hand_description/allegro_hand_description_right.urdf...
[ INFO] [1487926150.778952318]: Setting base model directory to /home/semih/catkin_ws/src/allegro-hand-ros/allegro_hand_description
[ INFO] [1487926150.965064477]: ### Converting files starting from link hand_root
[ INFO] [1487926150.966526657]: ### Preparing for DH conversion...
[ INFO] [1487926150.967414703]: ### Joining fixed links..
[ INFO] [1487926150.968870644]: ############### Joining fixed links
[ INFO] [1487926150.970220518]: Start from root link: hand_root
[ INFO] [1487926151.377681329]: ### Transforming rotation axes to z...
[ INFO] [1487926151.386079466]: ############### Aligning rotation axis to [0, 0, 1]
[ INFO] [1487926151.562896971]: ### Converting files...
[ INFO] [1487926151.572490861]: ### Urdf2GraspIt::pretConvert for robot allegro_hand_right
[ INFO] [1487926151.587700689]: ##### Computing DH parameters out of model
[ INFO] [1487926151.588878658]: ############### Getting DH params
[ INFO] [1487926151.591863032]: ### Starting DH conversion from link hand_root
[ INFO] [1487926151.595827418]: ======== Transforming joint joint_0.0 to DH parameters
[ INFO] [1487926151.599571419]: ======== Transforming joint joint_1.0 to DH parameters
[ INFO] [1487926151.654010936]: z-axises intersect! [0, 0, 1] at [0, 0, 0], [0, -1, 2.22045e-16] at [0, 0, 0.0164]
[ INFO] [1487926151.681763903]: ======== Transforming joint joint_2.0 to DH parameters
[ INFO] [1487926151.693956879]: ======== Transforming joint joint_3.0 to DH parameters
[ INFO] [1487926151.698719016]: ======== Transforming joint joint_12.0 to DH parameters
[ INFO] [1487926151.699231042]: ======== Transforming joint joint_13.0 to DH parameters
[ INFO] [1487926151.704190207]: DEBUG-INFO DHParams: Correcting alpha: [-1, 0, 2.22045e-16], [1, 0, 2.83277e-16]
[ INFO] [1487926151.705399803]: ======== Transforming joint joint_14.0 to DH parameters
[ INFO] [1487926151.710583091]: z-axises intersect! [-1, 0, 2.22045e-16] at [-0.0399, 0.005, -0.027], [-2.22045e-16, -1, 4.93038e-32] at [-0.0576, 0.005, -0.027]
[ INFO] [1487926151.715311563]: DEBUG-INFO DHParams: Correcting theta: [2.22045e-16, 0, 1], [-2.22045e-16, 6.12323e-17, -1]
[ INFO] [1487926151.716964824]: ======== Transforming joint joint_15.0 to DH parameters
[ INFO] [1487926151.720839545]: ======== Transforming joint joint_4.0 to DH parameters
[ INFO] [1487926151.721362170]: ======== Transforming joint joint_5.0 to DH parameters
[ INFO] [1487926151.724412896]: z-axises intersect! [0, 0, 1] at [0, 0, 0], [0, -1, 2.22045e-16] at [0, 0, 0.0164]
[ INFO] [1487926151.725296210]: ======== Transforming joint joint_6.0 to DH parameters
[ INFO] [1487926151.728603533]: ======== Transforming joint joint_7.0 to DH parameters
[ INFO] [1487926151.731693637]: ======== Transforming joint joint_8.0 to DH parameters
[ INFO] [1487926151.732185517]: ======== Transforming joint joint_9.0 to DH parameters
[ INFO] [1487926151.735179928]: z-axises intersect! [0, 0, 1] at [0, 0, 0], [0, -1, 2.22045e-16] at [0, 0, 0.0164]
[ INFO] [1487926151.736036659]: ======== Transforming joint joint_10.0 to DH parameters
[ INFO] [1487926151.739307587]: ======== Transforming joint joint_11.0 to DH parameters
[ INFO] [1487926151.747809304]: ############### Transform links to DH reference frames
[ INFO] [1487926151.957089628]: --- DH Parameters: ---
[ INFO] [1487926151.960041159]: joint_0.0: d=0.0164, r=0, theta=0, alpha=1.5708, dof_idx=0
[ INFO] [1487926151.960601932]: joint_1.0: d=0, r=0.054, theta=1.5708, alpha=0, dof_idx=1
[... cut ...]
==15312==    definitely lost: 8,936 bytes in 22 blocks
==15312==    indirectly lost: 6,720 bytes in 252 blocks
==15312==      possibly lost: 13,897,768 bytes in 64,590 blocks
==15312==    still reachable: 3,291,186 bytes in 6,551 blocks
==15312==         suppressed: 0 bytes in 0 blocks
==15312== Rerun with --leak-check=full to see details of leaked memory
==15312== 
==15312== For counts of detected and suppressed errors, rerun with: -v
==15312== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
Killed

JenniferBuehler, 2017-02-26:
It was just something small in the end which was easy to find with gdb instead of valgrind :)
The hand doesn't have an inertial in the URDF, and I didn't catch that case in the converter, boh! I fixed it now.
Pull a new version of graspit-pkgs and it should work:

[image]

JenniferBuehler, 2017-02-26:
Let me know if it works for you now and if it does you can close the issue.

ilaplace, 2017-03-02:
Yes I've created the file, but when I try to load world file it says: (If I make palm_link as wrist link)
```
Creating base...
Could not open /home/semih/catkin_ws/src/graspit-ros/graspit/graspit_source/models/robots/allegro_hand_right/iv/palm_link.xml
Failed to load base
Failed to load robot from file
Deleting Hand: 
Deleting grasp
Deleted robot: 
Could not open /home/semih/catkin_ws/src/graspit-ros/graspit/graspit_source/models/robots/allegro_hand_right/allegro_hand_right.xml
```

And if choose hand_root as the wrist link error becomes this:

```
Creating base...
invalid mass in dynamic body file: models/robots/allegro_hand_right/iv/hand_root.xml
Failed to load base
Failed to load robot from file
Deleting Hand: 
Deleting grasp
Deleted robot: 
Could not open /home/semih/catkin_ws/src/graspit-ros/graspit/graspit_source/models/robots/allegro_hand_right/allegro_hand_right.xml
```
And if I make the mass nonzero at hand_root.xml:

```xml
<?xml version="1.0" ?>
<root>
	<material>plastic</material>
	<mass>1</mass>
	<cog>0 0 0 </cog>
	<inertia_matrix>1 0 00 1 00 0 1</inertia_matrix>
	<geometryFile>hand_root.iv</geometryFile>
</root>
```

The new error:
```
Creating base...
Invalid Inertia Matrix Input
Failed to load base
Failed to load robot from file
Deleting Hand: 
Deleting grasp
Deleted robot: 
Could not open /home/semih/catkin_ws/src/graspit-ros/graspit/graspit_source/models/robots/allegro_hand_right/allegro_hand_right.xml

```

ilaplace, 2017-03-02:
Actually after modifying the inertia_matrix like this:
```xml
	<inertia_matrix>1 0 0 0 1 0 0 0 1</inertia_matrix>
```
I can open the world file but it seems a bit weird though:

[image]

JenniferBuehler, 2017-03-02:
I am guessing that these are the problems you describe when opening the files in GraspIt?
The inertia seems to be important so it looks like GraspIt needs it. You may want to just add it in the URDF then it will be used by urdf2grapit to generate the files.

The output you are posting indeed looks like there is a problem with the DH parameters as well. I didn't try to open it in GraspIt.

I will try to make some time this weekend to look at this and let you know.

JenniferBuehler, 2017-03-05:
Sorry, just saw the problem with the intertia matrix was because of the double 0s, my bad for not testing it in Graspit. I fixed it now so you should not have to manually edit the xml files any more.

As to the other problem with the thumb that is off, this is actually the same issue as #17. I'm looking into this, hopefully will be able to get back to you soon on this. Meanwhile you could try the in-between "fake link" as described in #17.

JenniferBuehler, 2017-03-06:
I have made some sort of progress, though it's still not quite there... there's still some problems with the thumb. The first link's visual element is a bit off, and the rotation axes go the wrong way, which you will notice when you go to "Grasp -> EigenGrasp Interface" and handle the slider. This is an issue which I'm looking at now.

[image]

Also note that the movement with the EigenGrasp Interface, and also the default position of the hand, is not really what it should be. This is because the automatically generated Eigen values are not particularly useful... this is something the user is expected to do manually (see also description [here](https://github.com/JenniferBuehler/graspit-pkgs/wiki/urdf2graspit#eigengrasp-notes) which refers to the file eigen/eigen.xml).

I have other problems with the allegro hand using the version of [simlab robotics](https://github.com/simlabrobotics/allegro_hand_ros) so I think this hand has exposed a few issues with urdf2graspit. I am working on it and will post updates here. Sorry this is not such a quick fix, I'm trying to allocate as much time as possible to fix this issue.

PS: As in the discussion of issue #17 it turns out the DH parameters *can* be calculated, this is what I fixed today. So you won't need the fake link, even if this may resolve the issue by coincidence and may be good to get a temporary fix.

JenniferBuehler, 2017-03-06:
Btw you need to pull a new version of [urdf-tools-pkgs](https://github.com/JenniferBuehler/urdf-tools-pkgs) as well

ilaplace, 2017-03-06:
Actually another interesting issue is that this is not the right hand :P. I have no idea why, I was just working with the right hand urdf but now it seems like I have a left hand.

JenniferBuehler, 2017-03-06:
Indeed, you're right... I didn't notice that. The urdf may have been the wrong one. I just re-generated the right hand from the xacro and it looks ok:

[image]

I also fixed the directional issue with the grasping motion, so it now works with this model (except that the one link visual is still off). Still having troubles with the other repository's allegro hand, but that's something I won't get around to before next weekend.

Do a fresh pull of urdf-tools-pksg, graspit-pkgs and also re-generate your URDF and try again - hopefully you'll have something that you at least can work with this week.

ilaplace, 2017-03-08:
I have generated the urdf from xacro like this: [https://github.com/felixduvallet/allegro-hand-ros](url) (from this fork)

`rosrun xacro xacro --inorder allegro_hand_description_right.xacro > allegro_hand_description_right.urdf`

And I still have the left hand: 
[image]

May I ask which xacro did you use?

JenniferBuehler, 2017-03-08:
Oh weird, I'm pretty sure I was using the same repository. I'm at a different computer now but will check asap.
I forgot the ``--inorder``, otherwise I used the same command.

The output (weird thumb) also looks like you don't have the newest version of urdf-tools-pkgs, did you pull and re-build both urdf-tools-pkgs and graspit-pkgs? I had the same strange thumb before.

ilaplace, 2017-03-11:
Sorry rookie mistake, I've forgotten to build again, mine looks like yours now

JenniferBuehler, 2017-03-12:
I now finally got it to work with the hand in the original repository as well:

[image]

It was a bug in urdf2graspit which remained undiscovered with the other hands - thanks for posting the issue so I could track it!

With the fork by felixduvallet I still have problem that the visual is a bit off. I'm not sure if it's a problem with the modified URDF (and maybe modified mesh files?), I haven't tracked it down. But even with my simple URDF viewer it also looks like that, which makes me think it may be an issue with the URDF itself (since it works with the original now):

[image]

Have you opened the hand with other URDF viewers, or Gazebo, and not observed the same problem with the thumb link? If so, please let me know and I'll see if this is another bug. 

You need to pull urdf-tools-pkgs and graspit-pkgs again.

JenniferBuehler, 2017-03-12:
***PS:*** You can use a launch files like this, which makes life a bit easier (root link name as for the repository by felixduvallet, change it if you are going to use the original repo):

For the URDF viewer:
```xml
<launch>                                                                                            
    <arg name="urdf_file" default="$(find allegro_hand_description)/allegro_hand_description_right.urdf"/>
                                                                                                    
    <!-- command line parameter -->                                                                 
    <arg name="iv_or_link" default=""/>                                                             
                                                                                                    
    <arg name="join_fixed_links" default="false"/>                                                  
    <arg name="rotate_axes_z" default="false"/>                                                     
    <arg name="display_axes" default="true"/>                                                       
    <arg name="axes_radius" default="0.001"/>                                                       
    <arg name="axes_length" default="0.015"/>                                                       
                                                                                                    
    # An axis and angle (degrees) can be specified which will transform *all*                       
    # visuals (not links, but their visuals!) within their local coordinate system.                 
    # This can be used to correct transformation errors which may have been                         
    # introduced in converting meshes from one format to the other, losing orientation information  
    # For example, .dae has an "up vector" definition which may have been ignored.                  
    <arg name="visual_corr_axis_x" default="1"/>                                                    
    <arg name="visual_corr_axis_y" default="0"/>                                                    
    <arg name="visual_corr_axis_z" default="0"/>                                                    
    <arg name="visual_corr_axis_angle" default="0"/>                                                
                                                                                                    
  <node name="urdf_viewer" pkg="urdf_viewer" type="urdf_viewer_node" respawn="false"                
        output="screen" args="$(arg urdf_file) $(arg iv_or_link)">                                  
    <param name="join_fixed_links" value="$(arg join_fixed_links)"/>                                
    <param name="rotate_axes_z" value="$(arg rotate_axes_z)"/>                                      
    <param name="display_axes" value="$(arg display_axes)"/>                                        
    <param name="axes_radius" value="$(arg axes_radius)"/>                                          
    <param name="axes_length" value="$(arg axes_length)"/>                                          
    <param name="visual_corr_axis_x" value="$(arg visual_corr_axis_x)"/>                            
    <param name="visual_corr_axis_y" value="$(arg visual_corr_axis_y)"/>                            
    <param name="visual_corr_axis_z" value="$(arg visual_corr_axis_z)"/>                            
    <param name="visual_corr_axis_angle" value="$(arg visual_corr_axis_angle)"/>                    
  </node>                                                                                           
</launch>     
```

For the urdf2graspit converter:

```xml
<launch>                                                                                            
  # the input urdf file                                                                             
  <arg name="urdf_file" default="$(find allegro_hand_description)/allegro_hand_description_right.urdf"/>
                                                                                                    
  # the directory onto which to write the output files                                              
  <arg name="output_directory"/>                                                                    
  <arg name="hand_root_link" default="palm_link"/>                                                  
  <arg name="finger_joint_names" default="joint_0.0 joint_12.0 joint_4.0 joint_8.0"/>               
  <arg name="scale_factor" default="1000"/>                                                         
                                                                                                    
  <arg name="output_material" default="plastic"/>                                                   
                                                                                                    
  <arg name="launch_prefix" default=""/>                                                            
                                                                                                    
  <include file="$(find urdf2graspit)/launch/urdf2graspit.launch">                                  
    <arg name="launch_prefix" value="$(arg launch_prefix)"/>                                        
    <arg name="urdf_file" value="$(arg urdf_file)"/>                                                
[... cut ...]
        <arg name="finger_joint_names" value="$(arg finger_joint_names)"/>                          
        <arg name="scale_factor" value="$(arg scale_factor)"/>                                      
        <arg name="display_dh_axes" value="$(arg display_dh_axes)"/>                                
    <arg name="negate_joint_movement" value="false"/>                                               
    </include>                                                                                      
</launch>
```

ilaplace, 2017-03-12:
Actually the original urdf file gives me the error:
```
[ INFO] [1489325562.415584363]: ############### Joining fixed links
[ INFO] [1489325562.415600085]: Start from root link: palm_link
[ERROR] [1489325562.415630760]: No link named 'palm_link'
[ERROR] [1489325562.415644672]: Could not join fixed links
[ERROR] [1489325562.415654185]: Could not prepare for DH conversion
[ERROR] [1489325562.415669295]: Failed to process.
```

Any idea what is this about?

JenniferBuehler, 2017-03-12:
You have to change the link name, in the original it's not palm_link, I
think it was base_link

JenniferBuehler, 2017-03-17:
Did you try again? Did it work? If it did, feel free to close the issue.

ilaplace, 2017-03-18:
It worked, thank you for your help!

commit e7f2037f (2017-02-26): Catching case in urdf2graspit for hands with no inertial. Fixes #19
closed 2017-03-18 by ilaplace