JenniferBuehler/graspit-pkgs #20: How to access planning

ilaplace, 2017-03-16:
Hi Jennifer,

First of all I think you have excellent code documentation although I couldn't find way to do what I want. What I need is to create my own planner. So how can I get access to whole joint space that has been reduced by eigengrasps?

JenniferBuehler, 2017-03-17:
To access the whole Graspit search space you will have to dive deeper into the Graspit code itself. You could look at [the code in EigenGraspPlanner::plan()](https://github.com/JenniferBuehler/graspit-pkgs/blob/master/grasp_planning_graspit/src/EigenGraspPlanner.cpp#L313) where the graspit planner is started. However it only runs the GraspIt planner and then gets the results, it doesn't go through all states - that would be silly, GraspIt itself already does that, why do it here again.

One way to access each search state could be if GraspIt! emits a Qt signal at each planning state, then you could make a Qt slot for it (that's like registering a callback). But I don't remember if GraspIt does that, I'd have to look into details of the framework again, it's been a while ;)

There are no tutorials for this kind of detail here, because this package is only intended to serve as an interface to the GraspIt planning algorithms.

JenniferBuehler, 2017-03-17:
Closing and refering to ongoing discussion in [issue 102 on graspit-simulator](https://github.com/graspit-simulator/graspit/issues/102).

closed 2017-03-17 by JenniferBuehler