JenniferBuehler/graspit-pkgs #24: urdf2graspit for my own hand

emigmo, 2017-06-13:
Hi, Jenny.
Bother you agein.
I use your urdf2graspit to convert my own hand( Made by our lab)--named **eagle_shoal**.
I used other software to generate the eagle_shoal.urdf. And there are no problem when I use
## 1:  roslaunch urdf_tutorial display.launch model:=<my-eagle_shoal.urdf-path>.All the DOF is OK.
## 2: roslaunch urdf2graspit example_urdf2graspit_eagleshoal.launch. 
the example_urdf2graspit_eagleshoal.launch : 
`
<launch>
    <arg name="urdf_file" default="$(find eagle_shoal)/urdf/eagle_shoal.urdf"/>
    <arg name="output_directory" default="$(find eagle_shoal)/urdf2graspit/"/>
    <arg name="hand_root_link" default="p_b"/>
    <arg name="finger_joint_names" default="f_3_0_j f_3_1_j f_2_0_m_j"/>
    <arg name="scale_factor" default="1000"/>
    <arg name="output_material" default="plastic"/>
    <include file="$(find urdf2graspit)/launch/urdf2graspit.launch">
        <arg name="urdf_file" value="$(arg urdf_file)"/>
        <arg name="output_directory" value="$(arg output_directory)"/>
        <arg name="hand_root_link" value="$(arg hand_root_link)"/>
        <arg name="finger_joint_names" value="$(arg finger_joint_names)"/>
        <arg name="scale_factor" value="$(arg scale_factor)"/>
        <arg name="output_material" value="$(arg output_material)"/>
        <arg name="negate_joint_movement" value="false"/>
    </include>
</launch>
`

It work well too . And the output info in the terminal is:

`
roslaunch urdf2graspit example_urdf2graspit_eagleshoal.launch
... logging to /home/dex_hand/.ros/log/93297868-4fe3-11e7-8673-38d5472591a7/roslaunch-Dexhand-26880.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://Dexhand:43926/

SUMMARY
========

PARAMETERS
 * /rosdistro: indigo
 * /rosversion: 1.11.21
 * /urdf2graspit_node/negate_joint_movement: False
 * /urdf2graspit_node/output_material: plastic
 * /urdf2graspit_node/scale_factor: 1000
 * /urdf2graspit_node/visual_corr_axis_angle: 0
 * /urdf2graspit_node/visual_corr_axis_x: 1
 * /urdf2graspit_node/visual_corr_axis_y: 0
 * /urdf2graspit_node/visual_corr_axis_z: 0

NODES
  /
    urdf2graspit_node (urdf2graspit/urdf2graspit_node)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[urdf2graspit_node-1]: started with pid [26944]
[ INFO] [1497340512.333904413]: URDF file: /home/dex_hand/catkin_ws/src/eagle_shoal/urdf/eagle_shoal.urdf
[ INFO] [1497340512.333968639]: Output dir: /home/dex_hand/catkin_ws/src/eagle_shoal/urdf2graspit/
[ INFO] [1497340512.333983845]: Hand root: p_b
[ INFO] [1497340512.334009260]: Joint: f_3_0_j
[ INFO] [1497340512.334024797]: Joint: f_3_1_j
[ INFO] [1497340512.334036268]: Joint: f_2_0_m_j
[ INFO] [1497340512.335545647]: output_material: <plastic>
[ INFO] [1497340512.336926511]: scale_factor: <1000.000000>
[ INFO] [1497340512.338362626]: negate_joint_movement: <0>
[ INFO] [1497340512.344182770]: Starting model conversion...
[... 100 lines cut ...]
[ INFO] [1497340513.075985096]: Converting mesh file /home/dex_hand/catkin_ws/src/eagle_shoal/meshes/f_2_0_b.stl with factor 1000
[ INFO] [1497340513.076172020]: Scaling the mesh /home/dex_hand/catkin_ws/src/eagle_shoal/meshes/f_2_0_b.stl with factor 1000
[ INFO] [1497340513.076428883]: Converting mesh file /home/dex_hand/catkin_ws/src/eagle_shoal/meshes/f_2_0_f.stl with factor 1000
[ INFO] [1497340513.076685156]: Scaling the mesh /home/dex_hand/catkin_ws/src/eagle_shoal/meshes/f_2_0_f.stl with factor 1000
[ INFO] [1497340513.149028906]: Convert mesh for link 'f_2_1_b'
[ INFO] [1497340513.149168207]: Converting mesh file /home/dex_hand/catkin_ws/src/eagle_shoal/meshes/f_2_1_b.stl with factor 1000
[ INFO] [1497340513.149981570]: Scaling the mesh /home/dex_hand/catkin_ws/src/eagle_shoal/meshes/f_2_1_b.stl with factor 1000
[ INFO] [1497340513.150777306]: Converting mesh file /home/dex_hand/catkin_ws/src/eagle_shoal/meshes/f_2_1_f.stl with factor 1000
[ INFO] [1497340513.151040903]: Scaling the mesh /home/dex_hand/catkin_ws/src/eagle_shoal/meshes/f_2_1_f.stl with factor 1000
[ INFO] [1497340513.236713815]: ### Urdf2GraspIt::postConvert for robot eagle_shoal
[ INFO] [1497340513.236875004]: ############### Getting XML
[ INFO] [1497340513.237560075]: Conversion done.
[ INFO] [1497340513.237735750]: urdf2inventor::FileIO::writeMeshFiles into /home/dex_hand/catkin_ws/src/eagle_shoal/urdf2graspit/
[ INFO] [1497340513.252261488]: Cleaning up...
[ INFO] [1497340513.252291655]: Done.
[urdf2graspit_node-1] process has finished cleanly
log file: /home/dex_hand/.ros/log/93297868-4fe3-11e7-8673-38d5472591a7/urdf2graspit_node-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
`

But When I input the world/robot file, There are nothing at all.
Can you help me?

JenniferBuehler, 2017-06-16:
What do you mean, there is nothing at all? You can't find the generated world files? Or when you open it in graspit simulator, it doesn't show anything?

JenniferBuehler, 2017-09-12:
I'm guessing this was resolved?

closed 2017-09-12 by JenniferBuehler