JenniferBuehler/graspit-pkgs #25: Is GraspIt library compatible with arbitrary hand model?

Booooooosh, 2017-06-25:
Does anyone know whether the GraspIt library works with arbitrary hand model? Can I still use this library if I want to grasp with a suction cup? Or use three finger gripper instead of two finger gripper?

Many thanks!

JenniferBuehler, 2017-06-26:
For each new hand and object, you will need to train/compute the grasps using the GraspIt simulator (or the C++ API as in my packages). You cannot re-use grasps for other hands.. Only if the grippers are extremely similar, you may be able to re-use the grasp and modify it accordingly (e.g. joint names and maybe slight adjustments) to match your hand, but that's not intended by the graspit framework and will require fiddling.

Suction cups are not supported as far as I know (maybe ask in the original repository?), but for a simple grasp like that you may not need a complex grasp planner at all, depending on how your hand looks like.

Hope this helps.

Booooooosh, 2017-06-26:
Got it, thanks very much! Seems that if you are using a sucking cup, all you need to compute is the position where the cup can reliably adhere on the object.

closed 2017-06-26 by JenniferBuehler