JenniferBuehler/graspit-pkgs #26: manipulation Grasp Results are not correct

sameergrey, 2017-07-17:
Hey Jennifer
We are using Graspit.Services to integrate our Custom Robot hand with UR5 Robot.So we are using graspit for Grasp execution and evaluation.
We successfully made graspit files for our Robot hand and able to spawn in graspit simulator.
So now we are using ROS to integrate with our Robot for that we are using Graspit ROS Services from your wiki.I made a simple client in ROS to communicate with the ROS Service **grasp_planning_service.cpp** node.
Everything is working OK except that when i am trying to get the pose and orientation of Robot hand i am not getting the correct values but at the same time when i check the output world file it shows the correct value.
My frame id is "1".
I am able to load and add the object and robot to the database model.
Do i am missing anything?

JenniferBuehler, 2017-07-21:
Hello,

Sorry about the delay in getting back to you. A bit more information would
be great to be able to help you. How are you trying to get the pose? The
world file has nothing to do with the state, it only decribes the robot
hand, it doesn't have the current state. The grasp file/result message
should contain the *target* state for the grasp. You could get the current
state from the joint state of the robot.

If you could describe your scenario in a bit more detail, that would be
great.

Cheers
Jenny

<https://github.com/JenniferBuehler/graspit-pkgs/issues/26>, or mute the
thread

sameergrey, 2017-07-25:
Dear JenniferBuehler,
Thanks a lot for your response.
We need to pick the object from  one bin of Rack and put it in another bin.So our environment is always a cluttered one.
For above application we are using UR5 as a robot and one 3Fingered custom made gripper.

We have object file in format .stl with Pose (eventually object will be in bin) based on that we start our 
planning in Graspit.So we feed the Object .stl and the Robot Hand and start the planning.
We are using same **grasp_planning_graspit_ros** package to communicate with the Graspit world.
We made our ROS client to communicate with the Services.Now we have few queries to ask

As per the attachment we are getting x,y,z and roll pitch yaw of robot hand on **Grasp.msg
** at our client and but it is not matching with the Output  result .world file...why so?

Can we give .pcd file of object to Graspit Services is it working?

For collision detection and better result i need to add Robot in Graspit world how can i do that because when we generate the Graspit file for Robot hand we only has to give root_link and the Joints names which is attached to the palm of robotic hand?

Why grasp_quality is not coming to 100% when we plan in free space how to compute and analyze the value?

Why mesh_rotation issue occur?

[image]
[modelo_pick_object_1.txt](https://github.com/JenniferBuehler/graspit-pkgs/files/1172390/modelo_pick_object_1.txt)

JenniferBuehler, 2017-08-12:
Apologies for the delay again. I'm extremely tied up at the moment and hardly find time to tend to the github issues ;)

The pose in Grasp.msg is relative to the object. Did you take this into account already?

The current version of urdf2graspit is set up for using it starting from the wrist. This is because if you add all degrees of freedom, the grasp planner in graspit will not give you good results. The collision detection to actually grasp the object should then be done by your motion planner, e.g. MoveIt, as you try to move the hand into grasping position.

No, there is no support for .pcd files.

The grasp quality coming out of graspit is a bit vague. If you want to know more about how it is computed, I suggest you read the GraspIt papers or ask in the graspit github repository maybe? :)

The rotation issue occurs because .dae files have an "up" vector which can be set to any axis, while we internally need it to always be the z axis. So if you have .dae files which use another axis, the links have to be rotated.

Hope this answers your questions!

sameergrey, 2017-08-16:
Thanks a lot for Your reply.

1.Yes we figured out the Grasp.msg is relative to the object so that issues is resolve now.

2.We have to use MoveIt Planner instead of to spawn whole Robotic Arm.

3.If possible can we dig down and make this happen to add the .pcd file support if yes than how much effort we need to put to make this work?

4.I will ask to Graspit Repo.

5.Yup i understand i will take care of the rotation while converting to Graspit Files.

JenniferBuehler, 2017-08-17:
Hi,

to 3.: I don't have time to do this right now unfortunately, but if you're happy to do this, you can send a pull request :)

Hope your project is moving along fine!

Cheers
Jenny

JenniferBuehler, 2017-09-12:
Was this resolved?

sameergrey, 2017-09-12:
Yes this issue is resolved.

closed 2017-09-12 by JenniferBuehler