JenniferBuehler/graspit-pkgs #27: Object addition to graspit world runs into infinite loop

akash-greyorange, 2017-09-07:
Hello @JenniferBuehler ,
   I am adding some graspable object in graspit world along with my robot.I do have some obstacle over which I am placing this graspable object.Pose for this object is basically fetched through pose estimation running on my machine. I am getting following logs and it goes on infinitily printing it without planning in some cases.

 INFO] [1504805605.651663695]: Now starting the planning process... - GraspItServices.cpp, 416
[ INFO] [1504805605.651717189]: Initializing planner type - EigenGraspPlanner.cpp, 953
[ INFO] [1504805605.651761143]: Initiating planning, trial #0 - EigenGraspPlanner.cpp, 359
[ INFO] [1504805605.651783316]: Waiting for planning algorithm to finish... - EigenGraspPlanner.cpp, 364
[ INFO] [1504805605.651820976]: ### Start planner! ### - EigenGraspPlanner.cpp, 220
[ INFO] [1504805605.651845964]: Entering planner loop... - EigenGraspPlanner.cpp, 237
created new L1 Norm GWS.
Neighbor gen loops: 11
Neighbor gen loops: 11
Neighbor gen loops: 11
Neighbor gen loops: 13
Neighbor gen loops: 11
Neighbor gen loops: 16
Neighbor gen loops: 12

Code never comes out of this loop.What can be wrong with this?

akash-greyorange, 2017-09-08:
Update on this issue.If I use SpaceSearch option as COMPLETE then it solves my problem and dosent run infinitily into Neighbor gen loops but obviously making space search COMPLETE takes some time to start EGPlanner.But Currently your package dosent support space search other then Axis Angle so I am kind of stuck.Is their any chance of integrating COMPLETE space search into existing package?Or can we solve this EGPlanner going into Neighbor gen loops forever by twiking Axis angle space search?

akash-greyorange, 2017-09-09:
Another Update, Is there any chance to update graspit binary files since I raised issue on graspit simulator repo and suggestion I got was to check for latest update of graspit source .so file.There are some problem in loading robotcontacts it seems to be.

JenniferBuehler, 2017-09-12:
Hello,

so sorry for the late reply, I'm pretty tied up lately. I've seen the issue on the graspit repo but haven't had the time to comment...

Yes absolutely, my repo needs to be synced with the main graspit repo again! This hasn't happened in a year or so. Reason being that I didn't get around to it. It should be a straight forward merge with upstream, but there may be some incompatibilities with the CMakeLists.txt/package.xml if I remember correctly. That's why I wanted to do it with care and subsequent proper testing. And again, haven't had the time :/ 

If you would like to merge with the main repo and test your code to see if it fixes the issue, give it a go! I will put this on my TODO list and try to do it ASAP, hopefully within the next couple of weeks when I get a free hour.

Cheers
Jenny

akash-greyorange, 2017-09-12:
Hello,
 Yes I tried merging main repo with existing interface but it dosent seem to work since there are some incompatibilities as mentioned by you.So please sync it once you find free time or else if there are any guidelines for merging you can share them with me.When I tried merging it showed that some .h files like robot.h , world.h are discontinued and not been used currently and which is heavily used in grasp_planning_graspit_ros package.Rest regarding my issue I am discussing it with graspit developer on its main repo. :)
Cheers,
Akash

JenniferBuehler, 2017-09-12:
Hi Akash,

ok I'll see to that it will be merged asap, will put it on the top of my TODO list for my free time. There are no guidelines, except merge the graspit repo (my fork) with upstream and resolve conflicts, then check that graspit-pkgs repo still compiles both with cmake only and with catkin (as described in tutorials). If you want to, give it a go! I can't promise to get around to it before in 1-2 weeks time, but will give my best to find time.

Cheers
Jenny

akash-greyorange, 2017-09-13:
Sure @JenniferBuehler  I might give it a try in upcoming few days and if you find any free time kindly do it and notify on this thread. :)

JenniferBuehler, 2017-09-24:
Hi, I have merged it with the newest version of graspit. It's not fully tested yet, therefore I still have the changes in special branches.

For my [graspit fork](https://github.com/JenniferBuehler/graspit/) you will need the branch ``merge-upstream-0916`` and for this repo (graspit-pkgs) you will need ``merge-graspit-upstream-0916``.

It compiles and works in the simulator, however when using the C++ interface the results still don't seem OK, but it may be because I haven't tested it properly. I'll have to look into this separately. Meanwhile you can try it out. I'll report back here when I have more time to fix up the C++ interface.

akash-greyorange, 2017-09-25:
Hi @JenniferBuehler  Thanks for taking this on your priority inspite of your busy schedule. I will surely try it out.Majorly I need C++ interface for my application so that's where I will be looking for.Once that gets stable I could be able to include it into my main stream.

JenniferBuehler, 2017-09-25:
Ok great, let me know how it goes. Once it's tested I'll merge it to the
master branch. Also I'll extend support for the other energy functions and
search types incl. COMPLETE. Hopefully I'll get around to do it soon ;)

By the way I forgot to say, you'll have to re-generate the contacts file
for your robot, in the new version of graspit it's now in xml format. The
old .vgr files won't work any more. I've updated urdf2graspit to generate
xml now.

Let me know if it works with the c++ interface.

akash-greyorange, 2017-09-27:
@JenniferBuehler Thanks for the help.Surely will test it and update to you.I added search type COMPLETE by uncommenting some of your code and by adding some.It works good for me now.

JenniferBuehler, 2017-10-05:
Glad it works now! Will test it again as well and add other search types, then report here when I merge with master.
Cheers

akash-greyorange, 2017-10-05:
Is CPP interface tested @JenniferBuehler for latest graspit source?I am not finding time to give it a try.

JenniferBuehler, 2018-01-05:
Sorry I just saw this now, oups... it's ages ago! 
I have just updated the master branch to work with the newest graspit source. Give it a try and let me know if you still experience this issue.

akash-greyorange, 2018-01-08:
:) It's okay @JenniferBuehler  .Glad you replied.I am currently not working on Graspit and will look into this whenever I will be back onto this.Closing this issue for now.

closed 2018-01-08 by akash-greyorange