JenniferBuehler/graspit-pkgs #28: use the urdf2graspit package, always error to generate contact points.

emigmo, 2017-09-09:
`[ INFO] [1504927114.111518864]: Clicked link jaco_6_hand_limb, point ,-0.102241,0.00252118,-0.0020643
contacts_generator_node: /usr/include/eigen3/Eigen/src/Core/DenseCoeffsBase.h:337: Eigen::DenseCoeffsBase<Derived, 1>::Scalar& Eigen::DenseCoeffsBase<Derived, 1>::operator()(Eigen::DenseCoeffsBase<Derived, 1>::Index, Eigen::DenseCoeffsBase<Derived, 1>::Index) [with Derived = Eigen::Matrix<double, 3, 3>; Eigen::DenseCoeffsBase<Derived, 1>::Scalar = double; Eigen::DenseCoeffsBase<Derived, 1>::Index = long int]: Assertion `row >= 0 && row < rows() && col >= 0 && col < cols()' failed.
`

JenniferBuehler, 2017-09-12:
Weird, I haven't seen that one before... can you describe your set-up a bit more please? Are you following [this tutorial](https://github.com/JenniferBuehler/graspit-pkgs/wiki/Conversion-with-urdf2graspit)? Are you on Ubuntu? Which version?

JenniferBuehler, 2018-01-05:
Is this issue still relevant, or can we close this?

echeng22, 2018-03-09:
Hi, I am also seeing this error, though when using a model of my own, not the tutorial model. The error message seems similar
` [ INFO] [1520618690.578285218]: Clicked link left_inner_finger, point ,-0.00361292,0.000723158,0.0439998
contacts_generator_node: /usr/include/eigen3/Eigen/src/Core/DenseCoeffsBase.h:353: Eigen::DenseCoeffsBase<Derived, 1>::Scalar& Eigen::DenseCoeffsBase<Derived, 1>::operator()(Eigen::Index, Eigen::Index) [with Derived = Eigen::Matrix<double, 3, 3>; Eigen::DenseCoeffsBase<Derived, 1>::Scalar = double; Eigen::Index = long int]: Assertion `row >= 0 && row < rows() && col >= 0 && col < cols()' failed.
`
I am on Ubuntu 16.04 using ROS kinetic if that helps.

JenniferBuehler, 2018-03-11:
Thanks for letting me know. I'll look into this.
Any chance you are able to share your URDF?

echeng22, 2018-03-12:
Yes, I've attached the urdf as well as the meshes I've been using for it. 

I did end up finding a workaround for adding contacts manually in this comment thread [here](https://github.com/graspit-simulator/graspit/issues/126#issuecomment-325983510) in the graspit! issues thread.

[robotiq_urdf.tar.gz](https://github.com/JenniferBuehler/graspit-pkgs/files/1803762/robotiq_urdf.tar.gz)

JenniferBuehler, 2018-03-13:
Ah awesome, I didn't know that this can now be done in GraspIt too, great! Good you have something to work with in the meantime.
I'll look at what's causing the issue as soon as I get around to, this week is looking pretty busy, so I'm glad it's not too pressing any more ;)

JenniferBuehler, 2018-03-13:
Ah and thanks for sharing the URDF!

JenniferBuehler, 2018-04-03:
I've just tested this on 16.04/Kinetic, and unfortunately could not reproduce it :( 

I ran:

``rosrun urdf2graspit urdf2graspit_node `rospack find robotiq_arg85`/urdf/robotiq_change.urdf ~/robotiq robotiq_85_base_link left_inner_finger_joint right_inner_finger_joint``

and then to generate the contacts:

``rosrun urdf2graspit contacts_generator_node  `rospack find robotiq_arg85`/urdf/robotiq_change.urdf ~/robotiq robotiq_85_base_link left_inner_finger_joint right_inner_finger_joint
``

All works fine.

Can you remember whether you got this error every time you clicked the hand to add a contact? Or was it only occasionally or when you clicked a specific spot on the finger?

Also I've tested all my packages with kinetic and did some changes, if you get the chance can you please update the packages moveit-pkgs, urdf-tools-pkgs and graspit-pkgs and try again? If you don't get the time to do it because you're not working on this any more, that's fine. Then please just let me know if you remember how this error happened (and if always).

echeng22, 2018-04-03:
Yes, this error occurred everytime I clicked the hand to add a contact. I did update the packages and still run into the same error. Since I found a workaround for it, and I'm further along on my project with it, I don't think I'll have too much time to look into it. Thanks for looking into it for me!

JenniferBuehler, 2018-04-05:
Hmmmm if it always happens on your system, that will make it difficult for me to fix it because I can't reproduce it. One more question: which exact package version of Eigen are you using? Can you send the output of 

``apt search libeigen3-dev``
and

``apt search ros-kinetic-eigen-conversions``

please? It will probably be the same as mine and won't make a difference, but at the lack of other ideas right now, let's see if we have different versions.

echeng22, 2018-04-05:
`apt search libeigen3-dev`
libeigen3-dev/xenial,xenial,now 3.3~beta1-2 all [installed]
lightweight C++ template library for linear algebra

`apt search ros-kinetic-eigen-conversions`
ros-kinetic-eigen-conversions/xenial,now 1.11.9-0xenial-20180222-204052-0800 amd64 [installed]
Conversion functions between: - Eigen and KDL - Eigen and geometry_msgs.

JenniferBuehler, 2018-04-17:
Hm, it's the same...
because I can't reproduce this, I won't look into this further for now, if anyone else has this problem, please put your hand up and I'll look at this again. Leaving this open for now.
Thanks again for reporting this issue!

JenniferBuehler, 2019-03-21:
Since there has been no movement here, I will close this for now. If anyone still has this issue, please re-open.

artemiialessandrini, 2019-04-30:
Clicked link simple_gripper_gripper_finger2_knuckle_link, point ,0.0280812,0.012,0.00535807
contacts_generator_node: /usr/include/eigen3/Eigen/src/Core/DenseCoeffsBase.h:353: Eigen::DenseCoeffsBase<Derived, 1>::Scalar& Eigen::DenseCoeffsBase<Derived, 1>::operator()(Eigen::Index, Eigen::Index) [with Derived = Eigen::Matrix<double, 3, 3>; Eigen::DenseCoeffsBase<Derived, 1>::Scalar = double; Eigen::Index = long int]: Assertion `row >= 0 && row < rows() && col >= 0 && col < cols()' failed.

Got it!:)

JenniferBuehler, 2019-05-03:
Which robot were you using? Can you share the urdf files?

artemiialessandrini, 2019-05-03:
I managed to fix it someways:D
Was having the same issue with wg_perseption tabletop detection, could it be a qt4 qt5 version conflict?
[this](https://github.com/wg-perception/object_recognition_ros_visualization/issues/5) or [this](https://github.com/PickNikRobotics/rviz_visual_tools/issues/79)?
I also reinstalled rviz_visual_tools package.
I'll check it again in a couple days.

JenniferBuehler, 2019-05-03:
Thanks for letting me know. Could you tell me how you someways managed to fix it?
Since I can't reproduce it, it's hard to tell where it comes from. The problem itself obviously comes from Eigen, I guess it would have been fed with an empty matrix or something. But it can't be related to the model because it's worked for me while others had issues on the same model.
I wish I could reproduce it, if I could check with debugging techniques I may be able to resolve it right away :(

Let me know what you did to somehow fix it, and if you could share the urdf files that would help too. Once you check this again, let me know as well :)
Thank you!

artemiialessandrini, 2019-05-03:
Sure, I'll let you know!

artemiialessandrini, 2019-10-25:
Hi @a-price
Great, please share your URDF file, I'm also interested, due to wasn't able to reproduce the issue last time.

a-price, 2019-10-26:
Ok, try this one: https://github.com/a-price/robotiq_3finger_description.
(I pulled it out of a larger project, so I apologize if there's a broken link that I missed.)

Assuming that you're building the GraspIt files in `~/grasping_ws/src/robotiq_graspit`, run with:

```
rosrun urdf2graspit urdf2graspit_node $(rospack find robotiq_3finger_description)/urdf/robotiq_3finger_description.urdf ~/grasping_ws/src/robotiq_graspit gripper_palm gripper_fingerA_joint_2 gripper_fingerB_knuckle gripper_fingerC_knuckle
```
and
```
rosrun urdf2graspit contacts_generator_node $(rospack find robotiq_3finger_description)/urdf/robotiq_3finger_description.urdf ~/grasping_ws/src/robotiq_graspit gripper_palm gripper_fingerA_joint_2 gripper_fingerB_knuckle gripper_fingerC_knuckle
```

a-price, 2019-10-26:
The latter fails when selecting a contact point on my machine with https://github.com/JenniferBuehler/urdf-tools-pkgs/commit/39b7638fbd530073cab7327be81ef5a0ce453951, but succeeds with https://github.com/JenniferBuehler/urdf-tools-pkgs/pull/9.

However, it isn't able to automatically sample grasps. Possibly because GraspIt considers collisions between adjacent links?

JenniferBuehler, 2019-10-26:
Thanks for reporting the issue and the PR! I will take a look tonight or tomorrow when I get a little time for it.

JenniferBuehler, 2019-10-27:
Thanks for the PR @a-price, that was definitely a bug. I have merged it.
Can everyone try again if it fixes the issues please?

a-price, 2019-10-28:
Sure thing! (Also, I found the issue with the hand model not automatically planning: there were non-adjacent links that were in collision by default.)

JenniferBuehler, 2019-10-29:
Yes I saw your PR to graspit, well done :)

I will close this issue for now, please anyone re-open if you still have this issue.

closed 2019-10-29 by JenniferBuehler