JenniferBuehler/graspit-pkgs #29: energy function

Shentheman, 2017-10-05:
Hi Jennifer:
By running `rosrun grasp_planning_graspit graspit_simulator`, I can see that the energy formulations include Contacts and Quality, Hand Contacts, Potential Quality, Autograsp Quality and so on. But from the original version, I see all of the [energy formulations](https://github.com/graspit-simulator/graspit/tree/master/include/graspit/EGPlanner/energy) are different. Are there a matching between your energy functions and the original ones?

Thank you!

JenniferBuehler, 2017-10-05:
Hi,
The graspit version used on the master branch is old. I'm currently merging with the new graspit version in the branch ``merge-graspit-upstream-0916`` (and for my graspit fork, if you are compiling it separately, you will need branch ``merge-upstream-0916``). The simulator will work like the new original graspit version,  but the c++ interface is still being tested with the new version. If you have any issues with it, please report here.

JenniferBuehler, 2018-01-05:
Update: I have finally gotten around to give this a test run. The current master version should now work with the newest graspit version. If this is still relevant to you, can you give it a test run? Otherwise, feel free to close this issue.

Shentheman, 2018-01-06:
Hi Jennifer!
Thank you so much for updating the code!
Currently, I am not using it. I will close the issue for now and test it in the future!

closed 2018-01-06 by Shentheman