JenniferBuehler/graspit-pkgs #31: Using urdf2graspit with single joint hand

tkelestemur, 2018-03-25:
Hello @JenniferBuehler 

First of all thanks for the great set of packages, they save a lot of time for getting started with GraspIt!.

I am currently trying to convert my robot's end-effector model from urdf to graspit. It has a parallel gripper controlled with one actuated joint. So when I try to run urdf2graspit I am getting following error:

```
[ERROR] [1521939504.622503746]: Not enough arguments!
[ INFO] [1521939504.622523805]: Usage: /home/tarik/ros_ws/devel/lib/urdf2graspit/urdf2graspit_node <urdf-file> <output-directory> <palm link name> <finger joint name 0> ... <finger joint name n>
```

I am running as follows: 
```
rosrun urdf2graspit urdf2graspit_node /opt/ros/kinetic/share/hsrb_description/robots/hsrb.urdf /home/tarik/ros_ws/src/frasier_graspit/models hand_palm_link hand_motor_joint
```
I checked the source code and it is throwing this error when the number of arguments is less than 6 so I guess it is expecting at least two joints. 
Do you think I can fix this somehow?

Best,

tkelestemur, 2018-03-26:
Thanks for the quick reply :) I have just changed the number of expected number of arguments to 5 and tried like that. It works like this, however, I'm getting following error:
```
tarik@rivalab-firat:~$ roslaunch frasier_graspit urdf2graspit.launch 
... logging to /home/tarik/.ros/log/2fcc723e-3106-11e8-84f6-408d5c50ac92/roslaunch-rivalab-firat-7303.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.18:42625/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.12
 * /urdf2graspit_node/negate_joint_movement: False
 * /urdf2graspit_node/output_material: plastic
 * /urdf2graspit_node/scale_factor: 1
 * /urdf2graspit_node/visual_corr_axis_angle: 0
 * /urdf2graspit_node/visual_corr_axis_x: 0
 * /urdf2graspit_node/visual_corr_axis_y: 0
 * /urdf2graspit_node/visual_corr_axis_z: 1

NODES
  /
    urdf2graspit_node (urdf2graspit/urdf2graspit_node)

auto-starting new master
process[master]: started with pid [7313]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 2fcc723e-3106-11e8-84f6-408d5c50ac92
process[rosout-1]: started with pid [7329]
started core service [/rosout]
process[urdf2graspit_node-2]: started with pid [7340]
[ INFO] [1522076328.109424292]: URDF file: /opt/ros/kinetic/share/hsrb_description/robots/hsrb.urdf
[ INFO] [1522076328.109460749]: Output dir: /home/tarik/ros_ws/src/frasier_graspit/models
[ INFO] [1522076328.109467666]: Hand root: hand_palm_link
[ INFO] [1522076328.109473552]: Joint: hand_motor_joint
[ INFO] [1522076328.110177352]: output_material: <plastic>
[ INFO] [1522076328.110816508]: scale_factor: <1.000000>
[ INFO] [1522076328.111333560]: negate_joint_movement: <0>
[ INFO] [1522076328.113757759]: Starting model conversion...
[ INFO] [1522076328.113849770]: ### Loading from URDF file /opt/ros/kinetic/share/hsrb_description/robots/hsrb.urdf...
[ INFO] [1522076328.114213533]: Setting base model directory to /opt/ros/kinetic/share/hsrb_description/robots
[ INFO] [1522076328.116007451]: ### Converting files starting from link hand_palm_link
[ INFO] [1522076328.116023938]: ### Preparing for DH conversion...
[ INFO] [1522076328.116029923]: ### Joining fixed links..
[ INFO] [1522076328.116040908]: ############### Joining fixed links
[ INFO] [1522076328.116055173]: Start from root link: hand_palm_link
[ INFO] [1522076328.116227519]: Joining fixed joint (hand_camera_gazebo_frame_joint) between hand_camera_frame and hand_camera_gazebo_frame
[ INFO] [1522076328.116680732]: Joining fixed joint (hand_camera_frame_joint) between hand_palm_link and hand_camera_frame
[ INFO] [1522076328.116784029]: Joining fixed joint (hand_l_finger_tip_frame_joint) between hand_l_distal_link and hand_l_finger_tip_frame
[ INFO] [1522076328.116849634]: Joining fixed joint (hand_l_finger_vacuum_frame_joint) between hand_l_distal_link and hand_l_finger_vacuum_frame
[ INFO] [1522076328.117468734]: Joining fixed joint (hand_r_finger_tip_frame_joint) between hand_r_distal_link and hand_r_finger_tip_frame
[ INFO] [1522076328.117574509]: ### Transforming rotation axes to z...
[ INFO] [1522076328.117598522]: ############### Aligning rotation axis to [0, 0, 1]
[ INFO] [1522076328.121359784]: ### Converting files...
[ INFO] [1522076328.121409099]: ### Urdf2GraspIt::pretConvert for robot hsrb
[ INFO] [1522076328.121459871]: ##### Computing DH parameters out of model
[ INFO] [1522076328.121473526]: ############### Getting DH params
[ INFO] [1522076328.121512897]: ### Starting DH conversion from link hand_palm_link
[ INFO] [1522076328.121532619]: ======== Transforming joint hand_l_proximal_joint to DH parameters.
[ INFO] [1522076328.121561183]: ======== Transforming joint hand_l_spring_proximal_joint to DH parameters.
[ WARN] [1522076328.121844007]: z-axises equal. No common normal can be obtained ([0, 0, 1] at [0, 0, 0], [0, 0, 1] at [0, 0, 0])
[ERROR] [1522076328.121861467]: Common normal can't be obtained
[ERROR] [1522076328.121878107]: could not obtain dh params
[ERROR] [1522076328.121893379]: Could not get DH parameters
[ERROR] [1522076328.121906374]: Could not transform to DH reference frames
[ERROR] [1522076328.121932076]: Failed to pre-convert
[ERROR] [1522076328.121948153]: Could not do the conversion
[ERROR] [1522076328.121965062]: Failed to process.
[urdf2graspit_node-2] process has finished cleanly
log file: /home/tarik/.ros/log/2fcc723e-3106-11e8-84f6-408d5c50ac92/urdf2graspit_node-2*.log
^C[rosout-1] killing on exit
```
I guess this is related to the urdf of my robot. [Here](https://gist.github.com/tkelestemur/f428927da4141e15f5597732b7b23ec2) is the urdf of the hand if it helps. I also changed the `visual_corr_axis_angle` to 90 from 0 as it was not showing correct with urdf_viewer.

Thanks,

JenniferBuehler, 2018-03-26:
Hi, thanks for sharing the URDF and the error.
There are issues with computing the DH parameters because the joints are aligned, I'll have to have a closer look at that but I'm flat out the next few days. I hope I'll get around to do that on the weekend, I hope you have enough other stuff to do until then ;)

JenniferBuehler, 2018-04-03:
Hello, just about to look into this. I need the package ``hsrb_description`` for the meshes. I guess this is the HSR hand or some modification of it? Can you let me know which repository you are using for this, or is it private? I guess it's not [this one](https://github.com/ToyotaResearchInstitute/hsr_description/tree/master/urdf) because it's called ``hsr_description`` and also the URDF is different.
I need the meshes or I won't be able to generate the GraspIt files...

tkelestemur, 2018-04-03:
Hello,

Thank you for spending your time on this. `hsr_description` package is the public version of the `hsrb_description`. Since this is a private package, I cannot share it without permission but I am going ask for it. I just took a look at the `hsr_description` [urdf](https://github.com/ToyotaResearchInstitute/hsr_description/blob/master/robots/hsrb4s.urdf) file and it is only slightly different than what I have. So whatever works for that should work for the private version.

Best,

JenniferBuehler, 2018-04-05:
Ok, thanks. I ended up just putting fake boxes instead of the meshes. I can't see if the visuals are fine, but the joint configuration looks like a hand to me.

[image]

I believe I've managed to fix it, in this special case the common normal which it failed to compute before can be any perpendicular vector.

However when I convert it with only the one joint ``hand_motor_joint`` and load the converted model into ``graspit_simulator``, it crashes. Not sure why, I haven't investigated this further. If I use more joints however, loading it into ``graspit_simulator`` it works:

``rosrun urdf2graspit urdf2graspit_node  `rospack find hrb_sample`/urdf/hsr_hand.urdf ~/hsr_results wrist_roll_link hand_motor_joint hand_l_proximal_joint hand_r_proximal_joint``

Can you pull the update and try again please?

JenniferBuehler, 2018-04-17:
Did it work? If yes, or if you don't need this thread any more, can you please close the issue :) Thanks!

JenniferBuehler, 2018-07-19:
Closing due to inactivity.

closed 2018-07-19 by JenniferBuehler