JenniferBuehler/graspit-pkgs #32: Ged rid of manipulation_msgs and household_objects_database_msgs dependency

JenniferBuehler, 2018-04-03:
The ROS interface can now use ``moveit_msgs`` messages as they contain all information required now.

ksjordan, 2018-05-21:
If I have moveit_msgs does that mean I can remove references to these dependencies on CMakeLists in graspit-pkgs?

JenniferBuehler, 2018-05-23:
Yes, though you will probably have to change the include directives and namespace directives in the code as well. The compiler will guide you through it ;) I've been wanting to do it for a while but don't get around to it. If you end up doing it, I'll be happy about a PR :))

JenniferBuehler, 2019-02-28:
This was already half done (currently still in branch `melodic-support`), but before completing and closing this properly, issues #40 and #41 need to be resolved.