JenniferBuehler/graspit-pkgs #33: graspit! I can not see the screen.

ambl2357, 2018-04-05:
[image]

I have run the 'graspit_simulator'

but  UI is blank.

QPixmap::fromMimeSource: Cannot find pixmap "filenew.xpm" in the mime source factory
QPixmap::fromMimeSource: Cannot find pixmap "fileopen.xpm" in the mime source factory
QPixmap::fromMimeSource: Cannot find pixmap "filesave.xpm" in the mime source factory
QPixmap::fromMimeSource: Cannot find pixmap "play.xpm" in the mime source factory
QPixmap::fromMimeSource: Cannot find pixmap "prevMark.xpm" in the mime source factory
QPixmap::fromMimeSource: Cannot find pixmap "mark.xpm" in the mime source factory
QPixmap::fromMimeSource: Cannot find pixmap "collide.xpm" in the mime source factory
QPixmap::fromMimeSource: Cannot find pixmap "translateTool.xpm" in the mime source factory
QPixmap::fromMimeSource: Cannot find pixmap "rotateTool.xpm" in the mime source factory
QPixmap::fromMimeSource: Cannot find pixmap "selectTool.xpm" in the mime source factory

Is this the cause of the error?

ambl2357, 2018-04-05:
my misstake sorry

xyaoab, 2018-06-27:
Hey, can I know how you solved it?

JenniferBuehler, 2018-07-03:
I will also have a look if I can reproduce this, but won't get the time before a while.

Meanwhile you can also just use the graspit simulator from the original repository to open the files.

m-asahi, 2018-07-11:
As a quick fix, copy all the image files in catkin_ws/build/graspit-pkgs/grasp_planning_graspit/graspit_external/repo/graspit/src/images to catkin_ws and always rosrun from catkin_ws.

At first viewing, grasp_planning_graspit/graspit_external/repo/graspit/CMakeList.txt under the build folder generates grasp_planning_graspit/graspit_external/graspit-build/qt_image_collection.cpp, but there seems to be no code accessing the staticImages instance defined here.

JenniferBuehler, 2018-07-19:
This is not relevant any more. The source has just been changed, now the graspit needs to be installed separetely from the original respository.
Please update my packages you are using and refer to the updated wiki for installation instructions (not much has changed, you will only need to install graspit separately, then re-build your catkin workspace, and the graspit simulator is then just started with ``graspit_simulator``).

closed 2018-07-19 by JenniferBuehler