JenniferBuehler/graspit-pkgs #34: Opening world loads the gripper unsuccessfully

joaocabogon, 2018-05-28:
After launching the urdf with the urdf_viewer package (and applying the mesh_rotation) every link is in the correct spot. 
But after using urdf2graspit the gripper looks like "unassembled", is it possible that the error is related to the dofValues parameter? I have tried using almost every joints combinations 

[image]
[image]

joaocabogon, 2018-05-28:
Yes, eventually i used the urdf2graspit launch file like i did for the urdf_viewer, but the result was still bad.

JenniferBuehler, 2018-05-29:
Ok. When it works in the URDF viewer it only means that you got your rotations right, the viewer itself is still using the URDF. What urdf2graspit does is convert it to Denavit-Hartenberg coordinates, which is used by graspit. So it can look entirely different there if the DH parameters aren't right.

So maybe the problem is somewhere in the DH Parameters computation, your robot may have some special case that hasn't been considered yet. Is there a chance you can share your source repo so I can have a look if I can find the issue? Or is your stuff private?

joaocabogon, 2018-05-29:
Well, more or less. I can share the folder with the meshes and the xacro file of the arm (please ignore the urdf file because it was not correctly generated).

[cyton_gamma_1500_description.zip](https://github.com/JenniferBuehler/graspit-pkgs/files/2047826/cyton_gamma_1500_description.zip)

joaocabogon, 2018-05-29:
I also noticed that changing the values of the visual_corr_axis in the urdf2graspit.launch leads to the same exact and incorrect display of the gripper when I open the world in the graspit interface.

JenniferBuehler, 2018-06-02:
Thanks for sharing the file, the meshes and xacro file is all I need.

I'll try to get some time early next week to look at it, sorry that I can't make it earlier, pretty flat out over the weekend.

JenniferBuehler, 2018-06-07:
Hi, I'm starting to look at this now, sorry it took a little longer.
May I ask which rosrun command you were using, or did you make a launch file?

I'm getting this in the contacts selector, is this how the robot is meant to look like? Just to confirm I'm testing the right thing ;) It looks like your screenshot above so I'm assuming it's OK like that.

[image]

I strangely get graspit crashes now, I probably need to check the graspit fork again and sync it with the most recent version. Will try to do this over the weekend and hopefully will be able to fix your problem.
Or is it not relevant any more? Did you have any updates on this from your side?

joaocabogon, 2018-06-10:
I still haven't tried to select the contacts, but the first launch file looks very similar to what I have tried. I have also tried using different arguments for the hand_root_link and the finger_joint_names. 

The only way I managed to open the model on Graspit interface (opening means I could visualize the gripper on the GUI but it was still "unassembled") was by launching the graspit_interface.lauch first and then on the top left side clicking on the open option and selecting the world that included my model. I guess if you try this maybe graspit won't crash when loading. 

This is the launch file I used:

    <arg name="urdf_file" default="/home/joao/ros_ws/src/mbot_description/xacro/testembot.urdf"/>
    <arg name="output_directory" default="/home/joao/ros_ws/src/graspit/models/robots/teste11"/>
     
    
    <arg name="hand_root_link" default="left_arm_Motor_+_Base"/>

    
    <arg name="finger_joint_names" default="left_arm_motor_base_to_left_link_back left_arm_motor_base_to_right_link_back"/>

    
    <arg name="scale_factor" default="" />
    
    
    <arg name="negate_joint_movement" default="false"/>
    
    <arg name="output_material" default="plastic"/>

    <arg name="launch_prefix" default=""/>

    
    <arg name="visual_corr_axis_x" default="1"/>
    <arg name="visual_corr_axis_y" default="0"/>
    <arg name="visual_corr_axis_z" default="0"/>
    <arg name="visual_corr_axis_angle" default="90"/>

    <node name="urdf2graspit_node" pkg="urdf2graspit" type="urdf2graspit_node" respawn="false"
      args="$(arg urdf_file) $(arg output_directory) $(arg hand_root_link) $(arg finger_joint_names)"
      output="screen" launch-prefix="$(arg launch_prefix)">

      <param name="output_material" value="$(arg output_material)"/>
      <param name="scale_factor" value="$(arg scale_factor)"/>
      <param name="negate_joint_movement" value="$(arg negate_joint_movement)"/>

      <param name="visual_corr_axis_x" value="$(arg visual_corr_axis_x)"/>
      <param name="visual_corr_axis_y" value="$(arg visual_corr_axis_y)"/>
      <param name="visual_corr_axis_z" value="$(arg visual_corr_axis_z)"/>
      <param name="visual_corr_axis_angle" value="$(arg visual_corr_axis_angle)"/>
    </node>

It is not an urgent matter, but I would like to fix this in the next few days so I can move forward.
Once again thanks for your efforts!

joaocabogon, 2018-06-28:
any updates on this?

JenniferBuehler, 2018-07-03:
Sorry, I'm still away on holiday. It's on my TODO list for checking as soon as possible.

JenniferBuehler, 2019-10-29:
So sorry, this completely fell under the radar. Are there any updates on this? Would you still like me to look at this more closely?

JenniferBuehler, 2020-12-07:
Closing due to inactivity. Please re-open if still relevant.

closed 2020-12-07 by JenniferBuehler