JenniferBuehler/graspit-pkgs #35: How to save a screenshot together with iv and xml files

mkokic, 2018-09-11:
Hi! I've managed to run 
**roslaunch grasp_planning_graspit_ros graspit_planner_jaco_example.launch \
    results_output_directory:=<your-results-directory>**
and save the inventor and world files. However, I would also like to also save a "screenshot" of the scene - similar to Save Image when running  ### graspit_interface. I've tried adding a function to ### GraspitSceneManager that does something like:
**core->getIVmgr()->saveImage(filename.c_str());**
The code can compile but it breaks when I try to execute it. Do you have any idea how to do this? Thanks a lot!

JenniferBuehler, 2018-09-13:
Hi, this should be possible, can you be more specific about how it "breaks", is it a segmentation fault at the call of `saveImage`?

JenniferBuehler, 2019-03-01:
Since I haven't heard back, I presume you have fixed this and will close the issue. Please re-open if you keep having problems.

closed 2019-03-01 by JenniferBuehler