JenniferBuehler/graspit-pkgs #38: Boost version conflict causes build error

liruiw, 2019-02-01:
Hello, I'm interested in using this package on other robot arms. 

My machine is on 16.04 with two boost versions 1.58 (system) and 1.66 (from source). It seems that in order to use the graspit_planning package, I would need at least 1.66. But rospack uses 1.58 by default in system ros installation. Namely, I won't be able to catkin_make successfully when using 1.58 solely. If I build with 1.66 and ignore the boost warnings, rospack will throw segfault when I run rosrun urdf2graspit urdf2graspit_node. So, I'm wondering if there's a workaround for this issue.

JenniferBuehler, 2019-02-03:
I guess this got resolved then?

liruiw, 2019-02-03:
I still don't know why there's a boost version conflict in my system. I end up hard coding the absolute path to urdf as a workaround. Now I try to use eigengrasp planner to store some grasp poses for hand. Is it normal to take 10 seconds to generate some valid grasps? The robot hand has two finger joints with 36 contact points generated from Graspit.

JenniferBuehler, 2019-02-03:
Hi, yes it is normal that GraspIt takes a while to compute. It has to search quite a large space so that takes a while. If you want it faster (at the cost of results which aren't as good) then you can decrease the max number of iterations. See also the Wiki.

liruiw, 2019-02-04:
Thank you for your response. It seems that the boost version conflict also prevents me from using the ros service tutorial. The exact problem that I had is 
grasp/devel/lib/libgrasp_planning_graspit.so: undefined reference to `boost::this_thread::hidden::sleep_until_realtime(timespec const&). Have you encountered this problem before?
Also in the api tutorial, GraspItSceneManagerNoGui should be updated as GraspItSceneManagerHeadless, right?

JenniferBuehler, 2019-02-07:
Hi,
I will try to reproduce this on the weekend or the coming week, sorry I won't get around to this earlier.

And yes you're right about the correction in the API tutorial! Will also update this.

JenniferBuehler, 2019-02-12:
Hi,
I just had a quick look at this again. I don't have Ubuntu Bionic 18.04 on my current system, I was thinking to upgrade my Xenial install but I first need to wrap up some work on it.

Meanwhile, one general thought. There is no boost version requirement for this package, the currently installed version should be OK. I'm guessing that because you have two versions installed, it takes the newest one, and that doesn't work.

Try the following: Put the version number in the CMakeLists.txt:

In `grasp_planning_graspit/cmake/grasp_planning_graspitConfig.cmake` and in
`/grasp_planning_graspit/CMakeLists.txt` (maybe you also need `grasp_planning_graspit_msgs/CMakeLists.txt` but try without first)
replace the following line
```cpp
find_package(Boost REQUIRED COMPONENTS<...>
```
with
```cpp
find_package(Boost REQUIRED 1.5.8 COMPONENTS<...>
```

If you get errors, you may need to add `date_time` to the components of that same command.

Can you try this and let me know if it worked?

liruiw, 2019-02-18:
Hello,
I've tried specified boost version 1.58 and add date_time to the component. However, the problem still exists. Since I'm able to compile the package with my 1.66 version and almost finish all I need for this package, I would be closing this issue (It might just be my local system's problem). Thank you for providing such a useful package and offering help.

closed 2019-02-18 by liruiw