JenniferBuehler/graspit-pkgs #39: Null pointer Issue

usman151mrd, 2019-02-03:
I faced the following Error in Ubuntu 18.04 and ROS melodic

^~~~~~~
/usr/include/urdf_model/model.h:92:8: note:   no known conversion for argument 2 from ‘urdf_traverser::LinkPtr {aka boost::shared_ptr<urdf::Link>}’ to ‘urdf::LinkSharedPtr& {aka std::shared_ptr<urdf::Link>&}’
/home/usman/Desktop/MSCS/jenifer/graspit_ws/src/urdf-tools-pkgs/urdf_traverser/src/UrdfTraverser.cpp: In member function ‘urdf_traverser::LinkConstPtr urdf_traverser::UrdfTraverser::readChildLink(const JointConstPtr&) const’:
/home/usman/Desktop/MSCS/jenifer/graspit_ws/src/urdf-tools-pkgs/urdf_traverser/src/UrdfTraverser.cpp:144:53: error: no matching function for call to ‘urdf::Model::getLink(const string&, urdf_traverser::LinkPtr&)’
     model->getLink(joint->child_link_name, childLink);
                                                     ^
In file included from /opt/ros/melodic/include/urdf/model.h:42:0,
                 from /home/usman/Desktop/MSCS/jenifer/graspit_ws/src/urdf-tools-pkgs/urdf_traverser/include/urdf_traverser/Helpers.h:47,
                 from /home/usman/Desktop/MSCS/jenifer/graspit_ws/src/urdf-tools-pkgs/urdf_traverser/src/UrdfTraverser.cpp:31:
/usr/include/urdf_model/model.h:52:22: note: candidate: urdf::LinkConstSharedPtr urdf::ModelInterface::getLink(const string&) const
   LinkConstSharedPtr getLink(const std::string& name) const
                      ^~~~~~~
/usr/include/urdf_model/model.h:52:22: note:   candidate expects 1 argument, 2 provided
/usr/include/urdf_model/model.h:92:8: note: candidate: void urdf::ModelInterface::getLink(const string&, urdf::LinkSharedPtr&) const
   void getLink(const std::string& name, LinkSharedPtr &link) const
        ^~~~~~~
/usr/include/urdf_model/model.h:92:8: note:   no known conversion for argument 2 from ‘urdf_traverser::LinkPtr {aka boost::shared_ptr<urdf::Link>}’ to ‘urdf::LinkSharedPtr& {aka std::shared_ptr<urdf::Link>&}’
/home/usman/Desktop/MSCS/jenifer/graspit_ws/src/urdf-tools-pkgs/urdf_traverser/src/UrdfTraverser.cpp: In member function ‘urdf_traverser::JointPtr urdf_traverser::UrdfTraverser::getParentJoint(const JointConstPtr&)’:
/home/usman/Desktop/MSCS/jenifer/graspit_ws/src/urdf-tools-pkgs/urdf_traverser/src/UrdfTraverser.cpp:150:45: error: conversion from ‘urdf::LinkConstSharedPtr {aka std::shared_ptr<const urdf::Link>}’ to non-scalar type ‘urdf_traverser::LinkConstPtr {aka boost::shared_ptr<const urdf::Link>}’ requested
     LinkConstPtr parentLink = model->getLink(joint->parent_link_name);
                               ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~
/home/usman/Desktop/MSCS/jenifer/graspit_ws/src/urdf-tools-pkgs/urdf_traverser/src/UrdfTraverser.cpp:152:24: error: could not convert ‘parentLink.boost::shared_ptr<const urdf::Link>::operator->()->urdf::Link::parent_joint’ from ‘const JointSharedPtr {aka const std::shared_ptr<urdf::Joint>}’ to ‘urdf_traverser::JointPtr {aka boost::shared_ptr<urdf::Joint>}’
     return parentLink->parent_joint;
            ~~~~~~~~~~~~^~~~~~~~~~~~
/home/usman/Desktop/MSCS/jenifer/graspit_ws/src/urdf-tools-pkgs/urdf_traverser/src/UrdfTraverser.cpp: In member function ‘urdf_traverser::JointConstPtr urdf_traverser::UrdfTraverser::readParentJoint(const JointConstPtr&) const’:
/home/usman/Desktop/MSCS/jenifer/graspit_ws/src/urdf-tools-pkgs/urdf_traverser/src/UrdfTraverser.cpp:157:45: error: conversion from ‘urdf::LinkConstSharedPtr {aka std::shared_ptr<const urdf::Link>}’ to non-scalar type ‘urdf_traverser::LinkConstPtr {aka boost::shared_ptr<const urdf::Link>}’ requested
     LinkConstPtr parentLink = model->getLink(joint->parent_link_name);
                               ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~
/home/usman/Desktop/MSCS/jenifer/graspit_ws/src/urdf-tools-pkgs/urdf_traverser/src/UrdfTraverser.cpp:159:24: error: could not convert ‘parentLink.boost::shared_ptr<const urdf::Link>::operator->()->urdf::Link::parent_joint’ from ‘const JointSharedPtr {aka const std::shared_ptr<urdf::Joint>}’ to ‘urdf_traverser::JointConstPtr {aka boost::shared_ptr<const urdf::Joint>}’
     return parentLink->parent_joint;
            ~~~~~~~~~~~~^~~~~~~~~~~~
/home/usman/Desktop/MSCS/jenifer/graspit_ws/src/urdf-tools-pkgs/urdf_traverser/src/UrdfTraverser.cpp: In member function ‘bool urdf_traverser::UrdfTraverser::hasChildLink(const LinkConstPtr&, const string&) const’:
/home/usman/Desktop/MSCS/jenifer/graspit_ws/src/urdf-tools-pkgs/urdf_traverser/src/UrdfTraverser.cpp:168:48: error: conversion from ‘const value_type {aka const std::shared_ptr<urdf::Link>}’ to non-scalar type ‘urdf_traverser::LinkPtr {aka boost::shared_ptr<urdf::Link>}’ requested
         LinkPtr childLink = link->child_links[i];
                                                ^
/home/usman/Desktop/MSCS/jenifer/graspit_ws/src/urdf-tools-pkgs/urdf_traverser/src/UrdfTraverser.cpp: In member function ‘int urdf_traverser::UrdfTraverser::traverseTreeTopDown(const LinkPtr&, boost::function<int(std::shared_ptr<urdf_traverser::RecursionParams>&)>, urdf_traverser::RecursionParamsPtr&, bool, unsigned int)’:
/home/usman/Desktop/MSCS/jenifer/graspit_ws/src/urdf-tools-pkgs/urdf_traverser/src/UrdfTraverser.cpp:202:78: error: conversion from ‘std::vector<std::shared_ptr<urdf::Link> >::iterator {aka __gnu_cxx::__normal_iterator<std::shared_ptr<urdf::Link>*, std::vector<std::shared_ptr<urdf::Link> > >}’ to non-scalar type ‘std::vector<boost::shared_ptr<urdf::Link> >::const_iterator {aka __gnu_cxx::__normal_iterator<const boost::shared_ptr<urdf::Link>*, std::vector<boost::shared_ptr<urdf::Link> > >}’ requested
     for (std::vector<LinkPtr>::const_iterator child = link->child_links.begin();
                                                       ~~~~~~~~~~~~~~~~~~~~~~~^~
/home/usman/Desktop/MSCS/jenifer/graspit_ws/src/urdf-tools-pkgs/urdf_traverser/src/UrdfTraverser.cpp:203:19: error: no match for ‘operator!=’ (operand types are ‘std::vector<boost::shared_ptr<urdf::Link> >::const_iterator {aka __gnu_cxx::__normal_iterator<const boost::shared_ptr<urdf::Link>*, std::vector<boost::shared_ptr<urdf::Link> > >}’ and ‘std::vector<std::shared_ptr<urdf::Link> >::iterator {aka __gnu_cxx::__normal_iterator<std::shared_ptr<urdf::Link>*, std::vector<std::shared_ptr<urdf::Link> > >}’)
             child != link->child_links.end(); child++)
             ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/x86_64-linux-gnu/c++/7/bits/c++allocator.h:33:0,

JenniferBuehler, 2019-02-03:
Hello, yes I haven't added support for Melodic and Gazebo 9 yet, I hope I'll get around to do this in the near future.

See also [this PR](https://github.com/JenniferBuehler/gazebo-pkgs/pull/23) (which unfortunately was not completed, see comments there). This may help you to fix it. If you do end up fixing this, I'm happy about PR's.

JenniferBuehler, 2019-03-01:
You can try again now, all my master branches have been updated and tested for Melodic.
I will close this for now, please re-open if you keep having the issue.

closed 2019-03-01 by JenniferBuehler