JenniferBuehler/graspit-pkgs #45: Error while loading world .xml file into the graspit_simulator

artemiialessandrini, 2019-05-03:
I run graspit_simulator, add my world file and gui crashes with:
```
Creating base...

Setting up 6 degrees of freedom...
Creating 2 kinematic chains (fingers)...
Chain 0:   Creating joints
   Joint 0
   Joint 1
  Creating links
   Link 0
   Link 1
  Creating dynamic joints
Chain 1:   Creating joints
   Joint 0
   Joint 1
  Creating links
   Link 0
   Link 1
  Creating dynamic joints
Segmentation fault (core dumped)
```

I am having a UR5e and Robotiq Gripper, mounted on the table. 
Gripper RobotIQ85-2F mesh is the one from [Stanley Innovation Robotiq 85 Visual Models](https://github.com/waypointrobotics/robotiq_85_gripper), all the urdf2graspit preparations from graspit-pkgs are ok.

artemiialessandrini, 2019-05-03:
[grasp_format.zip](https://github.com/JenniferBuehler/graspit-pkgs/files/3140996/grasp_format.zip) my models and worlds files I get

JenniferBuehler, 2019-05-06:
Hi, thanks for reporting this.
It was a stupid mistake with the continuous joints which were categorized as prismatic by mistake.
Please pull the repo again and try again.

I used the following launch file:

```xml
<launch>                                                                        
    # the input urdf file                                                       
    <arg name="urdf_file" default="$(find robotiq_85_description)/urdf/robotiq_85_gripper.urdf"/>
                                                                                
    # the directory onto which to write the output files                        
    <arg name="output_directory"/>                                              
                                                                                
    <arg name="hand_root_link" default="gripper_base_link"/>                    
    <arg name="finger_joint_names" default="gripper_finger1_joint gripper_finger1_inner_knuckle_joint gripper_finger2_joint gripper_finger2_inner_knuckle_joint"/>
    <arg name="scale_factor" default="1000"/>                                   
                                                                                
    <arg name="output_material" default="plastic"/>                             
                                                                                
    <include file="$(find urdf2graspit)/launch/urdf2graspit.launch">            
        <arg name="urdf_file" value="$(arg urdf_file)"/>                        
        <arg name="output_directory" value="$(arg output_directory)"/>          
        <arg name="hand_root_link" value="$(arg hand_root_link)"/>              
        <arg name="finger_joint_names" value="$(arg finger_joint_names)"/>      
        <arg name="scale_factor" value="$(arg scale_factor)"/>                  
        <arg name="output_material" value="$(arg output_material)"/>            
        <arg name="negate_joint_movement" value="false"/>                       
    </include>                                                                  
</launch>  
```

I haven't checked whether the transforms make sense because I don't know the robot, can you please report if there's anything else weird going on.

Cheers
Jennifer

JenniferBuehler, 2019-05-06:
By the way, I noticed that the robot has "mimic" joints, I don't think this is supported by GraspIt, so not sure how your results would turn out...

artemiialessandrini, 2019-05-09:
You're right, it's not supported.
I turned off the gazebo_plugin, graspit model got created, but that's what happened:
(left fingers got into collision)
[image]
[image]

Basically, there's not a single grasp created as well.
Terminal says:
` Collision found when looking for contacts`

May I get some tips about this? Is there any solution to avoid mimic_joints, do implement robotiq85 and arm  in gazebo simulation using moveit_ with+kinect and graspit!?

Or shell we make another thread regarding mimic_joints

JenniferBuehler, 2019-05-15:
You may have to tune your eigengrasps (see also [the wiki](https://github.com/JenniferBuehler/graspit-pkgs/wiki/urdf2graspit#eigengrasp-notes)), the generated values in `models/robots/<your robot>/eigen/eigen.xml` are just a guess, in most cases you have to edit them yourself. I think may be possible to handle mimic joints if you define the eigengrasps accordingly, but I haven't done this yet. Maybe you can ask in the graspit repository directly, they should be able to help you better with this. Not sure if [this issue](https://github.com/graspit-simulator/graspit/issues/126) may help, I haven't read through it all though.

majiqiang, 2019-05-23:
> You're right, it's not supported.
> I turned off the gazebo_plugin, graspit model got created, but that's what happened:
> (left fingers got into collision)
> [image]
> [image]
>
> Basically, there's not a [...]
嗨！你好，我看你的手爪是robotIQ的，跟我们的很像，我现在也是在想着如何导入3d模型到graspit，但是不知道如何下手，目前只有手的3D SolidWorks模型，我应该怎么做呢？谢谢！

JenniferBuehler, 2019-05-24:
Please see also [this issue](https://github.com/JenniferBuehler/graspit-pkgs/issues/47#issuecomment-494245613), maybe it helps for your robot configuration file as well?

Sorry at the moment I have very limited time to look into this, maybe next week I can look at this more closely.

majiqiang, 2019-05-24:
> Please see also [this issue](https://github.com/JenniferBuehler/graspit-pkgs/issues/47#issuecomment-494245613), maybe it helps for your robot configuration file as well?
> 
> Sorry at the moment I have very limited time to look into this, maybe next week I can look at this more closely.

Thank you very much! I am looking forward to your reply!

JenniferBuehler, 2019-10-29:
Have you had any more insights on this, did you check the issue I linked? If this is not relevant any more, let me know or just close the issue please :)

artemiialessandrini, 2019-10-29:
Hi, let me double-check this week, after changes made

artemiialessandrini, 2019-11-05:
Initial issue solved long time ago
Closed.

closed 2019-11-05 by artemiialessandrini

ItzMeJP, 2019-11-19:
Hi, 
I am trying to generate the same gripper with the same launch file described here by @JenniferBuehler. When I insert the generated .xml in Graspit, I get the following:

[image]

Could you help me? Maybe  @artemiialessandrini can attached your functional model of the gripper, please?

artemiialessandrini, 2019-11-20:
@ItzMeJP , yes, I faced this issue before
Please, refer to [mesh_rotation](https://github.com/JenniferBuehler/urdf-tools-pkgs/wiki/mesh_rotation) tool wiki page
As I remember, you need to change angle arg such 
 `<arg name="visual_corr_axis_angle" default="90"/>`
There is a great wiki explanation, good luck!

ItzMeJP, 2019-11-20:
@artemiialessandrini thank you for the help. I corrected the visual .ivs files as you can see below in the urdf_view tool:

[image]

However, when I add these new .ivs files inside the model in Graspit I get the following:

[image]

Is there any configuration to set? I tried to figure out but I can't.

artemiialessandrini, 2019-11-21:
Hi, @ItzMeJP

> I corrected the visual .ivs files as you can see below in the urdf_view tool:

If I understood your issue right, `urdf_viewer` is only for trying to rotate visuals, basically you "checked" the right way, but didn't apply it. To apply please refer to [urdf2inventor wiki (below)](https://github.com/JenniferBuehler/urdf-tools-pkgs/wiki/mesh_rotation). 
By changing args, just as you did in `urdf_viewer`, you can change an actual meshes orientation.

If it won't help, please, also take a look to #34.

iretiayo, 2019-11-21:
Hi,
I'm also running into issues when generating graspit files for this hand.

When I visualize the urdf, I get the following:
[image]

I tried using the [mesh_rotatation](https://github.com/JenniferBuehler/urdf-tools-pkgs/wiki/mesh_rotation) and tried different values for `visual_corr_axis_angle` but none seem to fix it.

@artemiialessandrini @ItzMeJP ,
What steps did you take to convert the urdf for this robot hand to graspit files?

iretiayo, 2019-11-23:
To answer my own question, I was using a robot description from a different [repo](git@github.com:ros-industrial/robotiq.git). When I switched to the [one](https://github.com/waypointrobotics/robotiq_85_gripper) that @artemiialessandrini mentioned above, I'm able generate a consistent hand in GraspIt!.

ItzMeJP, 2019-11-25:
> Hi, @ItzMeJP
>
>
> If I understood your issue right, `urdf_viewer` is only for trying to rotate visuals, basically you "checked" the right way, but didn't apply it. To apply please refer to [...]

Thank you. It works now. However, I have the contacts problem of the external knuckle link with the finger link... Did you solve that  @artemiialessandrini ? My solution was to remove that knuckle link, but I would like to get a more reliable model.

artemiialessandrini, 2019-11-27:
Hi, @ItzMeJP 
Does your model look like my, mentioned above?
I assume, it's due to usage of mimic joints.

> You're right, it's not supported.
> I turned off the gazebo_plugin, graspit model got created, but that's what happened:
> (left fingers got into collision)
> [image]
> [image]
>
> Basically, there's not a [...]

ItzMeJP, 2019-11-28:
@artemiialessandrini I removed all gazebo plugins, including the mimic joints setup. However, the Graspit calculate the collisions not only between objects and the robot but between the joints chain and the objects. Therefore, the generate .xml creates 2 chains that collide each other. We can see that with the image below:

[image]

 Probably, a solution is to generate the .xml file with only one joints chain, but this is very annoying since the graspit does not allow the creation of parents links with more than one child, and then, we have to manually create the link between the inner knuckle and the finger link (using DH is much more laborious).

My solution was to remove the inner knuckles links as follow, however, this approach is not so reliable.

[image]