JenniferBuehler/graspit-pkgs #47: urdf2graspit fails to generate wsg_50 contacts/fingers

mattcorsaro1, 2019-05-15:
Hi,

I'm trying to use the Schunk WSG 50 gripper in GraspIt. [nalt/wsg50-ros-pkg](https://github.com/nalt/wsg50-ros-pkg) contains a URDF of the hand that I'm trying to convert. I'm running Ubuntu 16.04 with ROS Kinetic and using the latest convenience-pkgs, graspit-pkgs. and urdf-tools-pkgs.

This conversion command seems to generate the GraspIt configuration correctly:
`rosrun urdf2graspit urdf2graspit_node src/wsg50-ros-pkg/wsg_50_simulation/urdf/wsg_50.urdf schunk_wsg_50 base_link base_joint_gripper_left base_joint_gripper_right`

The joints that the traverser finds are:

> base_joint_gripper_left
> base_joint_gripper_right
> guide_joint_finger_left
> guide_joint_finger_right

My generated configuration contains: `eigen  iv  wsg_50.xml`

When I try to load the model into GraspIt, the base of the hand appears, but the fingers don't. Here's the output from GraspIt:

> Creating base...
> 
> Setting up 2 degrees of freedom...
> Creating 2 kinematic chains (fingers)...
> Chain 0:   Creating joints
>    Joint 0
>   Creating links
>    Link 0
>   Creating dynamic joints
> Chain 1:   Creating joints
>    Joint 0
>   Creating links
>    Link 0
>   Creating dynamic joints
> DOF default value too large; setting to max
> DOF default value too small; setting to min
> No normalization data found; using factors of 1.0
> Read 1 eigengrasps from EG file
> Using eigengrasps from file: eigen/eigen.xml
> Could not open /home/matt/Software/graspit/models/robots/wsg_50/virtual/contacts.xml
> Failed to load virtual contacts from file virtual/contacts.xml

I also tried to generate a new URDF from wsg_50.urdf.xacro:
`rosrun xacro xacro --inorder src/wsg50-ros-pkg/wsg_50_simulation/urdf/wsg_50.urdf.xacro > src/wsg50-ros-pkg/wsg_50_simulation/urdf/new_wsg_50.urdf`

However, when I try to generate a GraspIt config using `rosrun urdf2graspit urdf2graspit_node src/wsg50-ros-pkg/wsg_50_simulation/urdf/new_wsg_50.urdf schunk_wsg_50 base_link base_joint_gripper_left base_joint_gripper_right`, I get this error:

> [ INFO] [1557961713.465940586]: URDF file: src/wsg50-ros-pkg/wsg_50_simulation/urdf/new_wsg_50.urdf
> [ INFO] [1557961713.466011676]: Output dir: schunk_wsg_50
> [ INFO] [1557961713.466043198]: Hand root: base_link
> [ INFO] [1557961713.466055359]: Joint: base_joint_gripper_left
> [ INFO] [1557961713.466069899]: Joint: base_joint_gripper_right
> [ INFO] [1557961713.466476938]: output_material: <plastic>
> [ INFO] [1557961713.467035813]: scale_factor: <1000.000000>
> [ INFO] [1557961713.467518663]: negate_joint_movement: <0>
> [ INFO] [1557961713.469093656]: Starting model conversion...
> [ INFO] [1557961713.469228798]: ### Loading from URDF file src/wsg50-ros-pkg/wsg_50_simulation/urdf/new_wsg_50.urdf...
> [ INFO] [1557961713.469360746]: Setting base model directory to /home/matt/ros/urdf/src/wsg50-ros-pkg/wsg_50_simulation/urdf
> [ERROR] [1557961713.469495546]: No link elements found in urdf file
> [ERROR] [1557961713.469534689]: Could not load model from XML string
> [ERROR] [1557961713.469543813]: Could not load file
> [ERROR] [1557961713.469553800]: Could not load file
> [ERROR] [1557961713.469569179]: Failed to process.

Please let me know if you need more info from me to reproduce this issue.
Thanks,
Matt

JenniferBuehler, 2019-05-17:
Hi, thanks for reporting this.

First, to your second problem (generate new URDF and then use it for urdf2graspit): `wsg_50_simulation/urdf/wsg_50.urdf.xacro` only defines a macro to generate the robot. You have to call this macro from a different .xacro file, e.g. like this:

```xml
<?xml version="1.0"?>                                                           
<robot  xmlns:sensor="http://playerstage.sourceforge.net/gazebo/xmlschema/#sensor"
  xmlns:controller="http://playerstage.sourceforge.net/gazebo/xmlschema/#controller"
  xmlns:interface="http://playerstage.sourceforge.net/gazebo/xmlschema/#interface"
        xmlns:xacro="http://ros.org/wiki/xacro"                                 
  name="wsg_50">                                                                
                                                                                
  <link name="robot_root">                                                      
  </link>                                                                       
                                                                                
  <xacro:include filename="$(find wsg_50_simulation)/urdf/wsg_50.urdf.xacro"/>  
                                                                                
  <xacro:wsg_50_xacro name="wsg" parent="robot_root" finger_length="0.023" finger_tip="0">
     <origin xyz="0 0 0" rpy="0 0 0" />                                         
  </xacro:wsg_50_xacro>                                                         
</robot>         
```

The file `wsg_50.urdf.xacro` is also a little old, you have to replace `insert_block` in line 14 with `xacro:insert_block` in order to make it work. Also, the joint and base link names have changed.
However I can confirm that I get the same behaviour with both files, though I would tend to use the freshly generated URDF because the existing one is quite old.

I get a lot of "Collision found when looking for contacts" messages after loading into graspit, which could be related to that the fingers aren't showing (maybe they're collapsed inside the base). Can you confirm that it's the same for you?

JenniferBuehler, 2019-05-17:
I found out why they can't be seen, the .STL mesh files must contain a transparency value which is set to fully transparent (or is read wrong by the mesh convertion to inventor files).
As a quick fix, you can replace all `transparency 1` values by `transparency 0` in the following files:
`<your-directory>/models/robots/wsg_50/iv/wsg_gripper_right.iv` and
`<your-directory>/models/robots/wsg_50/iv/wsg_gripper_left.iv`

When you do these changes, the IV files display correct in `urdf_viewer`:
`rosrun urdf_viewer urdf_viewer_node  <your-directory>/models/robots/wsg_50/iv/wsg_gripper_right.iv --iv`
However they still don't show in graspit, I'm not quite sure why, I will look into it.

I will see if the problem with the transparency resides in the .STL files themselves, or if the mesh converter is doing something wrong. Will report back here once I get around to that.

JenniferBuehler, 2019-05-17:
Got it, the transforms of the fingers are wrong, they are very far away (you see them when you zoom out far). The scale factor of the meshes is also wrong. In the same files I mentioned above, replace all entries of `scaleFactor 1000 1000 1000` by `scaleFactor 1 1 1` and it will work.

Here is how it looks (I edited the colors for testing), does this look correct?

[image]

The Eigengrasp file has to be edited, which is probably the reason why you get all these initial collisions. If you open the EigenGrasp tool (like in screenshot) then you can separate them with the slider. The values are not optimal yet. The generated EigenGrasp file is just a template, you have to find good values for yours. See also [this wiki](https://github.com/JenniferBuehler/graspit-pkgs/wiki/urdf2graspit#eigengrasp-notes).

I will report back here when I find the problem with the mesh conversion (transparency and scale).

mattcorsaro1, 2019-05-18:
Awesome, thanks so much for your help! The model looks correct in your screenshot and when I load it in GraspIt.

However, when I try to open the hand with graspit_commander, I get this error:
Robot joint interpolation error
Interpolation failed!

JenniferBuehler, 2019-05-18:
Have you fine tuned your EigenGrasp file? It would probably be necessary for this hand.

If you are experiencing issues with graspit_commander, I suggest you ask the guys on the original graspit repository, they'll be able to help you better with that. I'm sure they will be happy if you share your model files with them too, they can help you get it up and running then :)

mattcorsaro1, 2019-05-21:
I tried tuning the EigenGrasp file, but I'm not sure what the expected behavior is. I read through the wiki you posted as well as [this one](https://graspit-simulator.github.io/build/html/eigengrasps.html), so I think I understand what the values in the eigen.xml do. It makes sense that there's only one EigenGrasp and two dimensions in the template file that was generated since it's a parallel jaw gripper.

When I open the EigenGrasp Interface menu when the gripper is fully closed, the initial slider value is -36.98. After adjusting the slider, I can't move it beyond the -8.00 to 8.00 range. Is this to be expected? 0 looks to be close to the middle as you say in the wiki, but -8.00 is not fully closed and 8.00 is not fully open. I've tried changing the DimVals for the EG and ORIGIN, but I can't fully open the gripper. The slider starting at -36.98 but getting stuck in the +-8 range seems like a bug in GraspIt. What do you think?

Adjusting one of the EG's DimVals so d0 != -1*d1 makes one gripper close more than the other, which isn't desired. Adjusting them both by the same amount doesn't seem to have any effect on how much they open or close, and I don't think that has anything to do with the slider getting stuck between +-8. Changing the DimVals of the origin does change the behavior. I can get 0 to be the closed position, but I can't seem to change how much the grippers open. I've also tried changing the EigenValue field, but as mentioned in the doc I linked, it doesn't seem to be used in the code.

Sounds good, I'll create an issue on the main GraspIt package. I got the same error with the openclose plugin example, so the issue isn't caused by graspit_commander. A fork with my wsg_50 model is [here](https://github.com/mattcorsaro1/graspit/tree/wsg_50).

Thank you!

mattcorsaro1, 2019-05-24:
One of the problems with using GraspIt's openclose plugin with this model was that the fingers were initially in collision with each other. I increased the minimum joint values in [my model](https://github.com/mattcorsaro1/graspit/blob/wsg_50/models/robots/wsg_50/wsg_50.xml#L46), and that's fine.

The problem now is that the right finger's closing direction is wrong. When opening, GraspIt sends the fingers to the minimum joint values. This is fine for the left finger because the minimum value is -55 (open), but the minimum value for the right finger is 6 (closed). Is there a way to reverse the closing direction in the model? I've tried swapping the max and min values (didn't work because GraspIt swaps them back if min > max) in the right gripper's chain, as well as changing the transform and d.

If you have any advice, I'd greatly appreciate it! Thanks.

JenniferBuehler, 2019-05-24:
Great you've made some progress on this! :)

Have you tried to invert the signs in the EigenGrasp configuration?

Sorry I have very limited time this week as I'm moving house, hopefully I'll get a moment next week to look at this more closely.

mattcorsaro1, 2019-05-29:
I had tried inverting the signs in the EigenGrasp configuration, but that didn't do anything. I fixed it by inverting the sign on one of the defaultVelocities. Thanks for your help :)

JenniferBuehler, 2019-06-04:
Ok, good to know, thanks for the update! I'm glad you could fix it.

closed 2019-06-04 by JenniferBuehler