JenniferBuehler/graspit-pkgs #48: energy 0

TongWuEric, 2019-05-21:
Hi Jennifer,
When I try to run simple_planning_tutorial.cpp with other world like dlr_flask.xml instead of jaco, I always get the following results:

=== EigenGraspPlanner update === - EigenGraspPlanner.cpp, 1042
Current Step: 65000 - EigenGraspPlanner.cpp, 1064
Energy: 0 - EigenGraspPlanner.cpp, 1152
=== EigenGraspPlanner update === - EigenGraspPlanner.cpp, 1042
Current Step: 65100 - EigenGraspPlanner.cpp, 1064
Energy: 0 - EigenGraspPlanner.cpp, 1152
=== EigenGraspPlanner update === - EigenGraspPlanner.cpp, 1042
Current Step: 65200 - EigenGraspPlanner.cpp, 1064
Energy: 0 - EigenGraspPlanner.cpp, 1152
Neighbor gen loops: 11
=== EigenGraspPlanner update === - EigenGraspPlanner.cpp, 1042
Current Step: 65300 - EigenGraspPlanner.cpp, 1064
Energy: 0 - EigenGraspPlanner.cpp, 1152
=== EigenGraspPlanner update === - EigenGraspPlanner.cpp, 1042
Current Step: 65400 - EigenGraspPlanner.cpp, 1064
Energy: 0 - EigenGraspPlanner.cpp, 1152
Neighbor gen loops: 11
=== EigenGraspPlanner update === - EigenGraspPlanner.cpp, 1042
Current Step: 65500 - EigenGraspPlanner.cpp, 1064
Energy: 0 - EigenGraspPlanner.cpp, 1152

Why it only shows energy 0? Thanks!

JenniferBuehler, 2019-05-22:
The contacts for the DLR hand are not loaded by default. I have previously used `models/robots/DLR/virtual/phlx_distal.xml`. With `graspit_simulator` you also have to load them manually.
For the tutorial, you should put the loading of contacts into the robot xml file `models/robots/DLR/DLR.xml`. You can do this with the following tag

```xml
<virtualContacts>virtual/phlx_distal.xml</virtualContacts>
```

Let me know if it works after you include the contacts.

TongWuEric, 2019-05-22:
Yes you are right! It works! Thanks!

JenniferBuehler, 2019-05-22:
Great :)

closed 2019-05-22 by JenniferBuehler