JenniferBuehler/graspit-pkgs #50: GWS projection

TongWuEric, 2019-05-23:
Hi when I want to get the GWS projection like the paper Fig1:
[image]
Fig 1 paper image
I only get an empty result like Fig2 either I fix tx,ty,tz or fx,fy,fz with Fig3 config:
[image]
                                                    Fig2 my result
[image]
                                            Fig3 projection parameters
What I did was: 
1.  run simple_planning_tutorial.cpp with dlr_flask.xml (already added contact tag);
2.  open generated world in simulator and use GWS projection with the parameter mentioned above.

Is there a problem here? Thanks for your time and help!

JenniferBuehler, 2019-05-24:
Hi, I think regarding this issue the guys on the graspit repo should be able to help you better. I'll have to try it out with the same model files and see what the issue is, unfortunately this week I won't have time for that - hopefully next week. If you make any progress on this before that, please report back here!

You could also try to use the graspit simulator instead of the simple_planning_tutorial.cpp, doing the same as you did for the DLR hand. That should do the same. If not, let me know because then there may be an issue in the planning tutorial or the code in this package here.

Sorry I can't be of any more help in the short term, it's very busy around here at the moment.

TongWuEric, 2019-06-27:
Hi sorry for the delay, just an update: the GWS can work when the fingers all actually contact the object. Thanks!

closed 2019-06-27 by TongWuEric