JenniferBuehler/graspit-pkgs #51: error: urdf to robot.xml

majiqiang, 2019-06-21:
why show this error
[image]
load robot.xml failed.
this is my urdf model and robot.xml.
please help me!
[robots.zip](https://github.com/JenniferBuehler/graspit-pkgs/files/3314597/robots.zip)
[today615.zip](https://github.com/JenniferBuehler/graspit-pkgs/files/3314600/today615.zip)

JenniferBuehler, 2019-06-27:
Sorry I don't know what this says. It may be a segmentation fault? Which robot are you using?

majiqiang, 2019-06-29:
> Sorry I don't know what this says. It may be a segmentation fault? Which robot are you using?

closed 2019-06-29 by majiqiang

majiqiang, 2019-06-29:
> Sorry I don't know what this says. It may be a segmentation fault? Which robot are you using?

Thanks for your reply, the robot I'm using is usual, this is its picture and 3D mesh file. I use the plugin [sw_urdf_exporter](http://wiki.ros.org/sw_urdf_exporter) to get urdf file. Could you help me check if the urdf file is right?
[myurdf.tar.gz](https://github.com/JenniferBuehler/graspit-pkgs/files/3341953/myurdf.tar.gz)
[image]

JenniferBuehler, 2019-06-30:
I think the URDF is fine as long as you can view it correctly in RVIZ. I guess you are then using urdf2graspit to convert to graspit format?
If you could also translate what the error message says that would be helpful. I'm suspecting segmentation fault. I presume you are using the newest version of this repository and graspit?

majiqiang, 2019-07-01:
> I think the URDF is fine as long as you can view it correctly in RVIZ. I guess you are then using urdf2graspit to convert to graspit format?
> If you could also translate what the error message says [...]

Yes，I can view it in rviz, but it's a little wrong as this picture. I don't know how to form a closed loop chain in urdf. Maybe that plugin don't support the feature, I don't know.
I am using the newest version of this repository [urdf2graspit](https://github.com/JenniferBuehler/graspit-pkgs/wiki/Conversion-with-urdf2graspit)
[image]

JenniferBuehler, 2019-07-07:
Judging from your other issue, you managed to get rid of the segmentation fault?