JenniferBuehler/graspit-pkgs #53: New (?) transformations in graspit

gstavrinos, 2019-07-12:
Hello Jennifer,

I am not sure if this is the right place for this issue or the official graspit repo, but I will try here first, since this repository's response rate is much better!

I have developed and now maintaining this repo: https://github.com/gstavrinos/ez_pick_and_place, which uses graspit-pkgs as a basis to offer a friendly way to create pick and place tasks.

When I saw that you updated from household and manipulation msgs to moveit_msgs I did so with my package, to maintain compatibility. 

The problem is that after some months of not using my package, something else seems to be completely broken, making the package unusable. There seems to be a change in the transformation system of graspit (or graspit-pkgs), and now the grasping poses do not correspond to the same poses in ROS. Hence, even though the grasping poses seem fine in the graspit world, they are rotated in the ROS coordinate world, resulting in infeasible planning.

I have tried to dig into both graspit and graspit-pkgs for a striking change on transformation systems, but have not found anything too obvious.

Did you make such a change on graspit-pkgs or notice something like that in the graspit code base?

Thank you in advance, and sorry if I am spamming the wrong repo.

George

gstavrinos, 2019-07-12:
As an example, check out this image:

[image]

The expected candidate pose, should have the z axis (blue) pointing towards the object and not the y axis (green).

JenniferBuehler, 2019-07-13:
Hi George,

I haven't changed anything in the coordinate system.. And the new moveit_msgs are pretty much the same, so I doubt it's got to do with that. 
But I can look into it, at the moment I'm on holidays though, so I can't promise when I'll get around to try it out.

Meanwhile, if I understand it correctly, the output from graspit (specifically the grasp pose) is just rotated around the x axis by 90 degrees? A quick fix would be to correct that rotation right after it is returned from graspit. That would work until the actual reason this happens has been investigated. 

Thanks for reporting this. Let me know when there is any update on this, I'll post here as well once I get around to look at it. 

Cheers 
Jennifer

gstavrinos, 2019-07-13:
Hey Jennifer,

Thanks for the answer. There is a chance that there is a problem with a urdf and a gazebo plugin that causes the problem, so please don't bother with the it. At least not yet. I will keep you updated.

Thanks again,

George

JenniferBuehler, 2019-07-14:
Hi George,

Ok, let me know as soon as there's an update on this. 

Cheers 
Jennifer

gstavrinos, 2019-07-17:
Hey Jennifer,

It is definitely not a problem with your repo or graspit. I have cornered the issue to some urdf changes on parent repositories and kinematics libraries. I am closing this issue since it is not relevant.

Thanks for your time,

George

closed 2019-07-17 by gstavrinos

JenniferBuehler, 2019-07-17:
Ok, thanks for letting me know :)