JenniferBuehler/graspit-pkgs #54: i can't diplay 5 finger's shadow hand model on graspit_simulator.

masa1120, 2019-10-07:
Nice to meet you.
I challenged to display a five-finger shadow hand model on grasspit_simulator.
I read the getting started of "Step 1. Choose your robot model" on the following page.
https://github.com/JenniferBuehler/graspit-pkgs/wiki/Getting-started

However, an error message was displayed and could not be resolved.
Can you tell me how to deal with it?
The robot model file was attached into this issue page.

initial@adonis:~/automnp_ws/src/automnp_simulator/automnp_mujoco/graspit$ ./build/graspit_simulator -w shadowhand_motor_world
Creating base...

Setting up 22 degrees of freedom...
Creating 5 kinematic chains (fingers)...
Chain 0:   Creating joints
   Joint 0
   Joint 1
   Joint 2
   Joint 3
  Creating links
   Link 0
   Link 1
   Link 2
   Link 3
  Creating dynamic joints
Chain 1:   Creating joints
   Joint 0
   Joint 1
   Joint 2
   Joint 3
   Joint 4
  Creating links
   Link 0
   Link 1
   Link 2
Coin read error: Couldn't read value for field "translation" of Transform
	Occurred at line   9 in /home/initial/automnp_ws/src/automnp_simulator/automnp_mujoco/graspit/models/robots/shadowhand_motor/iv/rh_lfproximal.iv
A problem occurred while reading Inventor file/home/initial/automnp_ws/src/automnp_simulator/automnp_mujoco/graspit/models/robots/shadowhand_motor/iv/rh_lfproximal.iv
Failed to open geometry file: /home/initial/automnp_ws/src/automnp_simulator/automnp_mujoco/graspit/models/robots/shadowhand_motor/iv/rh_lfproximal.iv
Failed to load file for link 2
Failed to read chain 1
Failed to load robot from file
Deleting Hand: 
Deleting grasp
Deleted robot: 
Could not open /home/initial/automnp_ws/src/automnp_simulator/automnp_mujoco/graspit/models/robots/shadowhand_motor/shadowhand_motor.xml
Failed to Parse args.

masa1120, 2019-10-07:
This is the robot model which is composed by shadow hand and ur10 arm.
[shadowhand_with_ur10.zip](https://github.com/JenniferBuehler/graspit-pkgs/files/3696076/shadowhand_with_ur10.zip)

This is output folder made by urdf2graspit.launch(including urdf2graspit_node) and graspit_contacts_generator.launch 
[output.zip](https://github.com/JenniferBuehler/graspit-pkgs/files/3696075/output.zip)

This urdf2graspit_node could output following file.
[urdf2graspit.txt](https://github.com/JenniferBuehler/graspit-pkgs/files/3696095/urdf2graspit.txt)

In output\models\robots\shadowhand_motor\iv folder, rh_lfdistal.iv, rh_lfproximal.iv and rh_lfmiddle.iv have strange output within Transform variable which have elements of "nan".

When I compared the both axes of URDF(left in figure) and DH(right), the axes were not same.
Around the root of little finger, axes rotated a bit to different direction.
[image]

I could setup contact points without any trouble but the final output (shadowhand_motor_world.xml) of step1 was not displayed on graspit simulator.
[image]

JenniferBuehler, 2019-10-10:
Thanks for reporting this! I will try and look at it this weekend.

masa1120, 2019-10-11:
i use an official shadow hand urdf model ( included in shadowhand_with_ur10.zip) which was provided by shadow company. The urdf is used on GAZEBO or MOVEIT without any problem. When I use the urdf file, the first step of urdf2graspit seems to fail to convert urdf to graspit data format around the little finger. and then the latter step can not use the transform information. Indeedly, the converted xml has -NAN but not a specific value on the little finger transformation. In the urdf file, the dismal, middle and proximal link and joint information for little finger have completely same parameters as other fingers meaning first, middle and ring finger.  The joint "rh_LFJ5" which is root joint for little finger only has a bit twisted axis params comparing with other joint. As an example, I show a part of information of my urdf file.
   <joint name="rh_LFJ5" type="revolute">
     <parent link="rh_palm"/>
     <child link="rh_lfmetacarpal"/>
     <origin rpy="0 0 0" xyz="-0.033 0 0.02071"/>
     <axis xyz="0.573576436 0 0.819152044"/>
     <limit effort="1.0" lower="0" upper="0.785398163397" velocity="2.0"/>
     <dynamics damping="0.1"/>
   </joint>

JenniferBuehler, 2019-10-12:
Hello, I was just going to test it, but unfortunately the meshes don't come in your .zip file. I cannot find the official repository for that same file. Could you please provide the full directory `shadow_launch`?

masa1120, 2019-10-12:
I appreciate your reply. I pushed my packages with graspit modules on my repository. How to reproduce my error is written in README.md.  Link is below.
https://github.com/masa1120/graspit_sr_ur

JenniferBuehler, 2019-10-27:
Sorry it took me so long, I simply didn't have time in the past couple of weeks :/
I looked into it and I think I fixed it. At least there are no nan values any more, but I'm having separate issues with the graspit simulator at the moment (I can't load any worlds), so I couldn't open the converted files to check if it looks like it makes sense. Would you mind checking that? I'm not sure when I'll sort out the graspit simulator issue...

JenniferBuehler, 2019-10-27:
Thanks btw for providing your packages on github, and the description in the README. This made testing a lot easier :)

masa1120, 2019-10-30:
Hi, I appreciate your reply. Let me understand your code deeply to modify your code in the future if it is needed. From first, I started to understand your codes by JACO robot model. when display JACO URDF on RVIZ and your viewer, the direction of axis never match. why do you rotate an axis intentionally within your urdf_viewer.launch?

[image]

Your JACO only has three fingers and each finger only has one revolute joint. In the JACO model, the "contact_generator_node" never use DHParam::toDenavitHartenberg function to calculate DH-parameters. When I checked, your code skipped to call DHParam::toDenavitHartenberg function within Urdf2GraspIt::getDHParams function. If one finger only has one joint, the joint equals the root joint for each finger. The DH-parameters of the finger root joint are inserted all zero at the end of Urdf2GraspIt::getDHParams function. On case which each finger has multiple joints, I can not check if your code run well.

masa1120, 2019-10-30:
By using contacts_generator_node, can you display following robot model? that is shadow hand model without visual data. 

[shadowhand_with_ur10_left_simple2_without_visual.urdf.zip](https://github.com/JenniferBuehler/graspit-pkgs/files/3786984/shadowhand_with_ur10_left_simple2_without_visual.urdf.zip)

When you input abobe data into your contacts_generator_node, according to needs, please use following cpp. At the begin of main function, I wrote path scripts. You can change path information on the "shadowhand_with_ur10_left_simple2_without_visual.urdf". 

[contacts_generator_node.cpp.zip](https://github.com/JenniferBuehler/graspit-pkgs/files/3786991/contacts_generator_node.cpp.zip)

When "displayDHAxes" is true and  false, if your viewer shows same axes orientations on all joints, it means that your code never has problem. But when I check, the result was like this. On the following picture, there are two robot models. Left model is the axes orientation when "displayDHAxes" is false. This means the left axes orientation shows the axes of URDF definition. Right model is the axes orientation when "displayDHAxes" is true. This means the right axes orientation shows the axes of DH definition. Is it natural to have the same axis orientation on both definition of URDF and DH? In the case of JACO model, if each finger had multiple joints, I could search what was cause. Please tell me how to debug to confirm what is correct result.

[image]

masa1120, 2019-10-30:
URDF2GRASPIT can understand negative value and decimal value of axes tag in URDF definition? I changed all axes is positive int value from previous model. At present, what I want to ask you is way to solve the gap of axes orientation between URDF model and DH. But, I am wondering if the gap is problem or not...

closed 2019-11-13 by masa1120