JenniferBuehler/graspit-pkgs #55: Virtual Contacts of Panda Franka Emika

EZarbali, 2019-10-31:
my Name is Erguen Zarbali and im currently working on a cloud manufacturing project at the university of Stuttgart. 
The Robot we are working on is the Franka Emika Panda. We try to calculate the grasping points with GraspIt! and then send it to MoveIt! via ROS communication for futher process. 

I have a problem with the virtual contacts of the robot as im attaching it in this issue. 
I will highly appreciate it if you could help me and thanks in advance. 

[Panda.zip](https://github.com/JenniferBuehler/graspit-pkgs/files/3793876/Panda.zip)

JenniferBuehler, 2019-11-04:
Hi, I will look at this as soon as I can. Meanwhile, could you please describe in more detail what problem you have with the virtual contacts?

EZarbali, 2019-11-19:
First of all thank you for your reply.
I'm attaching a screenshot to this message to visualize the problem. You can see that the virtual contacts are not optimal for the Robot, as this is the "best grasp" which has been calculated. 
I've never worked with Virtual Contacts of robots so I am new to this topic. 
Again thank you in advance :)

[image]

JenniferBuehler, 2019-12-02:
Sorry for the late response, time just flies away ;)
On the screenshot I can't actually see contacts. They would be displayed with red lines. Did you set up the contacts [as described here](https://github.com/JenniferBuehler/graspit-pkgs/wiki/Conversion-with-urdf2graspit#step-2-optional-choose-your-contact-points-in-the-viewer)?

hoangcuongbk80, 2020-06-28:
Hi,

I'd like to use the gripper in Franka Emika Panda with GraspIt too.
Did you solve the problem?

JenniferBuehler, 2020-06-28:
Which issue exactly are you experiencing? I can only help if I have a bit more info :)

hoangcuongbk80, 2020-06-28:
Thank you for the reply.
I've tried to creat4e contacts. But I can not make it as expected. The contact should perpendicular to finger surface. However, as you can see from image below, it is not like that.

[image]

JenniferBuehler, 2020-06-28:
Oh ok, thanks for posting the details. This is strange, it shouldn't look like that. Which URDF are you using?

hoangcuongbk80, 2020-06-28:
I used Panda.xml in the model of @EZarbali uploaded above: [link](https://github.com/JenniferBuehler/graspit-pkgs/files/3793876/Panda.zip)
It is the gripper of Franka Emika robot.

JenniferBuehler, 2020-06-28:
Thank you! I will look into this as soon as I can and will get back to you

hoangcuongbk80, 2020-06-28:
I did this instead. How do you think?

[image]

JenniferBuehler, 2020-06-29:
Yes this looks as it should. Did the EGPlanner work with this?

hoangcuongbk80, 2020-06-29:
Hi Jennifer,

I got something like this: 

[image]

It looks good, doesn't it?

Could I have one more question?
I'd like to add a new object for grasp instead of the mug above. I have a stl file of its model. I know that the process should be: stl file -> urdf file -> .iv file + .xml file (graspit format). But I don't know which specific tools and softwares should be used. Could you give me some suggestions? (My system: ROS-Kinetic + Ubuntu 16.04).

Thank you for your help!

hoangcuongbk80, 2020-06-30:
I followed [the tutorial for urdf2graspit](https://github.com/JenniferBuehler/graspit-pkgs/wiki/Conversion-with-urdf2graspit) and got the error:

[image]

Note that I am trying to convert urdf file of an object (not robot) to .iv file and .xml file.

JenniferBuehler, 2020-07-01:
Yes that looks correct with the grasp.

Utdf2graspit is made for urdfs with gripper links, not objects. That would need a different (albeit similar) implementation. I can put it on my features TODO list,but not sure how soon I'd get around to it... Does your object only consist of one mesh? If yes, you could create an IV file of it and then hand craft the graspit xml file for it (taking eg the one for the small cube as template)

hoangcuongbk80, 2020-07-01:
I am not sure what you mean one mesh. They are YCB objects, for example:
[image]

If yes, so software should I use to create an IV file?

JenniferBuehler, 2020-07-02:
You mean it's an obstacle from the [ycb](https://www.ycbbenchmarks.com/object-models/) database?
Or YCB format (I haven't heard of)?

I haven't used that database before but it sounds like for each object there's several files with different info, like point cloud,mesh,images etc.
Which file have you opened (extension?) to show the mesh in your screenshot?

hoangcuongbk80, 2020-07-02:
Yes, it is an object from [YCB](http://ycb-benchmarks.s3-website-us-east-1.amazonaws.com/) database.

There's a textured.obj file (mesh), textured.mtl (info about material) and a xml file.
textured.obj is the file I opened.

[image]

JenniferBuehler, 2020-07-02:
Ok, that's a mesh file. You have to convert it to inventor (.iv) file. I have so far only converted files to IV via API but you should be able to find sth via Google, e..g [this](http://ivcon-tl.sourceforge.net/), though I didn't look at it very closely...

hoangcuongbk80, 2020-07-05:
Hey Jennifer,

I found that I can use (.wrl) file instead of (.iv). Therefore, I used Meshlab to convert (.obj) file to (.wrl) one.

Thank you very much for your help!

JenniferBuehler, 2020-07-05:
Ah yes, vrml works too! I forgot that :) I'm glad it worked!

I will close this for now, if you have any more questions just re-open.

Good luck with your project!

closed 2020-07-05 by JenniferBuehler