JenniferBuehler/graspit-pkgs #56: Unable to get DH parameters from URDF

SunnyKatyara, 2019-11-08:
Hi Jennifer, first of all let me appreciate your efforts on making such middlewares and frame works. Since, i am using URDF2graspit package to get DH parameters of robotic hand but unable to generate such values. Attached is the URDF of my robotic hand, i am using, can you look into it.
Looking forward to your response

Thank You

[urdf-five3.urdf.tar.gz](https://github.com/JenniferBuehler/graspit-pkgs/files/3823936/urdf-five3.urdf.tar.gz)

masa1120, 2019-11-08:
at first, did you check your urdf file to be able to import into gazebo or other simulator? if you succeed doing it, i think the program itself might not have problems. i recommend you upload stl files with your urdf file.

SunnyKatyara, 2019-11-08:
Thanks for the response. Yes i can load my urdf in Gazebo and control it. 

Attached is the complete folder with meshes and controller

Looking forward to hear from you.

Thank You

[urdf-five3.tar.gz](https://github.com/JenniferBuehler/graspit-pkgs/files/3824478/urdf-five3.tar.gz)

JenniferBuehler, 2019-11-17:
Hi, could you please describe the problem in more detail? What do you mean with you are unable to generate the DH parameters? Is there an error in urdf2graspit?

JenniferBuehler, 2020-12-07:
Closing due to inactivity, please re-open if still relevant.

closed 2020-12-07 by JenniferBuehler