JenniferBuehler/graspit-pkgs #58: Using urdf2graspit on Panda gripper - does not fully open or close in GraspIt!

karthikm-0, 2019-12-13:
Hi Jennifer,

I attempted to integrate the Panda gripper to GraspIt using your urdf2graspit package. I am using the urdf as defined here: https://github.com/erdalpekel/franka_ros/tree/simulation/franka_description/robots

However, upon converting and opening the gripper in GraspIt it does not fully open or close. I believe this is the case because in Gazebo/RViz I am able to fully open and close the gripper. Would you have some idea as to why this may be happening? I've attached two screenshots and the converted GraspIt hand model if you would like to inspect.

Thanks!
[image]
[image]
[Panda_GraspIt.zip](https://github.com/JenniferBuehler/graspit-pkgs/files/3962076/Panda_GraspIt.zip)

JenniferBuehler, 2019-12-17:
Hello, have you tried adjusting the eigengrasp values? The ones which are generated are just a rough example.

karthikm-0, 2020-01-02:
Thank you Jennifer. I tried adjusting the values but unfortunately, it does not fully extend.

closed 2020-01-02 by karthikm-0

chaiban, 2020-08-05:
Hello @karthikm-0 ,

Did you resolve the issue ? I am facing same issue.

Thank you !