JenniferBuehler/graspit-pkgs #59: package dependency

vignif, 2020-01-28:
during installation also the package in 
https://github.com/JenniferBuehler/joint-control-pkgs
should be installed due to dependency.

JenniferBuehler, 2020-02-07:
Thanks. Can you clarify why you needed the dependency? Maybe I've overseen a launch file or something which depends on that, but `joint-control-pkgs` is technically not required for the packages here. Are you using it from `grasp-execution-pkgs`, and had the dependency because of that? Or why did you need to install it? Thanks for pointing it out, it's possible I've overseen something ;)

vignif, 2020-02-09:
Hi Jennifer! and thanks for the reply! I cloned the whole graspit-pkgs in my <catkin_ws>/src and then run catkin_make in <catkin_ws> but the build failed. unfortunately i didn't save the log, but once joint-control-pkgs was included in the /src everything worked smoothed.

don't know how much it can help :) but thanks again for your package! 🥇

JenniferBuehler, 2020-12-07:
I just double checked. It doesn't need this dependency.
I suspect that you installed the jaco packages for the tutorial. That one needs the joint control packages. I guess the wiki install instructions weren't very clear on that the install instructions for the jaco packages have to be followed too. I amended the instructions.
Thanks for posting this!

closed 2020-12-07 by JenniferBuehler