JenniferBuehler/graspit-pkgs #61: EGPlanner inifinite loop for custom gripper

jaytjoa, 2020-05-29:
Hey, guys I am using a custom gripper and convert my urdf to the graspit format with the ros-pkg. It loads perfectly. The contact points were generated through the contact generator. But when it comes to  planning with graspit interface I have got a report with Neighbor gen loops: 11. And I do not know, what does exactly happen.

JenniferBuehler, 2020-06-01:
Hi, does it also crash, or is that only an output? Does it do any planning?
It may be worth asking this in the graspit repo, as they will know what is behind this message. If it's got to do with the generated contacts or the converted urdf, I can help look into this.

Cheers
Jennifer

jaytjoa, 2020-06-01:
Hi, thank you for your fast response. I have generated the urdf with the Solidworks and use the contact generator to define some contact points on the tip of the gripper. Unfortunately the egplanner does not work and I did not get any output or generated pose. 

Best regards,

Jay

JenniferBuehler, 2020-06-28:
Did you end up asking on the graspit repo? Any further insights?

JenniferBuehler, 2020-12-07:
Closing due to inactivity. Please re-open if still relevant

closed 2020-12-07 by JenniferBuehler