JenniferBuehler/graspit-pkgs #62: Code not working with ROS-neotic at ubuntu 20.04

usman151mrd, 2020-06-20:
Have you any plan to upgrade ROS Node for ROS-Neotic ???

JenniferBuehler, 2020-06-28:
At some point yes, but my time to put into this is extremely limited at the moment, so I'm not sure when... If you end up doing it, I'd welcome a PR :)

JenniferBuehler, 2020-12-07:
I just checked. Unfortunately, the Qt4 dependency is an issue on Noetic (on Ubuntu 20.04), which doesn't offer support for Qt4. While it is possible to install Qt4, the next stumble stone is SoQt4, which is not available (at least afaik) on 20.04.
So sorry to disappoint, but support for Noetic is going to be added only once GraspIt supports Qt5, or there is support for SoQt4 on 20.04.

closed 2020-12-07 by JenniferBuehler

vladi315, 2022-05-10:
Are there any updates or workaround on this matter? Can anyone recommend alternatives to graspit on Ubuntu 20.04?

JenniferBuehler, 2022-05-11:
I see you already asked [on the graspit repo](https://github.com/graspit-simulator/graspit/issues/174#issuecomment-1122475672). As far as I remember, Qt5 support was added but I'm not sure about the working state, I haven't checked in a while. SoQt4 is currently still a dependency here. There now is a Ubuntu package libsoqt520 which should work with Qt5, but again, I haven't had the time to test.
If anyone has time to try and test a migration to Qt5 and post a PR, that would certainly speed things up, as my time currently is extremely limited :(

DavidWatkins, 2022-05-29:
GraspIt! works on 20.04 and with noetic. I have no problems working with it at the moment using the following build instructions: 
```bash
# Graspit
$ sudo apt install -y cmake libsoqt520 libsoqt520-dev libblas-dev liblapack-dev libqhull-dev libeigen3-dev git
$ sudo apt update
$ mkdir $HOME/dependencies/
$ cd ~/dependencies
$ git clone https://github.com/graspit-simulator/graspit.git -b qt5
$ cd graspit
$ export GRASPIT=$PWD
$ mkdir build
$ cd build
$ cmake ..
$ make -j$(nproc)
$ sudo make install
```

RyanJiao, 2023-03-01:
Hi, I finished the installation as you instructed and I could suceessfully run the graspit_interface.launch. However, the gui will stop working when I import robot or .wld file. Would you kindly help me with that.

DavidWatkins, 2023-03-01:
Instead of responding to a thread from several months ago please open up a new issue with your context.

RyanJiao, 2023-03-01:
OK, thinks for reminding.