JenniferBuehler/graspit-pkgs #63: How to save grasp?

hoangcuongbk80, 2020-07-08:
Hi Jennifer,

I'd like to save grasp from EigeinGrasp Planners. Could you give me some suggestions?

I added:

`  FILE *f = fopen("grasp.txt","w");
  for (int i=0; i<mPlanner->getListSize(); i++) 
    mPlanner->getGrasp(i)->writeToFile(f);
  fclose(f);`

to egPlannerDlg.cpp and visualized Tx Ty Tz as point(x,y,z) as following:

[image]

I though the read points should be placed on object surface, but they do not.

JenniferBuehler, 2020-07-09:
Hi,

This question probably belongs to the graspit repo, I presume you weren't using my graspit-pkgs packages?
Generally, when you save a grasp, it would save the joint (denavit Hartenberg) parameters, and wrist pose, AFAIK. But it's been a while I dealt with the graspit core code in detail so I could remember this wrong...
I can take a look at the code and I'll be able to tell you exactly, but I won't have the time before next week unfortunately... Ping me if you haven't found the answer before then :)

hoangcuongbk80, 2020-07-13:
Hi Jennifer,

Yes I should ask the question in the graspit repo, but they are not likely be active. Thank you for replying all my questions, really really helpful.
 
For saving grasps, I solved it. Just need modify egPlannerDlg.cpp a little bit.

I am having a new question. EigenGrasp Planner works for object and robot in the scene. But if I add an obstacle to the scene, the planner doesn't functional anymore. Is there any way to make the planner work with object with obstacles?

JenniferBuehler, 2020-07-18:
Yes I've used obstacles (see tutorials in wiki) so it should work. However if your object lies on the table with suitable grasp points badly reachable by the gripper, the planner will fail because it finds collisions with the table for the suitable grasps. Make sure you have enough contact points on the object. Maybe increasing number of iterations helps, or re-planning a few times

hoangcuongbk80, 2020-07-20:
It works! Thank you!

closed 2020-07-20 by hoangcuongbk80