JenniferBuehler/graspit-pkgs #66: graspit_interface can't Import Robot on ROS-neotic, ubuntu 20.04

RyanJiao, 2023-03-01:
Hi, I finished the installation as @DavidWatkins instructed in this issue: #62 https://github.com/JenniferBuehler/graspit-pkgs/issues/62 . I could suceessfully run the graspit_interface.launch. However, the gui will stop working when I click the File -> Import Robot button. 

The gui will color will get a little bit darker than before. Is this something wrong with the qt5? 
[image]

RyanJiao, 2023-03-01:
[image]

KeerthivasanIITMadras, 2023-08-12:
I am too having the same issue

JenniferBuehler, 2023-08-13:
This would be a question for the `graspit-simulator` repo. I'm not actively working with Graspit at the moment, so I'm not as knowledeable about the current tricks and caveats ;)

closed 2023-08-13 by JenniferBuehler