JenniferBuehler/graspit-pkgs #67: Problem when using this package for robotiq_2f_85

gachiemchiep, 2024-07-18:
Hello @JenniferBuehler 
Thank you for your great package. 

I met a very strange bug. I search through entire wiki but couldn't find any possible solution. 
Would you mind take a look at this? 

I'm trying to convert robotiq_2f_85 urdf to graspit format

1. Here is my robotiq_2f_85 urdf

file:///datadrive/workspace/JIL/202311.aisin_object_tracking/dataset/aisin_works_bop/grasp/robotiq_2f_85_gripper_visualization.zip

2. Everything look fine when I view it

```bash
export XACRO=src/robotiq/robotiq_2f_85_gripper_visualization/urdf/robotiq_arg2f_85_model_2.xacro
rosrun xacro xacro --inorder $XACRO > ${XACRO}.urdf
rosrun urdf_viewer urdf_viewer_node ${XACRO}.urdf
```

[image]

3. Then I convert the urdf to graspit format and add some contacts point

```bash
# convert from urdf to graspit
rosrun urdf2graspit urdf2graspit_node ${XACRO}.urdf tmp robotiq_arg2f_base_link finger_joint left_inner_knuckle_joint right_inner_knuckle_joint  right_outer_knuckle_joint 
# add some contact points
rosrun urdf2graspit contacts_generator_node ${XACRO}.urdf tmp robotiq_arg2f_base_link finger_joint left_inner_knuckle_joint right_inner_knuckle_joint  right_outer_knuckle_joint 
```

everything look fine
[image]

4. But load the robotiq_arg2f_85_model_world.xml on graspit simulator gave me this 

[image]

JenniferBuehler, 2024-07-29:
Hello,
this looks to me like the joints are folded down to their extreme limits. If you open the gripper, does it look as expected? From what I can see, it loaded the contact points correctly, out of the green parts...

gachiemchiep, 2024-08-14:
@JenniferBuehler 
I tried to open the gripper in graspit simulation. 
But the following error occured

```bash
Robot joint interpolation error
Interpolation failed!

```

gachiemchiep, 2024-08-15:
@JenniferBuehler 
I have solved the problem

From graspit_simulation's menu, select *Grasp* -> *Eigen Grasp Interface* -> Edit the Amplitude value and the gripper is open. 

[image]

closed 2024-08-15 by gachiemchiep

JenniferBuehler, 2024-08-15:
Oh, great that you already solved it. And thanks for posting the solution too!